gotzl/hid-fanatecff #1: cannot locate "usbhid/usbhid.h" -

mkoreneff, 2020-11-07:
Sorry to ask here as I know it's (probably) not the right place, however i'm at a loss. 
I cannot find the ubshid header files package for ubuntu 1804. 
I'm getting this error when trying to make. 

```
/home/mickie/Documents/code/hid-fanatecff/hid-ftec.c:6:10: fatal error: usbhid/usbhid.h: No such file or directory
 #include "usbhid/usbhid.h"
          ^~~~~~~~~~~~~~~~~
compilation terminated.

```

gotzl, 2020-11-07:
For me, the header is found in the kernel sources. I don't know if there is a package for ubuntu.
What you can do is to download the kernel source tar and edit the `Kbuild` file to s.t. like
```
...
ccflags-y := -Idrivers/hid -I/path/to/linux-5.X.X/drivers/hid/
```

Hope that helps.

mkoreneff, 2020-11-07:
super helpful, thank you. 

I had to untar the linux-source files for my kernel version
`mickie@mickie-All-Series:/usr/src/linux-source-4.15.0$ sudo tar vxjf linux-source-4.15.0.tar.bz2`

then I updated Kbuild
`ccflags-y := -Idrivers/hid -I /usr/src/linux-source-4.15.0/linux-source-4.15.0/drivers/hid/`

I have another error now which is possibly due to an old version of usbhid in my kernel, maybe its time to update my ubuntu version from 18.04 LTS -> 20.04 LTS
`/home/mickie/Documents/code/hid-fanatecff/hid-ftec.c: In function ‘ftec_init’:
/home/mickie/Documents/code/hid-fanatecff/hid-ftec.c:125:35: error: ‘struct hid_report’ has no member named ‘application’
   report->id, report->type, report->application,
./include/linux/hid.h:1135:45: note: in definition of macro ‘dbg_hid’
   printk(KERN_DEBUG "%s: " fmt, __FILE__, ##__VA_ARGS__); \
                                             ^~~~~~~~~~~
`

gotzl, 2020-11-07:
Jea, seems like the `hid_report` changed.. I'll add a kernel switch. But you can just comment these lines 124-129, this is only debug output.

mkoreneff, 2020-11-07:
that worked, thanks for the help. Have compiled the driver now, will test and report back soon. 

failed
```
        dbg_hid(" ... %i %i %i %i %i %i\n%i %i %i %i\n\n",  
                /report->id, report->type, report->application,
```
commented 
```
        dbg_hid(" ... %i %i %i %i %i\n%i %i %i %i\n\n",
                report->id, report->type, // report->application,

```

closed 2020-11-07 by mkoreneff