gotzl/hid-fanatecff #4: ClubSport V2.5

mcoffin, 2020-11-14:
Continuation of the discussion from #3 about supporting the v2.5 clubsport wheelbase. Right now, the limiting factor seems to be disabling the centering spring that happens on startup.

@gotzl - here's the USB pcap from Windows. I *think* I did this correctly, but if not, then let me know how, and I'll snag a new one. Wheel was **off** at the start of the capture, was then turned **on** and I waited until the centering spring force was completely released, and gave the wheel a few turns before stopping the capture.

[wheel2.zip](https://github.com/gotzl/hid-fanatecff/files/5540900/wheel2.zip)

gotzl, 2020-11-14:
I've just had a quick look at your pcap and I've found some communication that might do the trick.. I've pushed another version to the cs-v25 branch, would be great if you could test it...
If it indeed works and you have some time to tinker around, it would be nice to nail it down to the minimum of statements that are actually needed. For this, you can (de)activate the bools in `hid-ftecff.c:438`. I think, chances are that either only `ab` or only `c` is needed ....

mcoffin, 2020-11-14:
Awesome. Thanks. I'll take a look at it tomorrow afternoon after my race is over (not gonna tinker so close to race time in case something goes tits up ;) ). Thanks again for the help; I've never really worked with USB anything on that level before, only graphics hardware.

mcoffin, 2020-11-16:
I can now confirm that, with *all* messages enabled, I get the expected behavior. I'll try to narrow it down to exactly what is completely necessary this afternoon, then gate the logic on the device ID, and get a PR ready along with [`mcoffin/hid-fanatecff@mcoffin.public-headers`](https://github.com/mcoffin/hid-fanatecff/tree/mcoffin.public-headers) which removes the requirement on extracting the kernel source for access to the private usbhid header.

gotzl, 2020-11-17:
Glad to hear! 

Sounds great, good idea to use `input.h`... I also don't mind to take over and add the device gate etc if you're running out of time!

mcoffin, 2020-11-19:
> I also don't mind to take over and add the device gate etc if you're running out of time!

Nah, we're all good, I just got around to running the testing.

It appears as if `c` is the only message that is necessary, so I'm just doing the device gating, and a little bit of cleanup and we'll be g2g.

gotzl, 2020-12-06:
Hi mcoffin,
because of a similar issue in #5 I went ahead and added the set_range into the init routine for all wheels. The master now also includes the CS v25 device ids, so in theory this should work for you then as well...
Cheers!

gotzl, 2021-11-26:
Please reopen if there are remaining issues.

closed 2021-11-26 by gotzl