gotzl/hid-fanatecff #5: Please Help

Croosi, 2020-12-06:
Hello, I happily found your driver for force feedback support. But I just can't get it to work. That's why I need to ask for help from you devs.
Little foreword... I just recently went Linux only, after a few attempts over the last years trying to do so. Since Linux is so much more than windows... I am learny every day something new.

I finaly got the driver package installed after the challenge of installing the missing library **usbhid** and compiling the module **ff-memless**. 
Currently I am running on **4.19.0-13-amd64 #1 SMP Debian 4.19.160-2 (2020-11-28) x86_64 GNU/Linux**.

Whenever I plugin the wheel **dmesg** tells me:

[12245.933330] usb 4-1.8: new full-speed USB device number 5 using ehci-pci
[12246.043570] usb 4-1.8: New USB device found, idVendor=0eb7, idProduct=0e03, bcdDevice= 6.56
[12246.043575] usb 4-1.8: New USB device strings: Mfr=1, Product=8, SerialNumber=0
[12246.043577] usb 4-1.8: Product: FANATEC CSL Elite Wheel Base
[12246.043579] usb 4-1.8: Manufacturer: Fanatec
[12246.045229] hid_map_usage: 34 callbacks suppressed
[12246.045232] Fanatec FANATEC CSL Elite Wheel Base: Invalid code 768 type 1
[12246.045236] Fanatec FANATEC CSL Elite Wheel Base: Invalid code 769 type 1
[12246.045241] Fanatec FANATEC CSL Elite Wheel Base: Invalid code 770 type 1
[12246.045243] Fanatec FANATEC CSL Elite Wheel Base: Invalid code 771 type 1
[12246.045245] Fanatec FANATEC CSL Elite Wheel Base: Invalid code 772 type 1
[12246.045247] Fanatec FANATEC CSL Elite Wheel Base: Invalid code 773 type 1
[12246.045248] Fanatec FANATEC CSL Elite Wheel Base: Invalid code 774 type 1
[12246.045255] Fanatec FANATEC CSL Elite Wheel Base: Invalid code 775 type 1
[12246.045261] Fanatec FANATEC CSL Elite Wheel Base: Invalid code 776 type 1
[12246.045265] Fanatec FANATEC CSL Elite Wheel Base: Invalid code 777 type 1
[12246.045387] input: Fanatec FANATEC CSL Elite Wheel Base as /devices/pci0000:00/0000:00:1d.0/usb4/4-1/4-1.8/4-1.8:1.0/0003:0EB7:0E03.0012/input/input32
[12246.045604] hid-generic 0003:0EB7:0E03.0012: input,hidraw12: USB HID v1.11 Joystick [Fanatec FANATEC CSL Elite Wheel Base] on usb-0000:00:1d.0-1.8/input0

Which is ok like that?

But, **lsmod** does not show me that **hid_fanatec.ko** or **ff_memless.ko** is running.

So I go to **/lib/modules/4.19.0-13-amd64/misc/** and start **ff-memless.ko** and **hid_fanatec.ko**  --- check **lsmod** and see that both are running.

That is pretty much how far I get here and do not know what else to do. 

I would really appreciate it, if someone could take on my issue, not only from the perspective that I wish for the force feedback to work, but also that I can learn something again about the process getting it to work.

Thanks alot  in adavance. Linux for the Win!

Greetings Thomas

gotzl, 2020-12-06:
Hi,
the issue is the VID/PID of your device 0EB7:0E03. My Wheel Base has a mode button and I've set it to 'PC mode', which is indicated by a red LED. This gives the expected device VID/PID 0EB7:0005.
The driver needs this VID/PID to know if it should activate for a certain device, see also hid-ftec.cc at the bottom.

So, the solutions are:
a) use the PC mode (if available)
b) adding youre VID/PID to the driver

Croosi, 2020-12-06:
Ohh, now it is working. I followed your advice and changed the PID in the hid-ftec.cc to my devices PID, because I am already running in PC-mode. :D Thank you so much!!

Croosi, 2020-12-06:
Sorry, I saw you edit a bit late. Here the dmesg in PS-Mode:

[ 2606.059896] usb 4-1.8: new full-speed USB device number 7 using ehci-pci
[ 2606.174252] usb 4-1.8: New USB device found, idVendor=0eb7, idProduct=0f31, bcdDevice= 6.56
[ 2606.174256] usb 4-1.8: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2606.174259] usb 4-1.8: Product: FANATEC CSL Elite Wheel Base
[ 2606.174261] usb 4-1.8: Manufacturer: Fanatec
[ 2606.174263] usb 4-1.8: SerialNumber: 00001096BF553920

Maybe you also are able to answere another question about the Force Feedback it self.

When I go to the "wrench" menu inside the wheel.... option FF on 100 the wheel has a permanent force to center itself and the force rises when uncentering it more. Unfortunatelly only on FF = 100 I get the full feedback from the game. Turning down FF decreses  the centering force but also the force from the game itself. Turning it FF = off kills the centering force at all but also the forces from the game. Is there a way to eliminate that constant center force? By the way. this constant centering force is already there when not in game. Maybe you know what I am talking about. :)

Greetings

gotzl, 2020-12-06:
Alright, I'll add your PID to the driver, although I'm abit puzzled about the number, I expected one like `000x`, but OK. I don't know anything about PS input devices, so no idea why it's not recognized at all in PS mode.

In this wrench menu, isn't there an option to modify autocenter/spring etc?
Maybe there is s.t. missing in the initialization of the wheel base. During some tests with the ClubSport v25, we found that the range has to be programmed, maybe this is required here as well. Can you check if
```
echo 1080 | sudo tee /sys/module/hid_fanatec/drivers/hid:ftec_csl_elite/0003:0EB7:0E03.*/range
```
has an effect? (not 100% sure about the sysfs path, but you'll figure it out ;))

Croosi, 2020-12-06:
Hey, that command did the trick. Is there some way to add that in the code or somewhere, so I do not need to execute that command each time?

gotzl, 2020-12-06:
Yes, I'll add that to the code. Thanks for reporting!

gotzl, 2020-12-06:
Just added the code. Could you please test it and report back? Thanks.

Croosi, 2020-12-06:
Just tested it... Works flawless :D 

Great job, I am impressed.

closed 2020-12-07 by gotzl