gotzl/hid-fanatecff #8: CSL Elite Racing Wheel + ClubSport Pedals V3

dinuxlt, 2021-01-15:
Hi,

i have `CSL Elite Racing Wheel - officially licensed for PlayStation™` and `ClubSport Pedals V3` and it works very good. It's much better then Lohitech G29 :) BIG thanks for you job here.

What works:
* stearing
* FFB
* all buttons
* pedals

What NOT works:
* RPM LEDs on base 
* LED on wheel (it's always green)
* these pedals has small motors, who can spin and generate vibration to pedals (ABS effect)

My question here, it's posible some how enable/disable pedals vibration from command line?

And if you need any help with these my devices, please ask. I have windows only on VirtualBox

gotzl, 2021-01-16:
Hi,

the issue with LEDs and display is (afaik), that applications use the 'Fanatec SDK' to communicate with the wheel.  I have no idea how this SDK works and how we can get it working in wine ...
As an alternative, some applications offer telemetry data, like AC and F1 2020. This telemetry data can be captured and fed to the driver. This is what I tried to do here: https://github.com/gotzl/hid-fanatecff-tools, but its work in progess.

To get your pedals vibration working, we need to learn the commands send to the pedals. Is it possible to test the vibration effect in the Fanatec UI in Windows ? If so, then you need to capture the USB traffic (using eg Wireshark) while enabling/disabling the effect.
I also need the product ID of the pedals if they are connected via USB.

dinuxlt, 2021-01-16:
Now i'm connecting these pedals to wheel base. But there is a possibility connect directly to PC.

When i have free time, i will try to connect these to windows (virtual box) and check if this posible manipulate with these pedals motors.

From: https://fanatec.com/ja-en/faq/details/id/80
```
No, not in general. If you want to use the Fanatec SDK please contact our support team and mention what your plans are to use the SDK. After reviewing those plans you might receive a NDA which has to be signed before the SDK would be sent to  you.
```

Maybe they (Fanatec) can provie some data how to implement all features to this driver?

dinuxlt, 2021-01-21:
Attaching ZIP file
[pedals.zip](https://github.com/gotzl/hid-fanatecff/files/5852664/pedals.zip)

Inside this zip file you will find two files, trottle.cap and brake.cap.

I run USBCap and pressed each button for three times

[image]

Pedals are connected to wheel base, so no new product ID here.

gotzl, 2021-01-24:
Thanks for the data. This is already helpful, I've found the instructions send to the wheel. But I'm not completely sure how to interpret them. I guess, I can make the pedals rumble, but what I don't understand right now is, how this is done from the game. Do all games support this just like that? Or is the `Fanatec SDK` involved? Also, is the rumble strength variable?

I also wonder how I can detect the different pedals one can connect to the wheel. Could you disconnect the pedals, then open the Fanatec Wheel Props, start recording USB traffic, and then connect the pedals?

dinuxlt, 2021-01-29:
Some updates from my side.

This time connected `ClubSport Pedals V3` directly to USB port.
`Bus 001 Device 032: ID 0eb7:183b Endor AG ClubSportPedal`

Please find attached zip file:
[pedals_part2.zip](https://github.com/gotzl/hid-fanatecff/files/5895670/pedals_part2.zip)

You will find three files. Brake ant throttle as for the first time, pressed three time eatch.

Last file `connect.cap`  is USB capture when it was connected.
I disconnected pedals, started capture and then connected. After that pressed `properties` (that small dialog in background) and then pressed one time rumble button (forgot which one that was).

Dialog on Windows when pedals are connected without wheel base, looks like this:
[image]

> Also, is the rumble strength variable?

No

> Do all games support this just like that? 

I have no idea :) I asked one person, which has same pedals and plays ACC on windows, he said that pedals not rumble on ACC for him.

> Or is the Fanatec SDK involved?

I think all these features like rumble, led's and displays on wheels are Fanatec SDK involved.
Found this dialog on ACC settings:
[image]

gotzl, 2021-02-03:
Hi,

if just pushed an update to the `next` branch. Maybe you can try this? If you do so, make sure to really use the new module ..

When you connect the wheel via USB, it should create a sysfs file called `rumble`. Try writing 
* 16776960 -> both pedals should rumble
* 16711680 -> throttle pedal rumble
* 65280 -> break pedal rumble
* 0 -> stop rumble
to it.

For example s.t. like `echo 65280 | sudo tee /sys/module/hid_fanatec/drivers/hid:ftec_csl_elite/0003:0EB7:183b.*/rumble`

> I think all these features like rumble, led's and displays on wheels are Fanatec SDK involved.

Yes, it looks like that... As long as we can't use FanatecSDK in linux, I think it should be possible to exploit telemetry data from games to get the pedals vibrating, similar to what I did for LEDs/display... I'll look into that.

gotzl, 2021-02-03:
Ah, one more thing. I still wonder how the base communicates connected pedals. Could you open up the fanatec properties page w/o connected pedals, then start capturing USB and connect the pedals to the base? (I assume that the fanatec properties page picks up the pedals when connecting, right?)

berarma, 2021-02-03:
Features other than axes/buttons and wheel FFB are handled through the FanatecSDK. That means there are no standard API for those features.

I don't know how the SDK works. It most probably communicates with the Fanatec driver. This can't work on Wine. I guess it would need some kind of process mimicking the Fanatec driver and using the Linux driver instead.

Our best bet for this to be supported sometime is that it gets standarized and included into DirectInput/XInput.

dinuxlt, 2021-03-12:
Sorry for very late answer here.

Tested all these commands, rumble works as you describet. There is no diference, pedals connected directly to PC or via wheel base all rumble command works as expected.

But one thing here, if pedals are connected directly to PC `dmesg` shows that "CSL Elite" are connected.
I think this device https://github.com/gotzl/hid-fanatecff/blob/64e8f84a87b339decc045ccfa7d91f25660239f9/hid-ftec.c#L290 should raport "ClubSport Pedals V3"

gotzl, 2022-02-05:
Closing this ticket, as basic support for rumble with "ClubSport Pedals V3" exists now. However, to make this work with games, some more work is needed in userspace (eg hid-fanatecff-tools).

closed 2022-02-05 by gotzl