gotzl/hid-fanatecff #9: Let's test it with the Fanatec Podium DD1

Gonzo2O28, 2021-01-16:
lsusb:
Bus 005 Device 010: ID 0eb7:0006 Endor AG FANATEC Podium Wheel Base DD1

The Pedals Clubsport V3 are plugged into my Wheel-Base.
I can later attach it by usb and check it out.

Let's see if we can bring solid simracing to linux finally. ;)

gotzl, 2021-01-17:
Hi,
I've added the PID to the driver in the 'dd1' branch.
You may check it out, compile and install/load the module, then plug in the wheelbase and check 'dmesg'.  There should be a line `ftec_csl_elite 0003:0EB7:0006.....`. Then use `fftest` to test if the constant force works.
If it doesn't work just like that, then maybe it needs some special initialization or Fanatec uses a different protocol... In this case, we need to reverse engineer abit ;) But let's see.

Btw: Why the duplicate #10 ?

Gonzo2O28, 2021-01-18:
it moves slightly but only once with and effect 1
fftest /dev/input/by-id/usb-Fanatec_FANATEC_Podium_Wheel_Base_DD1-event-joystick

Gonzo2O28, 2021-01-18:
Jfyi: I had to create this directory manually, otherwise the installation does not work:
/lib/modules/${KVERSION}/misc/

gotzl, 2021-01-18:
> it moves slightly but only once with and effect 1

Can you elaborate? 
For me, if I play the 'Constant Force' effect, the wheel turns anti-clockwise with constant force for ~20s.

> No effects in one game i tried with proton.

Which game did you test? In some games, FFB generally doesn't work. Also, I had some problems lately with steam input. See this comment: https://github.com/ValveSoftware/Proton/issues/2366#issuecomment-738989008

> I had to create this directory manually

Does `/lib/modules/${KVERSION}/kernel/drivers/misc/` exist? Which distribution do you use?

Gonzo2O28, 2021-01-18:
Arch Linux with linux-clear kernel.

it spinns fully to the left and stays there if previous spun to the right.

I tried rFactor 1 and rFactor 2 (because best known sims to me, which were working with the new-lgff logitech kernel drivers).
I can try another game (if i have it). What do you recommend?

That directory does not exists either.

l /lib/modules/5.10.8-1-clear/kernel/drivers/
drwxr-xr-x 22 root root 3,5K 18. Jan 15:16 ./
drwxr-xr-x 11 root root 3,5K 18. Jan 15:16 ../
drwxr-xr-x  2 root root 3,5K 18. Jan 15:16 acpi/
drwxr-xr-x  2 root root 3,5K 18. Jan 15:16 ata/
drwxr-xr-x  3 root root 3,5K 18. Jan 15:16 base/
drwxr-xr-x  2 root root 3,5K 18. Jan 15:16 bluetooth/
drwxr-xr-x  4 root root 3,5K 18. Jan 15:16 char/
drwxr-xr-x  3 root root 3,5K 18. Jan 15:16 crypto/
drwxr-xr-x  2 root root 3,5K 18. Jan 15:16 edac/
drwxr-xr-x  3 root root 3,5K 18. Jan 15:16 gpu/
drwxr-xr-x  2 root root 3,5K 18. Jan 15:16 hid/
drwxr-xr-x  3 root root 3,5K 18. Jan 15:16 hwmon/
drwxr-xr-x  4 root root 3,5K 18. Jan 15:16 i2c/
drwxr-xr-x  2 root root 3,5K 18. Jan 15:16 input/
drwxr-xr-x  3 root root 3,5K 18. Jan 15:16 leds/
drwxr-xr-x  2 root root 3,5K 18. Jan 15:16 md/
drwxr-xr-x  7 root root 3,5K 18. Jan 15:16 media/
drwxr-xr-x  4 root root 3,5K 18. Jan 15:16 net/
drwxr-xr-x  2 root root 3,5K 18. Jan 15:16 pinctrl/
drwxr-xr-x  3 root root 3,5K 18. Jan 15:16 platform/
drwxr-xr-x  3 root root 3,5K 18. Jan 15:16 usb/
drwxr-xr-x  3 root root 3,5K 18. Jan 15:16 video/

Gonzo2O28, 2021-01-18:
I just found this tool: https://github.com/berarma/ffbtools and ffb worked as expected.

tried this with a force-feedback file been shipped with it.
ffbplay -t constant.ffb -d /dev/input/by-id/usb-Fanatec_FANATEC_Podium_Wheel_Base_DD1-event-joystick

Gonzo2O28, 2021-01-18:
Jfyi: that base has three modes.

Compatibility Mode
Bus 005 Device 017: ID 0eb7:0004 Endor AG FANATEC Podium Wheel Base DD1
XBOX Mode
Bus 005 Device 018: ID 0eb7:0f54 Endor AG FANATEC Podium Wheel Base DD1
PC Mode
Bus 005 Device 019: ID 0eb7:0006 Endor AG FANATEC Podium Wheel Base DD1

gotzl, 2021-01-18:
Jeah, I think it should work already. Just checked rfactor2 myself, and this is also not working for me. I'll look into that.

Best would be to test with a native game like F1 2018 (?) or DirtRally1. 
I play alot AC/ACC, F1 2020 and DirtRally2. May you find one of those games?
(I think Automobilista 2, Project CARS2 and KartKraft worked for me as well).

Concerning the modes: use PC mode ;)

Gonzo2O28, 2021-01-18:
f1 2020 is downloading, i ll try that.
yeah pc mode for sure, just wanted to send you the usb ids.

gotzl, 2021-01-18:
Just checked F12020 again. I had to choose 'Disable Steam Input' in the games steam properties to get the controllers detected in-game. Also, the FF is not the best in this game, I think it needs more effects apart from FF_CONSTANT.
I'm already working on an updated driver that works analog to new-lgff.

Gonzo2O28, 2021-01-18:
hey, tried f1 2020, too. DX 12 crashed when entering track, beside of this it worked like a charme. The game detected the wheel and the ui control mapping worked out of the box. i mapped the axis to race and set up the ffb. the ffb felt bad. inverted maybe?
In WIndows you have some wheels which need the forces inverted as negative numbers.

We are close ;). rF2 has by far the best ffb out there, lets see if we can fix it.

gotzl, 2021-01-18:
Could you verify that FF realy works in rf2 with new-lgff?

Gonzo2O28, 2021-01-18:
> Could you verify that FF realy works in rf2 with new-lgff?

Yes it did!

Gonzo2O28, 2021-01-18:
https://github.com/berarma/new-lg4ff
Those who commented here say so, too: https://www.protondb.com/app/365960

Gonzo2O28, 2021-01-18:
dmesg some messages might be interesting

[ 1204.632522] usb 5-3.2: Product: FANATEC Podium Wheel Base DD1
[ 1204.632523] usb 5-3.2: Manufacturer: Fanatec
[ 1204.728686] Fanatec FANATEC Podium Wheel Base DD1: Invalid code 768 type 1
[ 1204.728688] Fanatec FANATEC Podium Wheel Base DD1: Invalid code 769 type 1
[ 1204.728688] Fanatec FANATEC Podium Wheel Base DD1: Invalid code 770 type 1
[ 1204.728689] Fanatec FANATEC Podium Wheel Base DD1: Invalid code 771 type 1
[ 1204.728690] Fanatec FANATEC Podium Wheel Base DD1: Invalid code 772 type 1
[ 1204.728690] Fanatec FANATEC Podium Wheel Base DD1: Invalid code 773 type 1
[ 1204.728691] Fanatec FANATEC Podium Wheel Base DD1: Invalid code 774 type 1
[ 1204.728691] Fanatec FANATEC Podium Wheel Base DD1: Invalid code 775 type 1
[ 1204.728692] Fanatec FANATEC Podium Wheel Base DD1: Invalid code 776 type 1
[ 1204.728692] Fanatec FANATEC Podium Wheel Base DD1: Invalid code 777 type 1
[ 1204.728736] input: Fanatec FANATEC Podium Wheel Base DD1 as /devices/pci0000:00/0000:00:08.1/0000:0b:00.3/usb5/5-3/5-3.2/5-3.2:1.0/0003:0EB7:0006.000F/input/input40
[ 1204.728829] input: Fanatec FANATEC Podium Wheel Base DD1 as /devices/pci0000:00/0000:00:08.1/0000:0b:00.3/usb5/5-3/5-3.2/5-3.2:1.0/0003:0EB7:0006.000F/input/input41
[ 1204.728910] hid-generic 0003:0EB7:0006.000F: input,hiddev101,hidraw14: USB HID v1.11 Joystick [Fanatec FANATEC Podium Wheel Base DD1] on usb-0000:0b:00.3-3.2/input0
[ 1205.270213] calling  ftec_csl_elite_init+0x0/0x1000 [hid_fanatec] @ 5212
[ 1205.270492] /tmp/hid-fanatecff/hid-ftec.c: ftec_probe: ifnum 0
[ 1205.270571] input: Fanatec FANATEC Podium Wheel Base DD1 as /devices/pci0000:00/0000:00:08.1/0000:0b:00.3/usb5/5-3/5-3.2/5-3.2:1.0/0003:0EB7:0006.000F/input/input42
[ 1205.322048] ftec_csl_elite 0003:0EB7:0006.000F: input,hiddev101,hidraw14: USB HID v1.11 Joystick [Fanatec FANATEC Podium Wheel Base DD1] on usb-0000:0b:00.3-3.2/input0
[ 1205.322051] /tmp/hid-fanatecff/hid-ftec.c:  ... 1 1 1 56 1
-128 127 -128 127

[ 1205.322052] /tmp/hid-fanatecff/hid-ftecff.c:  ... setting FF bits
[ 1205.322053] /tmp/hid-fanatecff/hid-ftecff.c: setting range to 1080
[ 1205.322153] initcall ftec_csl_elite_init+0x0/0x1000 [hid_fanatec] returned 0 after 50718 usecs

Gonzo2O28, 2021-01-18:
if you go the new-lg4ff wheel it would be awesome if the driver would be compatible with oversteer and pyLinuxWheel afterwards. Those tools halb you to configure it a lot.

i see the driver sets the range. i can set the range inside the wheelbase as well.

Gonzo2O28, 2021-01-18:
Oversteer
https://aur.archlinux.org/packages/oversteer/
pyLinuxWheel
https://gitlab.com/OdinTdh/pyLinuxWheel

Gonzo2O28, 2021-01-18:
https://github.com/berarma/ffbtools
that might help you as well.

berarma, 2021-01-18:
Hi. I'm the Oversteer, new-lg4ff and ffbtools author.

rFactor 2 uses sine wave effects but in a special way. It uses the offset of the wave effect to produce a constant effect, and all other parameters are set to values that don't produce any kind of wave.

Maybe there was a time when wheels didn't have constant effects but had sine wave effects. This means you could do a fast hack using just the offset of the sine wave effect to drive the gain of a constant effect and it would work perfectly well in this game.

Most games work well with just constant effects, rFactor 2 (don't remember about rF1) needs the sine-wave as a constant force. Some also use additional conditional effects. A few don't use constant force effects at all.

The direction of the effect is used, it can invert the force or even scale it down.

In the open/cloed issues for ffbtools there's logs with the FFB output for several games.

I want to help, it's just that I don't own a Fanatec wheel.

Gonzo2O28, 2021-01-18:
Hey great to have you helping here!
I have the DD1 plus the Formula V2 and the Porsche GT3 R Wheel along with the V3 Pedals

This is all i have and could provide to test:
https://fanatec.com/eu-de/racing-wheels-wheel-bases/wheel-bases/podium-wheel-base-dd1
https://fanatec.com/eu-de/lenkraeder/podium-lenkrad-porsche-911-gt3-r-wildleder
https://fanatec.com/eu-de/lenkraeder/clubsport-steering-wheel-formula-v2
https://fanatec.com/eu-de/pedale/clubsport-pedals-v3

The pedals are currently attached by rj11 of the podium wheelbase. They can be attached by usb as well. Both have advantages and disadvantages.

dinuxlt, 2021-01-19:
> Does /lib/modules/${KVERSION}/kernel/drivers/misc/ exist? Which distribution do you use?

I can confirm. On Archlinux `misc` directory does not exist, and `make install` fails. You need to create this directory manually.

berarma, 2021-01-19:
Why not use DKMS? It's very easy, just reuse https://github.com/berarma/new-lg4ff/blob/master/dkms.conf changing package, name, module name, version and destination.

mkoreneff, 2021-01-19:
This is certainly the best option IMO.
I looked at adding DKMS however as I run secure boot, signing the driver
looked a little complicated, and potentially insecure.

gotzl, 2021-01-19:
Welcome @berarma at this project :) And thank you very much for offering to help and the comment about rf2. In fact, I've just implemented the `FF_SINE` in the `next` branch and I get FF in rfactor2! Nice ;) I also have to say, that in the `next` branch I heavily copied from your new-lgff code. I hope you're OK with that. At some point we could try to merge, or extract common stuff...

@Gonzo2O28 Since the basic `FF_CONSTANT` seems to be working already on the DD1, next step would be to check for the other effects. Could you checkout the `next` branch and see whats working for you?

Concerning the installation directory, I've just changed it to `/lib/modules/${KVERSION}/kernel/drivers/hid`. I'll create a ticket for further improvements on the installation method. (Note, you may have to manually remove the old driver module from the `misc` dir).

Tools like `oversteer` or `pyLinuxWheel` can surly add support for Fanatec devices. The driver exposes some sysfs files the same way as LG driver does. Not sure how much overlap exists between configuring Fanatec and LG devices, though. 
Also, I've already started s.t. similar in this project: https://github.com/gotzl/hid-fanatecff-tools. But my approach is little different, since I use a DBus daemon to access the sysfs files and the UI then only talks to DBus.
But I'm open for any suggestions ;)

Gonzo2O28, 2021-01-19:
I will try asap! Thank you!

Gonzo2O28, 2021-01-19:
hm a first test did not work. Might i have to remove the older version? If so, how?

fftest /dev/input/by-id/usb-Fanatec_FANATEC_Podium_Wheel_Base_DD1-event-joystick
Force feedback test program.
HOLD FIRMLY YOUR WHEEL OR JOYSTICK TO PREVENT DAMAGES

Device /dev/input/by-id/usb-Fanatec_FANATEC_Podium_Wheel_Base_DD1-event-joystick opened
Features:
* Absolute axes: X, Y, Z, RX, RY, RZ, Throttle, Rudder, Hat 0 X, Hat 0 Y, Misc ,
[FF 00 03 00 00 01 00 00 ]
* Relative axes:
[00 00 ]
* Force feedback effects types: Constant, Gain,
Force feedback periodic effects:
[00 00 00 00 00 00 00 00 00 00 04 00 01 00 00 00 ]
* Number of simultaneous effects: 16

Setting master gain to 75% ... OK
Uploading effect #0 (Periodic sinusoidal) ... Error:: Invalid argument
Uploading effect #1 (Constant) ... OK (id 0)
Uploading effect #2 (Spring) ... Error: Invalid argument
Uploading effect #3 (Damper) ... Error: Invalid argument
Uploading effect #4 (Strong rumble, with heavy motor) ... Error: Invalid argument
Uploading effect #5 (Weak rumble, with light motor) ... Error: Invalid argument
Enter effect number, -1 to exit

Gonzo2O28, 2021-01-19:
Ok, looking good, first it didnt work until i restarted linux.

I have to reduce the power because on 100% this IS NOISY as hell, my daughter sleeps nextdoor :D.

fftest /dev/input/by-id/usb-Fanatec_FANATEC_Podium_Wheel_Base_DD1-event-joystick
Force feedback test program.
HOLD FIRMLY YOUR WHEEL OR JOYSTICK TO PREVENT DAMAGES

Device /dev/input/by-id/usb-Fanatec_FANATEC_Podium_Wheel_Base_DD1-event-joystick opened
Features:
* Absolute axes: X, Y, Z, RX, RY, RZ, Throttle, Rudder, Hat 0 X, Hat 0 Y, Misc ,
[FF 00 03 00 00 01 00 00 ]
* Relative axes:
[00 00 ]
* Force feedback effects types: Constant, Periodic, Spring, Damper, Rumble,
Force feedback periodic effects: Square, Triangle, Sine, Saw up, Saw down,
[00 00 00 00 00 00 00 00 00 00 2F 1F 00 00 00 00 ]
* Number of simultaneous effects: 16

Uploading effect #0 (Periodic sinusoidal) ... OK (id 0)
Uploading effect #1 (Constant) ... OK (id 1)
Uploading effect #2 (Spring) ... OK (id 2)
Uploading effect #3 (Damper) ... OK (id 3)
Uploading effect #4 (Strong rumble, with heavy motor) ... OK (id 4)
Uploading effect #5 (Weak rumble, with light motor) ... OK (id 5)

Gonzo2O28, 2021-01-19:
only this effect is working:
0:
Uploading effect #0 (Periodic sinusoidal) ... OK (id 0)
1: is not anymore

Gonzo2O28, 2021-01-19:
rF2 ffb works.
For whatever reason my rF2 under proton is very slow. other titles work fine, i tried everything, proton versions, proton ge custom, kernels,  nothing helped with rF2 yet.

Gonzo2O28, 2021-01-19:
can you influence the hz of the usb calls?
rF2 uses very high update-rates.
I can ask the devs for the exact number if needed.

berarma, 2021-01-19:
> Welcome @berarma at this project :) And thank you very much for offering to help and the comment about rf2. In fact, I've just implemented the `FF_SINE` in the `next` branch and I get FF in [...]

I'm glad you copied and didn't reinvent the wheel. I hope we can work together towards something that can be integrated into the kernel.

> Tools like `oversteer` or `pyLinuxWheel` can surly add support for Fanatec devices. The driver exposes some sysfs files the same way as LG driver does. Not sure how much overlap exists between configuring Fanatec and LG devices, though.

I can add the USB ids for the Fanatec devices supported by this module. I've seen there can be several modes with different ids. Do you want to list the ones Oversteer should detect in berarma/oversteer#47?

> Also, I've already started s.t. similar in this project: https://github.com/gotzl/hid-fanatecff-tools. But my approach is little different, since I use a DBus daemon to access the sysfs files and the UI then only talks to DBus.
> But I'm open for any suggestions ;)

I think that's better than accessing the sysfs entries directly. But adding a dependency that users must install manually is not ideal. Do you intend to make hid-fanatecff-tools more generic?

Maybe we should get to talk to some mainstream devs about the best way to do it. I've always thought that the libSDL proejct should improve its wheel support.

Gonzo2O28, 2021-01-19:
That one is PC mode of the DD1 i will ask friends for the dd2 codes.
Bus 005 Device 011: ID 0eb7:0006 Endor AG FANATEC Podium Wheel Base DD1

i see no reason why we should add Xbox/PS4 or compatibility mode of that wheel, this just opens door for bugs imho.

dinuxlt, 2021-01-19:
> Concerning the installation directory, I've just changed it to /lib/modules/${KVERSION}/kernel/drivers/hid

Tested on `master` and `next`, it works now as expected.

Looks like `FF_CONSTANT` works now on `CSL Elite` to. Tested with Automobilista 2. There is setting `menu spring` which one returns wheel to center position in main menu. With `next`  brach this efect is working as expected.

Gonzo2O28, 2021-01-19:
Tried Assetto Corsa: it has no ffb.
I assume this is because only "Periodic sinusoidal" is working for my dd1 and not "Constant", even if my fftest reports all as working, which is wrong.

gotzl, 2021-01-19:
> Ok, looking good, first it didnt work until i restarted linux.

When you install the new module, you first need to remove the 'old' one with `sudo rmmod hid-fanatec` and load the new one wiht `sudo modporbe hid-fanatec` (or replug the device).

> Tried Assetto Corsa: it has no ffb.

Could you try AC with the master branch? I'm a bit puzzled that the 'constant' effect doesn't work on the `next` branch... fftest only reports which capability bits are set in the driver, that doesn't mean that the effect actually is properly implemented.

Gonzo2O28, 2021-01-19:
hm, where is my comment?!
ffb works in AC and in rF2

fftest /dev/input/by-id/usb-Fanatec_FANATEC_Podium_Wheel_Base_DD1-event-joystick
Force feedback test program.
HOLD FIRMLY YOUR WHEEL OR JOYSTICK TO PREVENT DAMAGES

Device /dev/input/by-id/usb-Fanatec_FANATEC_Podium_Wheel_Base_DD1-event-joystick opened
Features:
* Absolute axes: X, Y, Z, RX, RY, RZ, Throttle, Rudder, Hat 0 X, Hat 0 Y, Misc ,
[FF 00 03 00 00 01 00 00 ]
* Relative axes:
[00 00 ]
* Force feedback effects types: Constant, Periodic, Spring, Damper, Rumble,
Force feedback periodic effects: Square, Triangle, Sine, Saw up, Saw down,
[00 00 00 00 00 00 00 00 00 00 2F 1F 00 00 00 00 ]
* Number of simultaneous effects: 16

Uploading effect #0 (Periodic sinusoidal) ... OK (id 0)
Uploading effect #1 (Constant) ... OK (id 1)
Uploading effect #2 (Spring) ... OK (id 2)
Uploading effect #3 (Damper) ... OK (id 3)
Uploading effect #4 (Strong rumble, with heavy motor) ... OK (id 4)
Uploading effect #5 (Weak rumble, with light motor) ... OK (id 5)
Enter effect number, -1 to exit

0-3 work, 4 and 5 show no reaction on the wheel.

Gonzo2O28, 2021-01-19:
i have to correct me, 0,1,2,3 are showing effects. I just did not notice them, because i set the ffb forces very low, because 0 is creating A LOT of noise on the direct drive! 4 and 5 do definately nothing.

berarma, 2021-01-19:
fftest isn't the best tool for testing wheels. It was created with joysticks in mind.

Gonzo2O28, 2021-01-19:
AC Competitizione also working fine. i need to fix my low rF2 fps....

berarma, 2021-01-20:
> can you influence the hz of the usb calls?
> rF2 uses very high update-rates.
> I can ask the devs for the exact number if needed.
> AC Competitizione also working fine. i need to fix my low rF2 fps....

The `next` branch uses code from `new-lg4ff` to serve requests and should be very fast processing them. It's well tested in Logitech wheels.

I remember reading somewhere that Fanatec wheels firmware processed requests at 100 Hz. Logitech wheels can process requests upto 500 Hz. You can change the code to work at 100 Hz by changing the value from 2 to 10 in line: https://github.com/gotzl/hid-fanatecff/blob/3f5d508420d5eaf503f565ae3ab016ea4e6c2ed9/hid-ftecff.c#L17

I'm not sure that's the problem but there's a chance the device firmware can't process so many requests and it's slowing down the CPU.

> I contacted Fanatec, lets see if we might get some more hardware to test.

Can you get them to send me one? Just for testing. :smiley:

Gonzo2O28, 2021-01-20:
Yeah, rF2 sends at at least 400 or 666HZ!
So 100Hz would be way too low. The direct drive wheels must be abĺe to do better!

Gonzo2O28, 2021-01-20:
Oversteer has partly support now for Fanatec.
I gave some feedback and it seems that some driver work has to be done, too.
Watch this: https://github.com/berarma/oversteer/issues/47

Gonzo2O28, 2021-01-20:
I am getting some errors with the ffbtools tool while playing the conditional.ffb file.
sine.ffb and constant.ffb are working fine. 
FFBTools
https://github.com/berarma/ffbtools

Driver or Tool?

./ffbplay -d /dev/input/by-id/usb-Fanatec_FANATEC_Podium_Wheel_Base_DD1-event-joystick -t ../tests/conditional.ffb
Force feedback playback tool.

Using device /dev/input/by-id/usb-Fanatec_FANATEC_Podium_Wheel_Base_DD1-event-joystick.

CAUTION: The forces applied might be dangerous.

Playing ../tests/conditional.ffb

00000000 # SPRING effect (5s)
# SPRING effect (5s)
00000000 > UPLOAD id:-1 dir:16384 type:SPRING left_coeff:32767 right_coeff:32767
00000000 < 0 id:0
00000000 > PLAY 0 1
05000000 # DAMPER effect (5s)
# DAMPER effect (5s)
05000000 > STOP 0
05000000 > REMOVE 0
05000000 > UPLOAD id:-1 dir:16384 type:DAMPER left_coeff:32767 right_coeff:32767
05000000 < 0 id:0
05000000 > PLAY 0 1
10000000 # FRICTION effect (5s)
# FRICTION effect (5s)
10000000 > STOP 0
10000000 > REMOVE 0
10000000 > UPLOAD id:-1 dir:16384 type:FRICTION left_coeff:32767 right_coeff:32767
ERROR: uploading effect failed (Invalid argument) [src/ffbplay.c:86]
10000000 < 0 id:0
10000000 > PLAY 0 1
15000000 # INERTIA effect (5s)
# INERTIA effect (5s)
15000000 > STOP 0
15000000 > REMOVE 0
ERROR: removing effect failed (Invalid argument) [src/ffbplay.c:115]
15000000 > UPLOAD id:-1 dir:16384 type:INERTIA left_coeff:32767 right_coeff:32767
ERROR: uploading effect failed (Invalid argument) [src/ffbplay.c:86]
15000000 < 0 id:0
15000000 > PLAY 0 1
20000000 > STOP 0
20000000 > REMOVE 0
ERROR: removing effect failed (Invalid argument) [src/ffbplay.c:115]

gotzl, 2021-01-24:
> can you influence the hz of the usb calls?
rF2 uses very high update-rates.

Why do you ask? 

> Driver or Tool?

Definitely driver. These effects are just not implemented yet.

In general: Could you maybe comment on the general feeling of the FFB? Is it comparable to what you get in windows?

Gonzo2O28, 2021-01-25:
I am asking, because i wonder if the HZ might be limited in any way under linux, or if its application related?
I will get the ids asap. And i will do some comparison in assetto corsa which runs smooth to me under Linux, rF2 is really slow, fps wise.

Gonzo2O28, 2021-01-25:
https://fanatec.com/eu-de/lenkraeder/clubsport-steering-wheel-formula-v2
Fanatec Formula V2
cat /sys/module/hid_fanatec/drivers/hid\:ftec_csl_elite/*/wheel_id
0x280a

https://fanatec.com/eu-de/lenkraeder/podium-lenkrad-porsche-911-gt3-r-wildleder
cat /sys/module/hid_fanatec/drivers/hid\:ftec_csl_elite/*/wheel_id
0x050c

berarma, 2021-01-25:
There's a limit in the speed of the device processing USB commands. That limit seems to be pretty high at 1000Hz for Fanatec wheels, it should be the same in Windows or Linux, or at least it should be high enough so it's no problem. Logitech wheels work well in rF2 with a maximum speed of 500Hz.

@gotzl Since you're using the timing code from `new-lg4ff` it should work the same in Fanatec wheels unless you change something in the way it works. When the game makes a request to the driver, the driver just registers the request with a very short lock, it's the fastest that we can get and it doesn't depend on the device throughput.

Then the timer code processes pending requests and send them to the device also very fast. In `new-lg4ff` I discard older requests that have been superseded by newest ones. I think you're doing the same. I had a check to avoid filling the kernel USB package buffer and creating FFB lag, I've seen you've commented this out. It shouldn't matter for rF2, it was needed for ETS2 and ACC.

Increasing the `DEFAULT_TIMER_PERIOD` constant should decrease the frequency at which commands are sent, and also decrease the number of commands sent. But the current value works well for us and should for you.

After writing all this chatter I've reminded two things that could be an issue.

The latest Nvidia driver version 460.32.03, and maybe some previous versions, make rF2 run very slowly. Versions up to 450.80.02 work perfectly.

The setting `Use thread` in `UserData/player/Controller.JSON` can be set to `true` so that rF2 processes FFB in a separate thread.

Gonzo2O28, 2021-01-25:
use thread = true in the controller json was made new default some months ago.
driver should have lowest latency possible for best ffb experience.

thanks for info about the nvidia drivers, i have to check out if older ones work for my rtx3090

berarma, 2021-01-25:
> thanks for info about the nvidia drivers, i have to check out if older ones work for my rtx3090

Try rF2 with the Fanatec driver unloaded, maybe just disconnect the wheel. You'll probably find it runs slow even then. My friends have the same problem. It seems there's some performance regression or a DXVK issue uncovered by the newest Nvidia driver.

Gonzo2O28, 2021-01-25:
i know it is not the fanatec driver, it was slow before. i had situations with my previous 2080 ti where it was fast and where it was slow...

Gonzo2O28, 2021-01-25:
I need at lease 455.x for my RTX 3090 card.
So downgrading is impossible. On arch this is not possible for me either, without having the old packages in the cache.

berarma, 2021-01-25:
> i know it is not the fanatec driver, it was slow before. i had situations with my previous 2080 ti where it was fast and where it was slow...

Then we should move that conversation here: https://github.com/ValveSoftware/Proton/issues/245

Gonzo2O28, 2021-01-26:
I just created this discord server.
not setup anything yet, but smalltalk might work better there.

https://discord.gg/K2dJKtNb

Gonzo2O28, 2021-01-27:
@gotzl do you join discord?
Oversteer made good progress so far with the basics.

gotzl, 2021-02-01:
Hi, yes, I'll join. I'm just quite busy lately .... Also, the link seems to be expired?

Gonzo2O28, 2021-02-01:
https://discord.gg/gxq2X2V8

meowthink, 2021-11-26:
Update:
Good job! I can confirm this works on DD2, with only a small patch on id

``
diff --git a/fanatec.rules b/fanatec.rules
index 4fe1e62..0e2959e 100644
--- a/fanatec.rules
+++ b/fanatec.rules
@@ -5,7 +5,7 @@ GOTO="fanatec_end"
 LABEL="fanatec_vendor"

 # remove deadzone/fuzz
-ATTRS{idProduct}=="0001|0004|0005|0006|0020|6204|0E03",  ATTRS{idVendor}=="0eb7", RUN{program}+="/usr/bin/evdev-joystick --evdev $devnode --deadzone 0 --fuzz 0"
+ATTRS{idProduct}=="0001|0004|0005|0006|0007|0020|6204|0E03",  ATTRS{idVendor}=="0eb7", RUN{program}+="/usr/bin/evdev-joystick --evdev $devnode --deadzone 0 --fuzz 0"
 # set access rights
 SUBSYSTEM=="usb|input|hid", GROUP:="plugdev"

diff --git a/hid-ftec.c b/hid-ftec.c
index e6b5d6c..7f14a27 100644
--- a/hid-ftec.c
+++ b/hid-ftec.c
@@ -293,6 +293,7 @@ static const struct hid_device_id devices[] = {
        { HID_USB_DEVICE(FANATEC_VENDOR_ID, CSL_ELITE_PS4_WHEELBASE_DEVICE_ID), .driver_data = FTEC_FF | FTEC_LEDS},
        { HID_USB_DEVICE(FANATEC_VENDOR_ID, CSL_ELITE_PEDALS_DEVICE_ID), .driver_data = FTEC_PEDALS },
        { HID_USB_DEVICE(FANATEC_VENDOR_ID, PODIUM_WHEELBASE_DD1_DEVICE_ID), .driver_data = FTEC_FF | FTEC_LEDS },
+       { HID_USB_DEVICE(FANATEC_VENDOR_ID, PODIUM_WHEELBASE_DD2_DEVICE_ID), .driver_data = FTEC_FF | FTEC_LEDS },
        { HID_USB_DEVICE(FANATEC_VENDOR_ID, CSL_DD_WHEELBASE_DEVICE_ID), .driver_data = FTEC_FF | FTEC_LEDS },
     { }
 };
diff --git a/hid-ftec.h b/hid-ftec.h
index 17c8e82..0e69ca4 100644
--- a/hid-ftec.h
+++ b/hid-ftec.h
@@ -7,6 +7,7 @@
 #define CLUBSPORT_V25_WHEELBASE_DEVICE_ID 0x0004
 #define CLUBSPORT_PEDALS_V3_DEVICE_ID 0x183b
 #define PODIUM_WHEELBASE_DD1_DEVICE_ID 0x0006
+#define PODIUM_WHEELBASE_DD2_DEVICE_ID 0x0007
 #define CSL_ELITE_WHEELBASE_DEVICE_ID 0x0E03
 #define CSL_ELITE_PS4_WHEELBASE_DEVICE_ID 0x0005
 #define CSL_ELITE_PEDALS_DEVICE_ID 0x6204
``

Though for unknown reason, rotation direction by scripts from ffbtools are inverted (left acts right)

gotzl, 2021-11-26:
Thanks for the patch and testing, I've added it to the souces.
How is the rotation direction in games? Inverted as well?

meowthink, 2021-11-27:
> Thanks for the patch and testing, I've added it to the sources. How is the rotation direction in games? Inverted as well?

Not tested. Unfortunately I was making a simulator, not gaming.
btw codes in ftecff.c like 
``
        parameters[0].level = (long)parameters[0].level * gain / 0xffff;
``
which will abstract to type `(long) * (unsigned int) / (int)`, should result `unsigned int` according to C99[1(https://stackoverflow.com/a/50632). This behaviors differently on different platforms.
As a result, I've patched like this in additional
```
diff --git a/hid-ftecff.c b/hid-ftecff.c
index 7d6d7f7..ccc1815 100644
--- a/hid-ftecff.c
+++ b/hid-ftecff.c
@@ -913,7 +913,7 @@ static __always_inline int ftecff_timer(struct ftec_drv_data *drv_data)
        unsigned long jiffies_now = jiffies;
        unsigned long now = JIFFIES2MS(jiffies_now);
        unsigned long flags;
-       unsigned int gain;
+       int gain;
        int current_period;
        int count;
        int effect_id;
```
I don't think this effects the direction - the result was traced `ftecff_timer` -> `ftecff_update_slot`, maybe some weird things like `fix_values`?

gotzl, 2021-11-30:
You are right, when playing `tests/constant.ffb` with ffbtest, the direction is opposite for me as well. I'm not sure why this is, I can only say that what's implemented does the correct things in games ^^

[2 more replies]

closed 2022-02-05 by gotzl