gotzl/hid-fanatecff #11: Deadzone Issue CSL Elite PS4

Acreen, 2021-02-13:
Today I tried version 170326d at Kubuntu 20.10 and followed the install instructions

After following your instructions the device was recognized. 

But it has an unusable high deadzone of like 75° to each side. I heard that there was a fix going arround in fanatec forum which seemed to be implemented already.

I tested the device using the KDE Gamectonroller Tool.

BTW: Are there any loadcell adjustment file seems to be missing here. 
What is the accepted value range for "BLI DPR DRI FEI FF FOR SEN SHO SPR"   , same as in fanatec windows?
This is my last needed step to screw windows though it is an important one since I love Assetto corsa and the CSL.

Please have mercy since I'm very new to linux for home usage!

Great work btw!

mkoreneff, 2021-02-13:
I wrote a post on reddit about fixing the deadzone

https://www.reddit.com/r/simracing/comments/jtldnb/setting_up_a_fanatec_csl_elite_ps4_wheelbase_in/?utm_medium=android_app&utm_source=share

You can use jstest-gtk auto-calibrate to remove the deadzone

Acreen, 2021-02-13:
Just tried it, it works. Didn't find your reddit post, sry!

mkoreneff, 2021-02-13:
Glad to hear it works for you!

Thanks,

Mike

closed 2021-02-13 by Acreen