gotzl/hid-fanatecff #13: Custom Force Feedback with Clubsport Wheelbase V2.5 from Ubuntu 18

andschi, 2021-04-16:
Hi, 

I want to achieve custom force feedback with my Fanatec Clubsport Wheelbase V2.5 from Ubuntu 18. Something similar as is done in the "Joystick" package's `ffcfstress` script. 
https://github.com/flosse/linuxconsole/blob/master/utils/ffcfstress.c 

I tried running this with my Fanatec wheel plugged in, but get 
> ERROR: device (or driver) has no constant force feedback support

Running `fftest` also gives me unsupported functions only: 
```
Force feedback test program.
HOLD FIRMLY YOUR WHEEL OR JOYSTICK TO PREVENT DAMAGES

Device /dev/input/event19 opened
Features:
  * Absolute axes: X, Y, Z, RX, RY, RZ, Throttle, Rudder, Hat 0 X, Hat 0 Y, 
    [FF 00 03 00 00 00 00 00 ]
  * Relative axes: 
    [00 00 ]
  * Force feedback effects types: 
    Force feedback periodic effects: 
    [00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ]
  * Number of simultaneous effects: 0

Uploading effect #0 (Periodic sinusoidal) ... Error:: Function not implemented
Uploading effect #1 (Constant) ... Error: Function not implemented
Uploading effect #2 (Spring) ... Error: Function not implemented
Uploading effect #3 (Damper) ... Error: Function not implemented
Uploading effect #4 (Strong rumble, with heavy motor) ... Error: Function not implemented
Uploading effect #5 (Weak rumble, with light motor) ... Error: Function not implemented
Enter effect number, -1 to exit
``` 

I have the feeling the driver implemented in this repo could achieve this. However, I am not sure how to use it. Do ( or @mcoffin ) you have an advice for me on how to use it from a C Code application? 
 
Thanks!

gotzl, 2021-04-21:
Hi,

the fftest should work. If you're using the `master` branch, only the 'Constant' effect will produce s.t. With the `next` branch, some more effects are supported.

I reckon that your wheel was not properly detected by the driver. Can you make sure that it is in the `PC` mode and not in the 'Compatibility` mode? You can check with `dmesg` or `lsusb`, your wheel should show up with VID:PID 0EB7:0004.

andschi, 2021-04-28:
Hi, 
many thanks for your reply! I tried checking out both branches (`master` and `next`), ran `make && sudo make install`, replugged the wheel. However, both, `fftest` and `ffcfstress`still give me
```
➜  ~ fftest /dev/input/event19
Force feedback test program.
HOLD FIRMLY YOUR WHEEL OR JOYSTICK TO PREVENT DAMAGES

Device /dev/input/event19 opened
Features:
  * Absolute axes: X, Y, Z, RX, RY, RZ, Throttle, Rudder, Hat 0 X, Hat 0 Y, 
    [FF 00 03 00 00 00 00 00 ]
  * Relative axes: 
    [00 00 ]
  * Force feedback effects types: 
    Force feedback periodic effects: 
    [00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ]
  * Number of simultaneous effects: 0

Uploading effect #0 (Periodic sinusoidal) ... Error:: Function not implemented
Uploading effect #1 (Constant) ... Error: Function not implemented
Uploading effect #2 (Spring) ... Error: Function not implemented
Uploading effect #3 (Damper) ... Error: Function not implemented
Uploading effect #4 (Strong rumble, with heavy motor) ... Error: Function not implemented
Uploading effect #5 (Weak rumble, with light motor) ... Error: Function not implemented
Enter effect number, -1 to exit
```
and 
```
➜  ~ ffcfstress -d /dev/input/event19
ERROR: device (or driver) has no constant force feedback support [ffcfstress.c:194]  
```  

I verified the wheel is in PC mode: `lsusb` gives me
```
Bus 001 Device 007: ID 0eb7:0004 Endor AG  
```  

`dmesg` gives me.
```
[  172.684735] usb 1-2: USB disconnect, device number 6
[  188.270660] usb 1-2: new full-speed USB device number 7 using xhci_hcd
[  188.420768] usb 1-2: New USB device found, idVendor=0eb7, idProduct=0004, bcdDevice= 6.28
[  188.420774] usb 1-2: New USB device strings: Mfr=1, Product=3, SerialNumber=0
[  188.420777] usb 1-2: Product: FANATEC ClubSport Wheel Base V2.5
[  188.420780] usb 1-2: Manufacturer: Fanatec
[  188.424517] hid_map_usage: 34 callbacks suppressed
[  188.424521] Fanatec FANATEC ClubSport Wheel Base V2.5: Invalid code 768 type 1
[  188.424527] Fanatec FANATEC ClubSport Wheel Base V2.5: Invalid code 769 type 1
[  188.424529] Fanatec FANATEC ClubSport Wheel Base V2.5: Invalid code 770 type 1
[  188.424533] Fanatec FANATEC ClubSport Wheel Base V2.5: Invalid code 771 type 1
[  188.424536] Fanatec FANATEC ClubSport Wheel Base V2.5: Invalid code 772 type 1
[  188.424539] Fanatec FANATEC ClubSport Wheel Base V2.5: Invalid code 773 type 1
[  188.424541] Fanatec FANATEC ClubSport Wheel Base V2.5: Invalid code 774 type 1
[  188.424544] Fanatec FANATEC ClubSport Wheel Base V2.5: Invalid code 775 type 1
[  188.424548] Fanatec FANATEC ClubSport Wheel Base V2.5: Invalid code 776 type 1
[  188.424551] Fanatec FANATEC ClubSport Wheel Base V2.5: Invalid code 777 type 1
[  188.424784] input: Fanatec FANATEC ClubSport Wheel Base V2.5 as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/0003:0EB7:0004.0004/input/input26
[  188.425191] input: Fanatec FANATEC ClubSport Wheel Base V2.5 as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/0003:0EB7:0004.0004/input/input27
[  188.425569] hid-generic 0003:0EB7:0004.0004: input,hiddev0,hidraw1: USB HID v1.11 Joystick [Fanatec FANATEC ClubSport Wheel Base V2.5] on usb-0000:00:14.0-2/input0  
```  

I am not sure if the "suppressed callbacks" and "invalid codes" are the issue?

gotzl, 2021-04-28:
Hmm... that's weird. The issue is this line `[  188.425569] hid-generic ...`. It picks the generic driver instead of `hid-fanatec` (i don't know what the `Invalid code` messages mean, but I get them as well and they seem no harm).

For a quick test, could you insert the module by hand `sudo insmod hid-fanatec.ko` and then replug the wheel?

andschi, 2021-04-28:
I tried inserting the module by hand using `sudo insmod hid-fanatec.ko`. At first, I got the error that `insmod: ERROR: could not insert module hid-fanatec.ko: Operation not permitted`.  
Apparently, the kernel refuses to load the kernel module, as it is not signed. (https://www.silabs.com/community/interface/forum.topic.html/cp210x_insmod_oper-kl7I )  
Following "the easiest way", described here: https://askubuntu.com/a/762255, I disabled Secure Boot in the UEFI (BIOS) settings. After this, I could `sudo insmod hid-fanatec.ko` the Kernel. Running `dmesg` now, I get
```
[   14.322567] Started bpfilter
[   14.640868] Initializing XFRM netlink socket
[   14.738983] Bluetooth: RFCOMM TTY layer initialized
[   14.738987] Bluetooth: RFCOMM socket layer initialized
[   14.738990] Bluetooth: RFCOMM ver 1.11
[   15.647564] rfkill: input handler disabled
[   28.571437] hid_fanatec: loading out-of-tree module taints kernel.
[   28.571534] hid_fanatec: module verification failed: signature and/or required key missing - tainting kernel
[   32.922871] usb 1-2: USB disconnect, device number 2
[   38.061057] usb 1-2: new full-speed USB device number 5 using xhci_hcd
[   38.210448] usb 1-2: New USB device found, idVendor=0eb7, idProduct=0004, bcdDevice= 6.28
[   38.210454] usb 1-2: New USB device strings: Mfr=1, Product=3, SerialNumber=0
[   38.210458] usb 1-2: Product: FANATEC ClubSport Wheel Base V2.5
[   38.210461] usb 1-2: Manufacturer: Fanatec
[   38.214015] hid_map_usage: 34 callbacks suppressed
[   38.214018] Fanatec FANATEC ClubSport Wheel Base V2.5: Invalid code 768 type 1
[   38.214021] Fanatec FANATEC ClubSport Wheel Base V2.5: Invalid code 769 type 1
[   38.214022] Fanatec FANATEC ClubSport Wheel Base V2.5: Invalid code 770 type 1
[   38.214027] Fanatec FANATEC ClubSport Wheel Base V2.5: Invalid code 771 type 1
[   38.214029] Fanatec FANATEC ClubSport Wheel Base V2.5: Invalid code 772 type 1
[   38.214031] Fanatec FANATEC ClubSport Wheel Base V2.5: Invalid code 773 type 1
[   38.214033] Fanatec FANATEC ClubSport Wheel Base V2.5: Invalid code 774 type 1
[   38.214034] Fanatec FANATEC ClubSport Wheel Base V2.5: Invalid code 775 type 1
[   38.214036] Fanatec FANATEC ClubSport Wheel Base V2.5: Invalid code 776 type 1
[   38.214038] Fanatec FANATEC ClubSport Wheel Base V2.5: Invalid code 777 type 1
[   38.214192] input: Fanatec FANATEC ClubSport Wheel Base V2.5 as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/0003:0EB7:0004.0003/input/input24
[   38.214637] input: Fanatec FANATEC ClubSport Wheel Base V2.5 as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/0003:0EB7:0004.0003/input/input25
[   38.215284] hid-generic 0003:0EB7:0004.0003: input,hiddev0,hidraw0: USB HID v1.11 Joystick [Fanatec FANATEC ClubSport Wheel Base V2.5] on usb-0000:00:14.0-2/input0  
```  
I guess, the problem remains that I need the driver to be signed: `hid_fanatec: module verification failed: signature and/or required key missing - tainting kernel`.

mkoreneff, 2021-04-28:
I self signed the module, I think this link
<https://ubuntu.com/blog/how-to-sign-things-for-secure-boot>  is what I
followed originally.

Something like this worked for me.

KVERSION := `uname -r`
/usr/src/linux-headers-${KVERSION}/scripts/sign-file sha256
myselfsignedmok.priv myselfsignedmok.der
/lib/modules/${KVERSION}/kernel/drivers/misc/hid-fanatec.ko

Thanks,

Mike
www.ironmikemusing.com

andschi, 2021-04-28:
Hi Mike,  
I am trying to follow the exact same instructions. I got to the point that I get the `signature appended`, when running the `hexdump -Cv hid-fanatec.ko | tail -n 5`. However, I still have the same `tainting kernel` issue.  
Did you place those four lines you wrote in the makefile of the driver?

mkoreneff, 2021-04-28:
It was a while back, from memory the steps I took were:
Generate a key with `openssl` here my example is named  myselfsignedmok.priv and myselfsignedmok.der
Enrol the key using `sudo mokutil --import  myselfsignedmok.der`
`make install`
Sign the hid-fanatec.ko module using `/usr/src/linux-headers-${KVERSION}/scripts/sign-file sha256 myselfsignedmok.priv yselfsignedmok.der /lib/modules/${KVERSION}/kernel/drivers/misc/hid-fanatec.ko`
Then I can use the module

I suspect you haven't loaded the self signed key into your uefi with `sudo mokutil --import`
I recall there being 2 steps to enrol the key. 
After running `sodo mokutil --import` you need to reboot and enter/enrol the key by entering the password used when generating the key.

andschi, 2021-04-28:
Thanks for your quick reply! I followed the steps including the reboot and key enrolling (part with the blue screen). After this, I built the driver using `make` and signed it using `kmodsign sha512 MOK.priv MOK.der hid-fanatec.ko`. Running `sudo make install` and/or `sudo insmod hid-fanatec.ko` places the files in `/lib/modules/${KVERSION}/kernel/drivers/hid`. When I replug the wheel now, and run `dmesg`, I get
```
[   14.682999] Started bpfilter
[...]
[   64.912382] hid_fanatec: loading out-of-tree module taints kernel.
[...]
[  100.758476] usb 1-2: USB disconnect, device number 2
[  106.342305] usb 1-2: new full-speed USB device number 5 using xhci_hcd
[  106.491956] usb 1-2: New USB device found, idVendor=0eb7, idProduct=0004, bcdDevice= 6.28
[  106.491962] usb 1-2: New USB device strings: Mfr=1, Product=3, SerialNumber=0
[  106.491965] usb 1-2: Product: FANATEC ClubSport Wheel Base V2.5
[  106.491968] usb 1-2: Manufacturer: Fanatec
[  106.495776] hid_map_usage: 34 callbacks suppressed
[  106.495779] Fanatec FANATEC ClubSport Wheel Base V2.5: Invalid code 768 type 1
[  106.495781] Fanatec FANATEC ClubSport Wheel Base V2.5: Invalid code 769 type 1
[  106.495783] Fanatec FANATEC ClubSport Wheel Base V2.5: Invalid code 770 type 1
[  106.495785] Fanatec FANATEC ClubSport Wheel Base V2.5: Invalid code 771 type 1
[  106.495786] Fanatec FANATEC ClubSport Wheel Base V2.5: Invalid code 772 type 1
[  106.495788] Fanatec FANATEC ClubSport Wheel Base V2.5: Invalid code 773 type 1
[  106.495790] Fanatec FANATEC ClubSport Wheel Base V2.5: Invalid code 774 type 1
[  106.495792] Fanatec FANATEC ClubSport Wheel Base V2.5: Invalid code 775 type 1
[  106.495793] Fanatec FANATEC ClubSport Wheel Base V2.5: Invalid code 776 type 1
[  106.495795] Fanatec FANATEC ClubSport Wheel Base V2.5: Invalid code 777 type 1
[  106.495953] input: Fanatec FANATEC ClubSport Wheel Base V2.5 as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/0003:0EB7:0004.0003/input/input24
[  106.496330] input: Fanatec FANATEC ClubSport Wheel Base V2.5 as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/0003:0EB7:0004.0003/input/input25
[  106.496733] hid-generic 0003:0EB7:0004.0003: input,hiddev0,hidraw0: USB HID v1.11 Joystick [Fanatec FANATEC ClubSport Wheel Base V2.5] on usb-0000:00:14.0-2/input0
[  280.379497] usb 1-2: USB disconnect, device number 5
[  287.148132] usb 1-2: new full-speed USB device number 6 using xhci_hcd
[  287.297764] usb 1-2: New USB device found, idVendor=0eb7, idProduct=0004, bcdDevice= 6.28
[  287.297770] usb 1-2: New USB device strings: Mfr=1, Product=3, SerialNumber=0
[  287.297773] usb 1-2: Product: FANATEC ClubSport Wheel Base V2.5
[  287.297776] usb 1-2: Manufacturer: Fanatec
[  287.301311] hid_map_usage: 34 callbacks suppressed
[  287.301314] Fanatec FANATEC ClubSport Wheel Base V2.5: Invalid code 768 type 1
[  287.301317] Fanatec FANATEC ClubSport Wheel Base V2.5: Invalid code 769 type 1
[  287.301319] Fanatec FANATEC ClubSport Wheel Base V2.5: Invalid code 770 type 1
[  287.301320] Fanatec FANATEC ClubSport Wheel Base V2.5: Invalid code 771 type 1
[  287.301322] Fanatec FANATEC ClubSport Wheel Base V2.5: Invalid code 772 type 1
[  287.301324] Fanatec FANATEC ClubSport Wheel Base V2.5: Invalid code 773 type 1
[  287.301325] Fanatec FANATEC ClubSport Wheel Base V2.5: Invalid code 774 type 1
[  287.301327] Fanatec FANATEC ClubSport Wheel Base V2.5: Invalid code 775 type 1
[  287.301329] Fanatec FANATEC ClubSport Wheel Base V2.5: Invalid code 776 type 1
[  287.301330] Fanatec FANATEC ClubSport Wheel Base V2.5: Invalid code 777 type 1
[  287.301495] input: Fanatec FANATEC ClubSport Wheel Base V2.5 as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/0003:0EB7:0004.0004/input/input26
[  287.301866] input: Fanatec FANATEC ClubSport Wheel Base V2.5 as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/0003:0EB7:0004.0004/input/input27
[  287.302586] hid-generic 0003:0EB7:0004.0004: input,hiddev0,hidraw0: USB HID v1.11 Joystick [Fanatec FANATEC ClubSport Wheel Base V2.5] on usb-0000:00:14.0-2/input0  
```  
So it still uses the `hid-generic`, however I do no longer the `hid_fanatec: module verification failed: signature and/or required key missing - tainting kernel` message, only that `hid_fanatec: loading out-of-tree module taints kernel.`

gotzl, 2021-04-28:
Damn, now that's my fault... Should've looked into the code right away... Apparently, I lost the CSv25 device id at some point :( Sorry for that, I'll put it in right now.

andschi, 2021-04-30:
okay, this definitely was the missing piece! (:  
I edited the issue description and added the steps that worked for me.  

Many thanks for your support! @gotzl @mkoreneff

closed 2021-11-26 by gotzl