gotzl/hid-fanatecff #14: Fanatec Porsche GT2 and other older devices

MrFuNK1, 2021-07-09:
Hi there,
I do own one of the old neat GT2 wheels and am curious whether the driver does work with it or what info/data would be required to get being compatibility. 
Would appreciate if anyone could get me started or point out which data I can collect/provide. 
Thx!

gotzl, 2021-07-11:
Hi,
is [this](https://fanatec.com/eu-de/racing-wheels-wheel-bases/racing-wheels/porsche-911-gt2-wheel) the product you own?

To support this wheel, we need to reverse engineer the USB communicate protocol. It might be the same as with the other, newer wheels, but you never know ^^ For this, you've to capture the USB traffic in windows while using the wheel. I'd recommend to use wireshark for capturing and an FFB test tool to generate 'known' input (for example [WheelCheck](https://www.dropbox.com/s/3zxf5yd1bvujyi5/WheelCheck_1_72.zip?dl=0)).
If the wheel software allows to change some wheel settings, like range or FFB strength, then change these as well while recording the USB traffic. Make sure to do the things in an 'orderly' fashion, so that your action can be matched with the recorded USB traffic ...
For starters, it would be enough to see the commands send for constant FFB effect. This is already enough for most games.

MrFuNK1, 2021-07-12:
Thanks and yes, that's the wheel I do have!

No problem collecting that data but unfortunately will not be able to do soon (other projects come first :rofl: ). Anyway, as soon as I have the data collected, I will let you know!