gotzl/hid-fanatecff #16: Support for CSR Elite

kozr, 2021-11-21:
Hello. I was trying to get CSR Elite working, but to no avail.

vendor:product_id
`0EB7:0011`

Here is what I've tried.
https://github.com/kozr/hid-fanatecff/commit/1b794c4d5b53daf3448e0512ef73d41bf259412f
`echo 500 | sudo tee /sys/module/hid_fanatec/drivers/hid\:ftec_csl_elite/0003\:0EB7\:0011.0013/range` does nothing.

To be honest I'm not very familiar with low level stuff but it seems like we ran into something? `implement() called with too large value` from `dmesg -w`
```
[ 9999.153262] input: HID 0eb7:0011 as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:0b:00.1/usb3/3-3/3-3:1.0/0003:0EB7:0011.0013/input/input44
[ 9999.213220] ftec_csl_elite 0003:0EB7:0011.0013: input,hidraw10: USB HID v1.11 Joystick [HID 0eb7:0011] on usb-0000:0b:00.1-3/input0
[ 9999.213227] /home/nicholaswong/Documents/wheel/hid-fanatecff/hid-ftec.c:  ... 0 1 1 56
               0 255 0 255

[ 9999.213231] /home/nicholaswong/Documents/wheel/hid-fanatecff/hid-ftecff.c:  ... setting FF bits
[ 9999.213233] /home/nicholaswong/Documents/wheel/hid-fanatecff/hid-ftecff.c: setting range to 1080
[ 9999.213236] ftec_csl_elite 0003:0EB7:0011.0013: implement() called with too large value -11 (n: 8)! (systemd-udevd)
[ 9999.213282] ------------[ cut here ]------------
[ 9999.213283] WARNING: CPU: 18 PID: 59158 at drivers/hid/hid-core.c:1432 implement.cold+0x5e/0x68 [hid]
[ 9999.213296] Modules linked in: hid_fanatec(OE+) veth xt_nat xt_tcpudp xt_conntrack nft_chain_nat xt_MASQUERADE nf_nat nf_conntrack_netlink nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 xfrm_user xfrm_algo nft_counter xt_addrtype nft_compat nf_tables nfnetlink br_netfilter bridge stp llc rfcomm ccm cmac algif_hash algif_skcipher af_alg overlay bnep nls_iso8859_1 intel_rapl_msr intel_rapl_common snd_hda_codec_realtek edac_mce_amd snd_hda_codec_generic ledtrig_audio snd_hda_codec_hdmi snd_hda_intel snd_intel_dspcfg soundwire_intel soundwire_generic_allocation soundwire_cadence snd_hda_codec snd_hda_core snd_hwdep soundwire_bus snd_soc_core iwlmvm snd_compress ac97_bus snd_pcm_dmaengine mac80211 snd_pcm snd_seq_midi snd_seq_midi_event libarc4 btusb snd_rawmidi btrtl btbcm iwlwifi input_leds btintel joydev kvm snd_seq bluetooth snd_seq_device eeepc_wmi snd_timer asus_wmi ecdh_generic sparse_keymap ecc rapl wmi_bmof efi_pstore snd video cfg80211 ccp k10temp soundcore mac_hid sch_fq_codel msr
[ 9999.213355]  parport_pc ppdev lp parport ip_tables x_tables autofs4 dm_crypt raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor raid6_pq libcrc32c raid1 raid0 multipath linear system76_io(OE) system76_acpi(OE) hid_generic usbhid hid amdgpu iommu_v2 gpu_sched i2c_algo_bit drm_ttm_helper ttm drm_kms_helper syscopyarea crct10dif_pclmul sysfillrect sysimgblt crc32_pclmul fb_sys_fops cec ghash_clmulni_intel rc_core aesni_intel drm crypto_simd cryptd glue_helper ahci xhci_pci libahci i2c_piix4 r8169 nvme xhci_pci_renesas realtek nvme_core wmi [last unloaded: hid_fanatec]
[ 9999.213395] CPU: 18 PID: 59158 Comm: systemd-udevd Tainted: G        W  OE     5.11.0-7620-generic #21~1626191760~21.04~55de9c3-Ubuntu
[ 9999.213399] Hardware name: System manufacturer System Product Name/TUF GAMING X570-PLUS (WI-FI), BIOS 3801 04/07/2021
[ 9999.213401] RIP: 0010:implement.cold+0x5e/0x68 [hid]
[ 9999.213408] Code: bf d0 18 00 00 89 d9 48 c7 c6 80 95 33 c0 65 48 8b 14 25 c0 7b 01 00 4c 8d 8a e8 0a 00 00 48 c7 c2 88 f4 32 c0 e8 87 20 4e e3 <0f> 0b 44 21 fb e9 13 49 ff ff 48 8b bb f0 b0 01 00 48 c7 c6 b8 95
[ 9999.213411] RSP: 0018:ffffb72d460938e0 EFLAGS: 00010046
[ 9999.213413] RAX: 0000000000000000 RBX: 00000000fffffff5 RCX: 0000000000000000
[ 9999.213415] RDX: 00000000ffffffd8 RSI: 0000000000000027 RDI: ffff918bcee98ac0
[ 9999.213417] RBP: ffffb72d46093908 R08: 0000000000000000 R09: ffffb72d460935f0
[ 9999.213419] R10: ffffb72d460935e8 R11: ffffffffa4953568 R12: ffff9184d7d07590
[ 9999.213420] R13: 0000000000000008 R14: 0000000000000000 R15: 00000000000000ff
[ 9999.213422] FS:  00007f6de8be28c0(0000) GS:ffff918bcee80000(0000) knlGS:0000000000000000
[ 9999.213424] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 9999.213426] CR2: 0000278440517430 CR3: 0000000295f5e000 CR4: 0000000000350ee0
[ 9999.213428] Call Trace:
[ 9999.213431]  hid_output_report+0xef/0x160 [hid]
[ 9999.213439]  __usbhid_submit_report+0x263/0x340 [usbhid]
[ 9999.213445]  usbhid_request+0x47/0x80 [usbhid]
[ 9999.213450]  ftec_set_range+0xc8/0x1e0 [hid_fanatec]
[ 9999.213456]  ftecff_init+0xf1/0x580 [hid_fanatec]
[ 9999.213460]  ? printk+0x58/0x6f
[ 9999.213467]  ftec_probe+0x1a8/0x1f0 [hid_fanatec]
[ 9999.213471]  hid_device_probe+0x10e/0x170 [hid]
[ 9999.213479]  really_probe+0xff/0x460
[ 9999.213483]  driver_probe_device+0xe9/0x160
[ 9999.213486]  __device_attach_driver+0x71/0xd0
[ 9999.213489]  ? driver_allows_async_probing+0x50/0x50
[ 9999.213492]  bus_for_each_drv+0x84/0xd0
[ 9999.213494]  __device_attach+0xde/0x1e0
[ 9999.213497]  ? __hid_register_driver+0x80/0x80 [hid]
[ 9999.213503]  device_attach+0x10/0x20
[ 9999.213506]  device_reprobe+0x4a/0x90
[ 9999.213509]  __hid_bus_reprobe_drivers+0x6e/0x70 [hid]
[ 9999.213515]  bus_for_each_dev+0x7e/0xc0
[ 9999.213517]  ? hid_destroy_device+0x60/0x60 [hid]
[ 9999.213524]  __hid_bus_driver_added+0x2c/0x40 [hid]
[ 9999.213530]  bus_for_each_drv+0x84/0xd0
[ 9999.213533]  __hid_register_driver+0x77/0x80 [hid]
[ 9999.213539]  ? 0xffffffffc0fa2000
[ 9999.213541]  ftec_csl_elite_init+0x23/0x1000 [hid_fanatec]
[ 9999.213546]  do_one_initcall+0x48/0x1d0
[ 9999.213550]  ? kmem_cache_alloc_trace+0xf6/0x200
[ 9999.213555]  ? do_init_module+0x28/0x290
[ 9999.213559]  do_init_module+0x62/0x290
[ 9999.213561]  load_module+0x6fd/0x780
[ 9999.213564]  __do_sys_finit_module+0xc2/0x120
[ 9999.213567]  __x64_sys_finit_module+0x1a/0x20
[ 9999.213570]  do_syscall_64+0x38/0x90
[ 9999.213573]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[ 9999.213577] RIP: 0033:0x7f6de909af6d
[ 9999.213579] Code: 28 0d 00 0f 05 eb a9 66 0f 1f 44 00 00 f3 0f 1e fa 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d cb de 0c 00 f7 d8 64 89 01 48
[ 9999.213582] RSP: 002b:00007ffd09a06108 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
[ 9999.213584] RAX: ffffffffffffffda RBX: 00005614e9d2a160 RCX: 00007f6de909af6d
[ 9999.213586] RDX: 0000000000000000 RSI: 00007f6de9240e2d RDI: 0000000000000006
[ 9999.213588] RBP: 0000000000020000 R08: 0000000000000000 R09: 00005614e7d2c8dd
[ 9999.213589] R10: 0000000000000006 R11: 0000000000000246 R12: 00007f6de9240e2d
[ 9999.213591] R13: 0000000000000000 R14: 00005614e9cf16b0 R15: 00005614e9d2a160
[ 9999.213593] ---[ end trace 65a585eb499ccb81 ]---
```

kozr, 2021-11-21:
https://github.com/kozr/hid-fanatecff/blob/next/csr_elite_cycle

This is the wireshark file I was able to capture while running wheelcheck on "cycle" until the last couple of seconds (where it was disabled).

gotzl, 2021-11-21:
Hi,
so, from what I see in the debug is that the HID reports of your device are different from what the other devices have; the min/max is 0/255, whereas for my device it's -128/127. 
For my device, I had to introduce a 'hack' to get to get the correct values sent out:
https://github.com/gotzl/hid-fanatecff/blob/13a569c0ab804a346c9be138deee1705daadcb0c/hid-ftecff.c#L78
Your min/max values seem to be better and this fix might not be needed. So maybe try to just remove this code for the moment.

kozr, 2021-11-21:
I [commented out](https://github.com/kozr/hid-fanatecff/commit/328b95c208444e56292b5f60a9ab1ade8e3fd643) your implementation of `fix_values`, ran `sudo rmmod hid_fanatec` and `sudo make install`. The error does not exist anymore, but `echo 100 | sudo tee /sys/module/hid_fanatec/drivers/hid\:ftec_csl_elite/0003\:0EB7\:0011.0016/range` still doesn't seem to be moving the wheel.

In addition, I also [added csr elite to the list with reduced max range](https://github.com/kozr/hid-fanatecff/commit/5d75241fad9a7c807cd6381fd066f08df1340a78).

```
[ 3860.934558] usb 3-3: new full-speed USB device number 5 using xhci_hcd
[ 3861.107300] usb 3-3: string descriptor 0 read error: -32
[ 3861.107308] usb 3-3: New USB device found, idVendor=0eb7, idProduct=0011, bcdDevice=12.22
[ 3861.107311] usb 3-3: New USB device strings: Mfr=1, Product=3, SerialNumber=0
[ 3861.119410] input: HID 0eb7:0011 as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:0b:00.1/usb3/3-3/3-3:1.0/0003:0EB7:0011.0016/input/input44
[ 3861.119526] hid-generic 0003:0EB7:0011.0016: input,hidraw10: USB HID v1.11 Joystick [HID 0eb7:0011] on usb-0000:0b:00.1-3/input0
[ 3861.250530] /home/nicholaswong/Documents/wheel/hid-fanatecff/hid-ftec.c: ftec_probe: ifnum 0
[ 3861.250632] input: HID 0eb7:0011 as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:0b:00.1/usb3/3-3/3-3:1.0/0003:0EB7:0011.0016/input/input45
[ 3861.250712] ftec_csl_elite 0003:0EB7:0011.0016: input,hidraw10: USB HID v1.11 Joystick [HID 0eb7:0011] on usb-0000:0b:00.1-3/input0
[ 3861.250714] /home/nicholaswong/Documents/wheel/hid-fanatecff/hid-ftec.c:  ... 0 1 1 56
               0 255 0 255

[ 3861.250716] /home/nicholaswong/Documents/wheel/hid-fanatecff/hid-ftecff.c:  ... setting FF bits
[ 3861.250718] /home/nicholaswong/Documents/wheel/hid-fanatecff/hid-ftecff.c: setting range to 900
[ 3861.250815] ftec_csl_elite 0003:0EB7:0011.0016: Hires timer: period = 2 ms
```

kozr, 2021-11-21:
I also ran your other repo's `hid-fanatecff-test.py` and ran `handle.interruptWrite(0x01, payload(CMD_SIMPLE_SPRING_ENABLE+CMD_SIMPLE_SPRING_ARGS))` with `DEVICE_ID = 0x0011`. It printed the payload `bytearray(b'\x01\x11\x0b\x80\x80\x10\x00\x00')` and exited. The wheel did not move. To make this clear, if I do an interrupt write after installing the driver, I should be able to see the wheel turning by itself without any games opened right?

kozr, 2021-11-21:
It seems like I was mistaken. Changing range only limits the maximum angle I can turn the wheel? Although when I set range to be 100, it wasn't doing anything either. Since I play F1 2021, I have it set to 370 from the wheel itself. Setting the range with sysfs isn't changing the number

kozr, 2021-11-22:
Although other than that, I was able to test FFB by using https://github.com/berarma/ffbtools/blob/master/tests/sine.ffb and it seems to work.

kozr, 2021-11-22:
Awesome, yeah. Things seems to be working.

I was also able to program it myself. If you don't mind I'll make a PR for CSR Elite?

kozr, 2021-11-22:
Although I'm not exactly sure the 0 255 vs -127 128? Maybe the older models use unsigned? Anyway, I came across https://elixir.bootlin.com/linux/latest/source/drivers/hid/hid-core.c#L255 and they had to deal with it as well.

gotzl, 2021-11-25:
> The wheel did not move. To make this clear, if I do an interrupt write after installing the driver, I should be able to see the wheel turning by itself without any games opened right?

What you've found there is some very crude code to send URB requests, *w/o* needing the driver. It is just directly creating/sending the URB. It's not really supposed to be used 'in public', but I see you like to get your hands dirty ;)

> Changing range only limits the maximum angle I can turn the wheel?

Yes, in my case, if I put 90deg, it puts a very strong force so I can't turn further. It might very well be that your wheel doesn't support that, or the register address is different.
Does the windows driver offer the possibility to set the wheel range? If so, you need to capture the USB traffic while playing around with this setting. Then we can see how it works for your device.

> Awesome, yeah. Things seems to be working.

Great ;) So, you've also tried some games with success?

> Let's figure out how to deal with it before merging in #17?

PR is very welcome ;) I agree, we should figure out a proper way to deal with this ....

kozr, 2021-12-01:
> Yes, in my case, if I put 90deg, it puts a very strong force so I can't turn further. It might very well be that your wheel doesn't support that, or the register address is different.
Does the windows driver offer the possibility to set the wheel range? If so, you need to capture the USB traffic while playing around with this setting. Then we can see how it works for your device.

Turns out it is the same, but I just forgot to set SEN to off on the wheel so it wasn't reading from the driver but the wheel itself. 
[image]

> Great ;) So, you've also tried some games with success?

Not yet, I'll try with ACC soon. I used ffbtools and was able to upload and run some effects

I'll look into why I don't need fix_values in a few days...

kozr, 2022-01-19:
this is still on my todo list, just been busy recently.

PR #25 (2022-03-15): Add support for CSR Elite
Closes #16 

### Add support for CSR Elite ###
- Add device code `0x0011`
- Limit CSR Elite's max range to 900

### Continue use of `fix_values` unless it's CSR Elite ###
It seems like [hid specs page 38](https://www.usb.org/sites/default/files/documents/hid1_11.pdf) says value should be signed integer unless both `logical_minimum` and `logical_maximum` are positive (which seems to be the case for CSR Elite). That was probably why we needed `fix_values` in the first place.
[image]

I refactored and grouped `fix_value` and `hid_hw_request` to be in a single method for  [... cut]
closed 2022-03-15 by gotzl