gotzl/hid-fanatecff #18: CSL Elite Pedals

qdlmcfresh, 2021-12-21:
Hi, id like to use this with [CSL Elite Pedals](https://fanatec.com/eu-en/pedals/csl-elite-pedals). Ultimatly i need to read brake and gas values from the pedals. I have a [Clubsport-USB-Adapter](https://fanatec.com/eu-en/accessories/pedal-accessories/clubsport-usb-adapter) to connect it to my PC. 

The Pedals show up as:
```
Bus 003 Device 003: ID 0eb7:1a95 Endor AG ClubSport USB Pedal
```
So i changed the following in the driver:
https://github.com/qdlmcfresh/hid-fanatecff/commit/cf581c390bcd203556621fb11cea72a1dafc5910

After that `/sys/module/hid_fanatec` looks like this so i think the driver is getting loaded:
```
.
├── coresize
├── drivers
│   └── hid:ftec_csl_elite -> ../../../bus/hid/drivers/ftec_csl_elite
├── holders
├── initsize
├── initstate
├── notes
├── parameters
│   └── profile
├── refcnt
├── sections
│   ├── __dyndbg
│   ├── __jump_table
│   ├── __mcount_loc
│   └── __param
├── srcversion
├── taint
└── uevent
```

But i dont get a path like this `/hid:ftec_csl_elite/0003:0EB7:0005.*/` so i guess i messed something up.

gotzl, 2021-12-21:
Hi,
you don't need this driver for the pedals. Reading throttle/brake values just works with the default HID driver.
The only addition my driver brings would be if you had the brake-pedal with load-cell. Then, the driver allows to adjust the load cell sensitivity.

That said, I think the issue your seeing with my driver (`not enough values ..`) could be related to what we've seen in #16.

qdlmcfresh, 2021-12-21:
Thanks came to the same conclusion after testing with jstest. Thanks anyways :+1: :smile:

closed 2021-12-21 by qdlmcfresh