gotzl/hid-fanatecff #19: No FF with CSL elite and F1 esports wheel

DashieTM, 2022-01-03:
First, thanks for the incredible work!
Please have some mercy, I am still new to the linux environment and I have no idea where else I should post this.

I am currently running a CSL Elite base alongside the F1 esports wheel and the csl elite pedals. (Pedals work flawlessly)
This combination results in strange FF behavior. Namely I only get speed wobble, curb rumble and off track effects, the regular FF doesn't seem to work. I currently tried F1 2020 and BeamNG drive. 
The second game has some "FF" but it's behavior is extremely strange.
My guess is that the driver somehow can't access my wheel settings as my range is off as well.

I tried fiddling with the code to somehow force the driver to use 100% FF to see if that somehow does something but to no success. If you have any advice at all it would be appreciated. 
If you need more info I will gladly provide, right now I have no Idea what else to post.

gotzl, 2022-01-04:
Hi,

I just tried F1 2020 for myself and I can confirm that the FFB is not working properly for me in this game as well. I started to dig around in wine/proton to see the FFB commands that are coming from the game, but so far they don't make too much sense for me... There seems to be no centering-forces and I wonder if it works with other wheels in Linux. With F1 2019 and the DFGT, I had a similar issue but was able to work around it by using the `autocenter` feature. I'll have to investigate some more.

A don't own BeamNG, so I can't say anything about that. You could create a log of the games FFB calls by setting `WINEDEBUG=+dinput,+joystick` in protons `user_settings.py` file. This might give me an idea what's going wrong ...

DashieTM, 2022-01-04:
So sorry for the delay, I managed to record the calls, and it pretty much says everything -> "fixme" 

36637.598:0124:02a0:fixme:dinput:IDirectInputDevice2WImpl_GetForceFeedbackState (0x79a0050)->(0x2335fc40): stub!
this about 100 million times.
I doubt this will be of any use but if there is anything else that i can provide don't hesitate to ask.

[steam-1080110.log](https://github.com/gotzl/hid-fanatecff/files/7810119/steam-1080110.log)

gotzl, 2022-01-04:
I actually was more interested in the the log for BeamNG. Still, looking at the log for F1, I'm missing a bunch of entries from `dinput`.. Could you verify that the entry in your `user_settings.py` is correct? And then maybe attach the log for BeamNG? The log should be cluttered with `*:trace:dinput:fill_DataFormat Copying*` and occasionally lines looking like `*:trace:dinput:_dump_DI*`, the latter are the ones I'm after ;)

DashieTM, 2022-01-04:
Forgot to rename the file. I figured that out once I had just +dinput in there.
The whole files were too large so these are snippets, but it should do the trick.
[F12020.log](https://github.com/gotzl/hid-fanatecff/files/7810640/F12020.log)
[BeamNG.log](https://github.com/gotzl/hid-fanatecff/files/7810641/BeamNG.log)

gotzl, 2022-02-05:
I've created #22 and #23 since these issues happen with all devices.

closed 2022-02-05 by gotzl