gotzl/hid-fanatecff #21: GT DD Pro support

24788329, 2022-02-05:
Thanks for your efforts to bring Fanatec hardware support to Linux.  

Can anyone confirm if the new GT DD Pro works with this driver? It seems internally similar to the CSL DD, which was [straightforward to add](https://github.com/gotzl/hid-fanatecff/issues/15). Hope Fanatec continues to use a common USB data format across their wheel bases.  

Beyond FFB support, what can we expect on LED and digital display support for this wheel?

gotzl, 2022-02-05:
It seems that the GT DD Pro is 'just' a bundle, so I would expect that the USB-PID of the CSL DD wheel base isn't changed and the driver should just work. But yea, since I don't own one, it's just a guess ;) In any case, there is very little chance that Fanatec changed data format for this wheel base, so adding it would be straight forward as well.

In terms of LED/display, they won't work. There is some limited support in the driver for CSL Elite and F1 wheel, but whether this translates to other bases/wheel I don't know, since the CSL Elite is the only device I own.
That said, the bigger issue with LEDs/displays is that there is no standard API for games to talk to them. For example, ACC has some proprietary Fanatec library built-in. In Linux (wine), this library doesn't work ...

24788329, 2022-02-06:
An educated guess that makes sense. I will test the GT DD Pro when it arrives and share results here.

Do Fanatec pedals, shifter, and handbrake accessories also work plugged directly to a wheel base, or do they need to be connected directly to the PC via USB?

24788329, 2022-03-12:
I tried the `next` branch with the GT DD Pro. I can see the wheel in [Oversteer](https://github.com/berarma/oversteer) when it's set to PC compatibility mode (yellow), but not in PC native mode (red). FFB in ACC appears to be working. 

I also see a number of `Fanatec FANATEC Wheel: Invalid code 768 type 1` errors.

Perhaps these are because the GT DD Pro is using slightly different IDs from the CSL DD: `idVendor=0eb7, idProduct=0004, bcdDevice= 6.57`

Let me know if what other info you need. Happy to continue testing and helping.

<details>
  <summary>sudo dmesg output:</summary>

```
[   75.613009] usb 1-3: new full-speed USB device number 5 using xhci_hcd
[   75.778711] usb 1-3: New USB device found, idVendor=0eb7, idProduct=0004, bcdDevice= 6.57
[   75.778718] usb 1-3: New USB device strings: Mfr=1, Product=3, SerialNumber=0
[   75.778720] usb 1-3: Product: FANATEC Wheel
[   75.778722] usb 1-3: Manufacturer: Fanatec
[   75.805053] Fanatec FANATEC Wheel: Invalid code 768 type 1
[   75.805058] Fanatec FANATEC Wheel: Invalid code 769 type 1
[   75.805060] Fanatec FANATEC Wheel: Invalid code 770 type 1
[   75.805061] Fanatec FANATEC Wheel: Invalid code 771 type 1
[   75.805062] Fanatec FANATEC Wheel: Invalid code 772 type 1
[   75.805063] Fanatec FANATEC Wheel: Invalid code 773 type 1
[   75.805064] Fanatec FANATEC Wheel: Invalid code 774 type 1
[   75.805065] Fanatec FANATEC Wheel: Invalid code 775 type 1
[   75.805066] Fanatec FANATEC Wheel: Invalid code 776 type 1
[   75.805067] Fanatec FANATEC Wheel: Invalid code 777 type 1
[   75.805122] input: Fanatec FANATEC Wheel as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:06:00.1/usb1/1-3/1-3:1.0/0003:0EB7:0004.0008/input/input33
[   75.805282] input: Fanatec FANATEC Wheel as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:06:00.1/usb1/1-3/1-3:1.0/0003:0EB7:0004.0008/input/input34
[   75.805378] hid-generic 0003:0EB7:0004.0008: input,hiddev2,hidraw5: USB HID v1.11 Joystick [Fanatec FANATEC Wheel] on usb-0000:06:00.1-3/input0
[   75.979985] /home/userone/bin/hid-fanatecff-next/hid-ftec.c: ftec_probe: ifnum 0
[   75.980167] input: Fanatec FANATEC Wheel as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:06:00.1/usb1/1-3/1-3:1.0/0003:0EB7:0004.0008/input/input35
[   76.039805] ftec_csl_elite 0003:0EB7:0004.0008: input,hiddev2,hidraw5: USB HID v1.11 Joystick [Fanatec FANATEC Wheel] on usb-0000:06:00.1-3/input0
[   76.039810] /home/userone/bin/hid-fanatecff-next/hid-ftec.c:  ... 1 1 1 56
               -128 127 -128 127

[   76.039814] /home/userone/bin/hid-fanatecff-next/hid-ftecff.c:  ... setting FF bits
[   76.039816] /home/userone/bin/hid-fanatecff-next/hid-ftecff.c: setting range to 900
[   76.039950] ftec_csl_elite 0003:0EB7:0004.0008: Hires timer: period = 2 ms
[   88.355494] usb 1-3: USB disconnect, device number 5
[  102.468590] usb 1-3: new full-speed USB device number 6 using xhci_hcd
[  102.634441] usb 1-3: New USB device found, idVendor=0eb7, idProduct=0004, bcdDevice= 6.57
[  102.634447] usb 1-3: New USB device strings: Mfr=1, Product=3, SerialNumber=0
[  102.634449] usb 1-3: Product: FANATEC Wheel
[  102.634451] usb 1-3: Manufacturer: Fanatec
[  102.660591] /home/userone/bin/hid-fanatecff-next/hid-ftec.c: ftec_probe: ifnum 0
[  102.660710] hid_map_usage: 78 callbacks suppressed
[  102.660712] Fanatec FANATEC Wheel: Invalid code 768 type 1
[  102.660714] Fanatec FANATEC Wheel: Invalid code 769 type 1
[  102.660715] Fanatec FANATEC Wheel: Invalid code 770 type 1
[  102.660716] Fanatec FANATEC Wheel: Invalid code 771 type 1
[  102.660717] Fanatec FANATEC Wheel: Invalid code 772 type 1
[  102.660718] Fanatec FANATEC Wheel: Invalid code 773 type 1
[  102.660719] Fanatec FANATEC Wheel: Invalid code 774 type 1
[  102.660720] Fanatec FANATEC Wheel: Invalid code 775 type 1
[  102.660721] Fanatec FANATEC Wheel: Invalid code 776 type 1
[  102.660722] Fanatec FANATEC Wheel: Invalid code 777 type 1
[  102.660776] input: Fanatec FANATEC Wheel as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:06:00.1/usb1/1-3/1-3:1.0/0003:0EB7:0004.0009/input/input36
[  102.660941] ftec_csl_elite 0003:0EB7:0004.0009: input,hiddev2,hidraw5: USB HID v1.11 Joystick [Fanatec FANATEC Wheel] on usb-0000:06:00.1-3/input0
[  102.660946] /home/userone/bin/hid-fanatecff-next/hid-ftec.c:  ... 1 1 1 56
               -128 127 -128 127

[  102.660950] /home/userone/bin/hid-fanatecff-next/hid-ftecff.c:  ... setting FF bits
[  102.660952] /home/userone/bin/hid-fanatecff-next/hid-ftecff.c: setting range to 900
[  102.661079] ftec_csl_elite 0003:0EB7:0004.0009: Hires timer: period = 2 ms
[  105.820979] usb 1-3: USB disconnect, device number 6
[  106.228149] usb 1-3: new full-speed USB device number 7 using xhci_hcd
[  106.392526] usb 1-3: New USB device found, idVendor=0eb7, idProduct=0e07, bcdDevice= 6.57
[  106.392533] usb 1-3: New USB device strings: Mfr=1, Product=9, SerialNumber=0
[  106.392536] usb 1-3: Product: FANATEC WHEEL(PS5 / PS4 MODE)
[... 76 lines cut ...]
[  356.951004] usb 1-3: Manufacturer: Fanatec
[  356.979142] /home/userone/bin/hid-fanatecff-next/hid-ftec.c: ftec_probe: ifnum 0
[  356.979256] hid_map_usage: 34 callbacks suppressed
[  356.979257] Fanatec FANATEC Wheel: Invalid code 768 type 1
[  356.979259] Fanatec FANATEC Wheel: Invalid code 769 type 1
[  356.979260] Fanatec FANATEC Wheel: Invalid code 770 type 1
[  356.979261] Fanatec FANATEC Wheel: Invalid code 771 type 1
[... cut ...]
[  356.979476] ftec_csl_elite 0003:0EB7:0020.000F: input,hiddev2,hidraw5: USB HID v1.11 Joystick [Fanatec FANATEC Wheel] on usb-0000:06:00.1-3/input0
[  356.979480] /home/userone/bin/hid-fanatecff-next/hid-ftec.c:  ... 1 1 1 56
               -128 127 -128 127

[  356.979484] /home/userone/bin/hid-fanatecff-next/hid-ftecff.c:  ... setting FF bits
[  356.979486] /home/userone/bin/hid-fanatecff-next/hid-ftecff.c: setting range to 1080
[  356.979613] ftec_csl_elite 0003:0EB7:0020.000F: Hires timer: period = 2 ms
[  409.554635] usb 2-1.1: current rate 16000 is different from the runtime rate 48000
```

</details>

gotzl, 2022-03-12:
Hi,

could you clear up which ID belongs to which mode? From the original dmesg I deduce:
1. 0004: 'yellow' mode, where it identifies as ClubSport Wheel Base V2.5
2. 0E07: 'Fanatec FANATEC WHEEL(PS5 / PS4 MODE)'
3. 0020: 'red' mode (PC native mode)
4. 0E04: 'Fanatec FANATEC WHEEL(CSL E WB PS4 Comp. Mode)')

So, you should put the wheel into 'red' mode, ie PID 0020. This PID is already correctly assigned in the driver, and from the `dmesg`, I can also see that the driver is properly picked-up, as the line `ftec_csl_elite 0003:0EB7:0020.000B: ....` indicates.

> Do Fanatec pedals, shifter, and handbrake accessories also work plugged directly to a wheel base, or do they need to be connected directly to the PC via USB?

The 'safe' way would be to connect the things directly to the wheel base. However, for all devices that are just generic HID devices (some axis/some buttons) w/o any special function, connecting with USB should work just fine as well.

24788329, 2022-03-13:
Your deduced mappings are correct, but also see the full results in each mode below for convenience. 

PC compatibility mode (yellow) correctly shows the wheel in Oversteer, but range is incorrectly set to 900. PC native mode (red) does not correctly show the wheel in Oversteer, but range is correctly set to 1080. Thank you for creating the Oversteer ticket to pick up the wheel in PC native mode (red). I'll continue using that for now since range is also correctly set to 1080. Out of curiosity, why does range change in the different modes?

In both PC compatibility and native modes, the [colored 5-way d-pads](https://fanatec.com/media/image/dc/2b/05/Product_thumbnail_GT_DD_PRO_composition_600x600.jpg) on the GT DD Pro steering wheel do not work in ACC. All of the other "regular" buttons work. What else can I try to get the colored buttons to work?

What do the different `Fanatec FANATEC Wheel: Invalid code 768 type 1` errors indicate? Are they related to the colored 5-way d-pad buttons not working above?

---

### PC compatibility mode (yellow)
idProduct=0004
hid-fanatecff sets range to 900
Oversteer sees the wheel
<details>
  <summary>dmesg output (yellow):</summary>

```
[ 1420.552080] usb 1-3: new full-speed USB device number 11 using xhci_hcd
[ 1420.717036] usb 1-3: New USB device found, idVendor=0eb7, idProduct=0004, bcdDevice= 6.57
[ 1420.717044] usb 1-3: New USB device strings: Mfr=1, Product=3, SerialNumber=0
[ 1420.717046] usb 1-3: Product: FANATEC Wheel
[ 1420.717048] usb 1-3: Manufacturer: Fanatec
[ 1420.745187] /home/userone/bin/hid-fanatecff-next/hid-ftec.c: ftec_probe: ifnum 0
[ 1420.745301] hid_map_usage: 34 callbacks suppressed
[ 1420.745302] Fanatec FANATEC Wheel: Invalid code 768 type 1
[ 1420.745305] Fanatec FANATEC Wheel: Invalid code 769 type 1
[ 1420.745306] Fanatec FANATEC Wheel: Invalid code 770 type 1
[ 1420.745307] Fanatec FANATEC Wheel: Invalid code 771 type 1
[ 1420.745308] Fanatec FANATEC Wheel: Invalid code 772 type 1
[ 1420.745309] Fanatec FANATEC Wheel: Invalid code 773 type 1
[ 1420.745310] Fanatec FANATEC Wheel: Invalid code 774 type 1
[ 1420.745310] Fanatec FANATEC Wheel: Invalid code 775 type 1
[ 1420.745311] Fanatec FANATEC Wheel: Invalid code 776 type 1
[ 1420.745312] Fanatec FANATEC Wheel: Invalid code 777 type 1
[ 1420.745388] input: Fanatec FANATEC Wheel as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:06:00.1/usb1/1-3/1-3:1.0/0003:0EB7:0004.000E/input/input41
[ 1420.804242] ftec_csl_elite 0003:0EB7:0004.000E: input,hiddev2,hidraw5: USB HID v1.11 Joystick [Fanatec FANATEC Wheel] on usb-0000:06:00.1-3/input0
[ 1420.804249] /home/userone/bin/hid-fanatecff-next/hid-ftec.c:  ... 1 1 1 56
               -128 127 -128 127

[ 1420.804253] /home/userone/bin/hid-fanatecff-next/hid-ftecff.c:  ... setting FF bits
[ 1420.804255] /home/userone/bin/hid-fanatecff-next/hid-ftecff.c: setting range to 900
[ 1420.804387] ftec_csl_elite 0003:0EB7:0004.000E: Hires timer: period = 2 ms
```

</details>

### PC native mode (red)
idProduct=0020
hid-fanatecff sets range to 1080
Oversteer *does not* see the wheel
 
<details>
  <summary>dmesg output (red):</summary>

```
[ 1697.753344] usb 1-3: new full-speed USB device number 13 using xhci_hcd
[ 1697.919399] usb 1-3: New USB device found, idVendor=0eb7, idProduct=0020, bcdDevice= 6.57
[ 1697.919405] usb 1-3: New USB device strings: Mfr=1, Product=3, SerialNumber=0
[ 1697.919408] usb 1-3: Product: FANATEC Wheel
[ 1697.919410] usb 1-3: Manufacturer: Fanatec
[ 1697.943499] /home/userone/bin/hid-fanatecff-next/hid-ftec.c: ftec_probe: ifnum 0
[ 1697.943614] hid_map_usage: 34 callbacks suppressed
[ 1697.943615] Fanatec FANATEC Wheel: Invalid code 768 type 1
[ 1697.943617] Fanatec FANATEC Wheel: Invalid code 769 type 1
[ 1697.943618] Fanatec FANATEC Wheel: Invalid code 770 type 1
[ 1697.943619] Fanatec FANATEC Wheel: Invalid code 771 type 1
[ 1697.943620] Fanatec FANATEC Wheel: Invalid code 772 type 1
[ 1697.943621] Fanatec FANATEC Wheel: Invalid code 773 type 1
[ 1697.943622] Fanatec FANATEC Wheel: Invalid code 774 type 1
[... 30 lines cut ...]
[ 1850.302955] hid-generic 0003:0EB7:0E04.0011: input,hiddev2,hidraw5: USB HID v1.11 Gamepad [Fanatec FANATEC WHEEL(CSL E WB PS4 Comp. Mode)] on usb-0000:06:00.1-3/input0
```

</details>

### PS5 / PS4 mode (blue)
idProduct=0e07
hid-fanatecff *does not* set any range
Oversteer *does not* see the wheel
 
<details>
  <summary>dmesg output (blue):</summary>

```
[ 2080.178991] usb 1-3: new full-speed USB device number 16 using xhci_hcd
[ 2080.345186] usb 1-3: New USB device found, idVendor=0eb7, idProduct=0e07, bcdDevice= 6.57
[ 2080.345191] usb 1-3: New USB device strings: Mfr=1, Product=9, SerialNumber=0
[ 2080.345193] usb 1-3: Product: FANATEC WHEEL(PS5 / PS4 MODE)
[ 2080.345195] usb 1-3: Manufacturer: Fanatec
[ 2080.373376] input: Fanatec FANATEC WHEEL(PS5 / PS4 MODE) as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:06:00.1/usb1/1-3/1-3:1.0/0003:0EB7:0E07.0013/input/input46
[ 2080.373596] hid-generic 0003:0EB7:0E07.0013: input,hiddev2,hidraw5: USB HID v1.11 Gamepad [Fanatec FANATEC WHEEL(PS5 / PS4 MODE)] on usb-0000:06:00.1-3/input0

```

</details>

gotzl, 2022-03-13:
> Out of curiosity, why does range change in the different modes?

Because in compat mode, the device identifies as a ClubSport Wheel Base, which is limited to 900 degs.

> What else can I try to get the colored buttons to work?

Do you see these d-pads when using https://github.com/Grumbel/sdl-jstest ?

> What do the different Fanatec FANATEC Wheel: Invalid code 768 type 1 errors indicate? Are they related to the colored 5-way d-pad buttons not working above?

I see these messages as well for the `CSL Elite`. I don't know what they mean, but they seem harmless.. I don't think that these messages are related to the issue with the d-pads.

zdv80, 2022-07-28:
Hi! I'm trying to use the DD Pro in Linux, thanks for all this work!! :)
In my case is strange because the wheel is well detected in Oversteer but when trying to use it, no games work with it correctly... :S In some games is impossible to map anything, and in others (like F1 2017 and Project Cars) the wheel produces crazy clicks in menus (like if some buttons where pressed constantly).

What can be the problem?

Regarding the 4 D-Pads  that has the wheel I tested it with evtest but doesn't produce any response... All other buttons sent events...

thanks!

gotzl, 2022-07-28:
Hi,
hmm.. sounds really strange. These issues are actually not related to the driver here - it is only for FFB. All the input shananigans are handled by the generic linux kernel HID driver.

Still, I think there are some things you can try:
* make sure your device is either in `yellow` or `red` mode (see https://github.com/gotzl/hid-fanatecff/issues/21#issuecomment-1065989765)
* try to disable disable steam input
* try swtching to proton 6.3
* the 'constant input' might come from a axis that is not centered - so maybe try to disconnect your pedals and check if s.t. changes

Let me know if any of these points have an effect ...  

With regards to the 4 D-Pads I cannot say much other than it seems they are not covered by the HID system,  since you say they don't produce an output with `evtest`. But w/o the device, I can't do much. Someone would have to record the USB traffic in windows while using these buttons to understand more about them ...

zdv80, 2022-07-29:
Aaahh ok I understand!! I'll try this things! :)

I captured the 4 D-Pads USB traffic in a windows machine... I don't know if this data is useful enough! :S 
I moved every Pad  1 time Up-Down-Left-Right-Push

[dpads.txt](https://github.com/gotzl/hid-fanatecff/files/9218063/dpads.txt)

thanks a lot!

gotzl, 2023-03-29:
Hi, just had some time and remembered this open issue with the D-Pads. The usb-dump you've provided suggests that the device doesn't send s.t. unexpected. So I checked the HID descriptor and found that (in case of my CSL Elite) there are 128 buttons reported.. And apparently, the linux HID code has some issue with this many buttons, [see this proposed patch](https://patchwork.kernel.org/project/linux-input/patch/CACa7zykn0q9XJAUvrqnNATr4DUv3Kc7XujF3vm6sfRB5pE6YNQ@mail.gmail.com/).
Not sure how deep you want to get involved here, but I think it would be worth to test if this patch helps here.

zdv80, 2023-03-30:
Hi gotzl! As far as I understand, is a patch that must be applied to the kernel? I never compiled it before, I'll investigate how to do it... 

thanks!!

gotzl, 2023-03-30:
Yes, it's a kernel patch... 
In the meantime, I try to come up with a solution that doesn't involve compiling the kernel ^^ I think I can remap buttons in the driver....

gotzl, 2023-03-30:
Hi again,
I've just pushed a test to the 'dpads' branch. With this code, I see much more buttons mapped in `evtest`. Although I'm not yet sure of the whole consequences by the chance ^^
Still, @zdv80 could you try it out? (s.t. like `git fetch && git checkout dpads`, then `make && make install && rmmod hid_fanatec && modprobe hid_fanatec`)

zdv80, 2023-04-01:
Hi!!
I tested it with evtest and the buttons are recognised! :)
But I did a very quick test in steam trying to assign it to the controller buttons but doesn't seem detected there... Maybe that can be another problem that hasn't involve the driver...

thanks!

gotzl, 2023-04-02:
Hi,
interesting ;) Could you send me the output of evtest (maybe doing s.t. similar as before in `dpads.txt` file)?
Also, could you try to bind the buttons in some racing game?

zdv80, 2023-04-17:
Hi, this is the evtest result:
[evtest.txt](https://github.com/gotzl/hid-fanatecff/files/11253903/evtest.txt)

I test a couple of games and it seems that the dpads can't be mapped :S

thanks!

gotzl, 2023-04-22:
Hi,
thanks for the test! So, there are no events of `type 1 (EV_KEY)`.... Hmm.. Guess I've to go dig a bit ...

gotzl, 2023-05-10:
Did you also try with the `Fanatec Tuning Menu` directly on the wheel? I think the menu-point is called `SEN`.

Yuuyatails, 2023-07-06:
While I don't have any Fanatec wheel base, I do have to mention that button layouts may vary depending on the equipped steering wheel as Fanatec wheel bases has a quick-release system that can swap between different steering wheels.

It is possible that the hid-fanatecff can't detect what wheel the wheel base currently have.

melroy89, 2023-12-17:
Can I safely assume I can buy the Fanatec GT DD Pro https://gtddpro.fanatec.com/ and using this kernel module?

I was looking into the Fanatec GT DD Pro Kit, which is a CSL wheel base and Fanatec CSL Pedals.

Please say yes :) If no 😢  , what is my best next option that is fully supported??

zFra65, 2023-12-17:
> Can I safely assume I can buy the Fanatec GT DD Pro https://gtddpro.fanatec.com/ and using this kernel module?
> 
> 
> 
> I was looking into the Fanatec GT DD Pro Kit, which is a CSL wheel base and Fanatec CSL Pedals.
> 
> 
> 
> Please say yes :) If no 😢  , what is my best next option that is fully supported??
> 
> 

Yes you can :), i have dd pro and using this kernel on arch linux, working good

melroy89, 2023-12-17:
> Yes you can :), i have dd pro and using this kernel on arch linux, working good

Great.. And also the Fanatec CSL Pedals are supported? Notice these are not the elite pedals I think. Maybe the readme needs to be updated. Hence my questions.

zFra65, 2023-12-17:
> > Yes you can :), i have dd pro and using this kernel on arch linux, working good
> 
> 
> 
> Great.. And also the Fanatec CSL Pedals are supported? Notice these are not the elite pedals I think. Maybe the readme needs to be updated. Hence my questions.

With arch they work

melroy89, 2023-12-19:
Thanks. I'm 99% sure I will buy the Fanatec DD Pro kit now. Let's go!

gotzl, 2023-12-19:
Hi there, just some notes:

The pedals don't require an extra driver. They are `Human Interface Devices (HID)` which use the generic HID kernel module and have no function on top. If you would by the load-cell module, then you can use this driver here which then allows you to set the load (when you connect the pedals to the PC directly).

Now, the wheelbase base itself (FFB) should work fine, but afaik, the 4 colored nobs on the wheel do not work. And I also don't know about it's LEDs and the display. Maybe others that own the wheel can confirm.

melroy89, 2023-12-19:
> Hi there, just some notes:
>
> The pedals don't require an extra driver. They are `Human Interface Devices (HID)` which use the generic HID kernel module and have no function on top. If you would by [...]

In that case I'm now looking into a [Fanatec CSL DD Ready2Race Bundle ](https://fanatec.com/eu-en/ready2race/csl-dd-ready2race-bundle-for-pc-5-nm)(for PC). With a CSL DD base and CSL Steering Wheel P1 V2 instead of the Gran Turismo bundle/steer, after all I'm **not** planning to use it on Xbox nor Playstation. And I believe the CSL Steering Wheel P1 V2 might feel and look better then the Grand Turismo steer. Any thoughts?

Upgrading with load cell pedals would be indeed by first upgrade. 

Ps. I go for the 5nm on purpose instead of 8nm, because of my own safety.

gotzl, 2024-01-18:
Closing as the GT DD Pro should work now. Issues with the wheel rim remain though, I've created a separat ticket for it.

closed 2024-01-18 by gotzl