gotzl/hid-fanatecff #22: FFB in F1 2020 / F1 2021

gotzl, 2022-02-05:
FFB in F1 2020 and F1 2021 is not working properly. It is very week and there are no centering forces at all.

BODYPRINT, 2023-09-06:
Has there been any progress with this? Would be nice to get full FF working in the F1 series of games.

gotzl, 2023-10-12:
Hi,

no progress, at least I did not work on it. I've just fired up F1 2021 and the situation did not change, although I seem to have no some weird oscillation which wasn't there before.

However, I've also just tested F1 22, and there, the FFB does make a lot more sense. I'm not sure how it is supposed to feel, but there are sane curb effects as well as centering forces. So, maybe the F1 series is now fine for games >= F1 22?

BODYPRINT, 2023-10-27:
Hi gotzl,

Thanks for your reply.
I tried F1 22 and FFB does indeed work and seems to work very well. Thanks for looking into it. I haven't got F1 23 yet, but hopefully when I do, it'll work in that as well. It will be a little while yet before I get, but when I do I'll report back.

Thanks again!

BODYPRINT, 2023-11-20:
I have a thought about FFB not working in F1 2021 and older.
Could it not be working because Fanatec setups like the CSL DD need to be in "compatibility" mode in order to be detected in those games, whereas in F1 22 this is not required anymore?

sean-skt, 2024-10-25:
Just wanted to confirm that force feedback works for both F1 2020 and F1 2021.

I tested with a CSL DD (8Nm) using Proton 7.0-6 in Pop_OS! 22.04. The CSL DD had to be in normal mode (NOT compatibility mode as recommended in the Fanatec forums) for the force feedback to work. As far as I could tell, all forces were there and it worked well. I was pretty impressed.

closed 2026-10-23 by gotzl