gotzl/hid-fanatecff #23: FFB in BeamNG.drive

gotzl, 2022-02-05:
FFB in BeamNG.drive behaves very strange.

ThatSkepticalGamer, 2022-12-27:
It seems to work fine when using the native version of Beamng Drive for Linux instead of running it through Proton.

michal2229, 2023-01-28:
Same thing on Logitech - need to use older proton or linux version.

gamingdoom, 2023-10-14:
I don't have a Fanatec wheel but I have a g29 and it doesn't work in Proton unless I turn on Full update type in the force feedback settings. Maybe that can fix it?

dinuxlt, 2023-12-15:
Thank you @gamingdoom for tip. CSL Elite FFB started to work after I selected `Full (slow fallback)`

[image]

gotzl, 2024-07-13:
For me, the `Fast` mode now works as well.

gotzl, 2024-07-13:
Sorry, false alarm. Still `Fast` is not working, I somehow missed the `Apply` button ...

Cykyrios, 2024-09-28:
Not sure if the 0.33 update helps at all (it includes FFB changes), but since it reset all my settings, I noticed I had no issues with FFB at all (other than inverting the FFB scale), Fast mode worked just fine without having to go through Full mode first.
If others can confirm this, I think the game can go in the "expected to work" list.

gotzl, 2025-02-01:
Works for me as well, closing.

commit 070ac02f (2024-07-13): fix start-solo, fixes #23
commit 7462b0ee (2024-07-14): fix start-solo (fixes #23), fix nullptr access in
commit e6d929a9 (2024-07-14): fix start-solo (fixes #23), fix nullptr access
commit ea884e3a (2024-08-20): fix start-solo (fixes #23), fix nullptr access
closed 2025-02-01 by gotzl