gotzl/hid-fanatecff #26: FFB in ACC and others (CSL DD)

dncrash, 2022-03-05:
Hi,

Using a CSL DD I can only get FFB in AC (the original one) while in ACC, Project Cars 2, Dirt Rally 2.0 and others it doesn't work.
At this point I'm not sure if in ACC it worked and then stopped or if it never worked in the 1st place.
I wanted to see if anyone else has this issue, if there are any known workarounds, and if there's anything I can do to help/troubleshoot.

Thanks!

gotzl, 2022-03-06:
Hi,
just tried ACC, and for me it was not working as well. It seems that this is a regression of Proton 7, though. I switched to Proton 6 and FFB was working fine again. I've updated the issue title accordingly.
So, for the moment, try with Proton 6. I'll try to figure out the issue with Proton 7 asap ...
Regards

gotzl, 2022-03-07:
Seems it's a regression that's affecting other Linux drivers as well. I see similar comments in the issue trackers for Automobilista, Dirt2 and BeamNG. I think Valve should already be aware of this and it should not take long for them to come up with something ... Still, I might look into it myself when I've time.

dncrash, 2022-03-14:
Thanks for replying! I had a chance to test over the weekend and can confirm ACC and Project Cars 2 work.
Hope Proton fixes the regression :crossed_fingers:

gotzl, 2022-03-14:
If you don't mind getting your hands dirty, there is a workaround that you could try: https://github.com/ValveSoftware/Proton/issues/2366#issuecomment-1062822328

dncrash, 2022-03-17:
Tried the regex workaround with proton GE_Proton7_9-1 in ACC and Project Cars 2 (which are working in Proton 6) - neither game receives any input from the wheel.

On a different topic, are Automobilista 2 and Dirt Rally 2.0 supposed to work? For Dirt Rally 2.0 FFB doesn't work no matter what Proton version I have (everything else like steering/buttons/pedals work fine). And in Automobilista 2 - again no FFB and I can't get it to assign the steering because it says multiple inputs detected. Don't know if there are any particular settings I should be applying for these.

dncrash, 2022-03-27:
Happy to report that I got Automobilista 2 working, I think it was due to in-game settings rather than anything else, but I am disabling steam input on every game now just in case :) 
Also got RaceRoom and rFactor2 working (everything with Proton 6.3-8).

odinu, 2023-10-10:
Seems to work fine now with latest version.

closed 2023-10-10 by odinu