gotzl/hid-fanatecff #29: Submitting driver to the Linux Kernel

PieterWaegeman, 2022-05-17:
Hi there!

I love your work. Any plans on submitting this driver to be added to the Linux Kernel?

Kind regards,
Pieter

gotzl, 2022-07-07:
Hi,
I would love to see it there as well, but it's nowhere near to meet the coding standards required to get it there. In addition, I think it would be favorable to have a common solution for all these out-of-tree FFB drivers, like [new-lg4ff](https://github.com/berarma/new-lg4ff) (were I pretty much stole all the code from), or [hid-tmff2](https://github.com/Kimplul/hid-tmff2) ... Unfortunately, I don't have the time to work in this direction atm ... Buy let's see what the future brings ;)

ZakMcKrack3n, 2024-02-17:
@gotzl  Maybe we can start small by collecting all the wheel base ids and making a pull request for systemd , eliminating the need for setting fuzz/deadzone ?

Someone did that for simucube devices, at least they use git pull requests (unlike main kernel development):
https://github.com/systemd/systemd/pull/25501/commits/fbe73bda72bf6e12ac1e38229e02bca55ac7f804