gotzl/hid-fanatecff #35: Makefile issues

ParzivalWolfram, 2022-12-02:
Provided makefile tries to invoke Makefile from `/usr/lib/modules/<kernel>/build`, despite never copying anything to that location, so nothing actually happens.

```
root@john-deb:/home/john/hid-fanatecff# make
-e 
:: Compiling Fanatec kernel module
========================================
make -C /lib/modules/`uname -r`/build M=$PWD
make[1]: Entering directory '/usr/lib/modules/6.0.0-5-amd64/build'
make[1]: *** No targets specified and no makefile found.  Stop.
make[1]: Leaving directory '/usr/lib/modules/6.0.0-5-amd64/build'
make: *** [Makefile:8: default] Error 2
```

gotzl, 2022-12-03:
This module is a so called `out-of-tree` kernel module. There is no copying of files involved building such a module, it just has to enter the kernel-module-build context, which is below the path you've mentioned.
However, the kernel headers/sources are required for this to work, ie `/usr/lib/modules/<kernel>/build` has to point to the location of your kernel headers/sources. So could you please check if the kernel headers/sources are located there. If not, check with your distribution and install at least the kernel headers (for debian/ubuntu probably s.t. like `sudo apt install linux-headers-$(uname -r)`).

ParzivalWolfram, 2022-12-05:
I have my kernel's headers installed. Other dkms modules work fine, as well, but only when being actively managed by `apt`. This seems to be a completely unrelated issue. Closing.

closed 2022-12-05 by ParzivalWolfram