gotzl/hid-fanatecff #37: Unable to send strings like " 1 ", " 2 ", " N ", " R ", "91" to the driver

GoLD-ReaVeR, 2023-02-20:
I've been messing around with the fanatec led python file to do some stuff with the gearing and to my dismay I found that there's several issues at foot.

For one I cannot center the gear on the LED display and it will instead shows 00X numbers and neutral (N) as well as reverse (R) cannot be sent. Whenever there's a 9 to be displayed, it will also not be sent to the display and I will instead see "X X" or "XX " or "X  " or something in that vein (where X are numbers). As an alternative I can use 0, but similarly I can also not send a -1 to the driver.

The thing I was working on was displaying gear switches as well as actual speed. I got something implemented, but I'm kinda fighting with this bug.

gotzl, 2023-02-20:
Hi,
I think some reverse engineering is required to get this wheel supported.. All I did so far was done for the [Club Sport steering wheel F1](https://fanatec.com/eu-en/steering-wheels/clubsport-steering-wheel-f1-esports-v2), and some things work with other wheels, some don't.
Here, my wheel had only a display composed of three 8segs, while your wheel seems to have a proper display where individual pixels can be controlled?
I am not sure how deep you want to get involved, but I think a USB traffic dump would be required to learn the commands for controlling the display...

GoLD-ReaVeR, 2023-02-20:
I do not know how to do a USB traffic dump.

gotzl, 2023-02-22:
The general procedure would be to boot into Windows, install and fire-up Wireshark with USB monitoring.
Then, we need some application that tries to use the display / LEDs. I think the Fanatec configuration tool should have a feature to test the various functions of the wheel.. If so, start capturing with Wireshark, then trigger the display / LED test, stop capturing and safe the capture file. (Also maybe check in wireshark that there is really s.t. happening when triggering the test...) 
I would then need the capture file and a (precise) description of what the display / LED was doing during the test.

ZakMcKrack3n, 2023-04-30:
@GoLD-ReaVeR you could try my branch [here](https://github.com/ZakMcKrack3n/hid-fanatecff/tree/seven-segment-test).
 I was just adding a crude implementation to display letters on my seven segment display e.g: "bAt", "Go" "YEL" "BLU" "rEd" .

Also single characters now display on my wheel, but I can only center numbers/letters by using unsupported characters like: "x4x" (i am echoing to the sysfs path, did not use the python scripts yet but using a custom one for testing ams2).

I also added brackets , so depending on the python script "[2]" could work , in the shell I have to escape it: "[2\]"

I am not well versed in C so hopefully nothing is broken , I only played around with the display code however , and since nothing is accessing it automatically it should be save to test with.

gotzl, 2023-12-08:
@GoLD-ReaVeR could you test with the most recent code in the `next` branch? Is this sufficient for you?

GoLD-ReaVeR, 2023-12-08:
Hi, I did a yay -S hid-fanatecdd-dkms which should be getting the next branch. My findings are:

- 999 is still not rendered (completely blank).
- r/R is shown not as an R but rather 3rd and 4th line of a 2 in the center.
- n/N is similarly shown as an R with the added extra vertical line creating a kind of n shape.
- -1 is rendered as a 1 on the right most side.
- Everything seems to center now as the code did indicate.

It's definitely better, but considering the device's firmware is showing far more complex things I'm kinda puzzled. Is this thing falling back to some kind of emulation mode or something?

gotzl, 2023-12-08:
Yes, I think so. The existing code is intended for the displays found on 'older' wheels, which consist of three 7seg displays. It seems that the commands that worked for this display partially work for the newer LCD displays as well, but not fully... In order to get them fully supported, I need to reverse engineer the protocol used to talk to the display... Let's see Santa brings a new wheel ;)

closed 2026-10-18 by gotzl