gotzl/hid-fanatecff #38: I've got 6 more LEDs on my Formula 2.5X

GoLD-ReaVeR, 2023-02-20:
Could you look for them? :D

I'd like to bind traction control and abs triggers to the lights. As well as perhaps the current gear.

gotzl, 2023-02-20:
Kind of similar situation as for #37, we would need a USB traffic dump to see how this LEDs are controlled..

Cykyrios, 2023-12-23:
I just got myself the Clubsport F1 2023 wheel, which is basically the same as the Formula 2.5 wheel. Here's a Wireshark trace of the LED test and a gif of the LED animation (starting with the side LEDs).
[led.pcapng.zip](https://github.com/gotzl/hid-fanatecff/files/13758747/led.pcapng.zip)

[image]

I did the same for the OLED display (the test starts with the empty white screen): [display.pcapng.zip](https://github.com/gotzl/hid-fanatecff/files/13758748/display.pcapng.zip)

[image]

@gotzl let me know if you need anything else, I'd be happy to help.

Cykyrios, 2023-12-27:
So, I had a look at the trace and found the following:
* Packets are length 64, though only 21 values are actually used.
* Packets start with `ff 01`, then `00` or `01` to select either rev LEDs or flag LEDs, then 2 values for the color of each LED (18 values for rev LEDs and 12 values for flag LEDs).
* 5 colors are used in the test: `00 00` (off), `00 1f` (red), `07 e0` (green), `f8 00` (blue) and `ff ff` (white).

Colors are encoded in the BGR565 format.
With the above, the first line of the test being `ff 01 01 00 1f 00 00 00 00 00 1f 00 00 00 00`, the 1st and 4th flag LEDs (bottom on both sides) are red.

gotzl, 2023-12-27:
Hey, thanks for the dumps and already analyzing the LEDs function. I'm gonna work on that soon, I'll wire this up in the kernel  using the [multicolor](https://docs.kernel.org/leds/leds-class-multicolor.html) subsys.
Question is how to organize the sysfs files. What do you think about having `LFLAG1..LFLAG3` and `RFLAG1..RFLAG3` for the flags and the `RPM1..RPM9` for the others?
Also, could you post the value of the `wheel_id` sysfs file?

Cykyrios, 2023-12-27:
`LFLAG`, `RFLAG` and `RPM` sounds good to me, I just went through the wheels on Fanatec's website, and it does seem like all wheels that have them have 9 RPM LEDs and 2x3 flag LEDs.
I'll get back to you about `wheel_id` on friday or saturday as I won't be able to check before then, but I assume it won't be the same value as the `0020` that was associated with my CS DD wheelbase, and rather something to tell apart the wheel itself from other wheels?

gotzl, 2023-12-27:
Yep, AFAIU the `wheel_id` should identify the mounted wheel rim...

Cykyrios, 2023-12-29:
Hey, got back to my rig, `wheel_id` is `0x000a`.

gotzl, 2023-12-29:
K, guess that checks out. Apparently, it has the same ID as the `Formula V2` wheel.

Cykyrios, 2024-04-21:
Hi, while I'm waiting for this to be added, I'd like to ask a few things:
* Are RPM LEDs supposed to be working at all even without this PR? On my wheel, the first RPM LED lights up in solid white and stays there the entire time, on the desktop or in game (LFS, ACC, EA WRC).
* I've now got a second wheel, a round one, using the Button Module Endurance: would it be useful if I got the same Wireshark trace I did for the Formula wheel? (FYI it does the same as my other wheel, first LED is on (but solid yellow))
* When the PR is completed and merged, will the flag LEDs be controlled through hid-fanatecff-tools or directly through this repo? For LFS in particular, I have implemented the InSim protocol using Godot/GDScript, which means I can get various events (such as yellow flags), but I don't know how I would send them to the wheel.

gotzl, 2024-04-23:
Hi!

* The multicolor LEDs use a different protocol (you've decoded it, packets start with 0xff) compared to the 'normal' LEDs (packets start with 0xf8). I cannot say if the 'normal' protocol is doing s.t. for the multicolor LEDs. You can check that easily by writing to the sysfs files `.../leds/0003:0EB7:0005.*::RPM*/brightness`
* The Button Module Endurance has single-color RPM LEDs? Then those should already work. The 'flag' LEDs should then work with the PR. Can you give me the reported `wheel_id` for this thing? The display is another story. Unfortunately, the dump you've provided for the other wheel doesn't contain s.t. useful. So it would be great if you could provide another dump for this. Maybe a dump from in-game where the display is really showing s.t.?
* This repo only provides the sysfs interface that allows access to these 'extended' wheel features from userspace. `hid-fanatecff-tools` is then using the sysfs interface with info from games. So you could add LFM to `hid-fanatecff-tools`, or  you could write directly to the sysfs files I would guess.

Cykyrios, 2024-04-23:
The Button Module Endurance has multi-color LEDs just like the Formula wheel, the wireshark trace for the LED test appears to be the same (or at least very close) for both wheels. I'm not sure I'll be able to do anything useful about the display though, it only works in Legacy mode (without all the telemetry data), I'll have to see if I can find a drive to put a non-VM Windows on and test that. LFS _kinda_ works on VirtualBox but I get no LEDs at all, though it may be because of the game itself.

I'm somewhat confused about `wheel_id` for the Button Module Endurance as I first found `0x04`, but it sometimes changes to `0x06` or even `0x00` when I switch wheels, right now I have `0x04` for the Formula wheel too...

I managed to write to the sysfs file (0 or 1) for RPM LEDs and they do turn on or off. Is #60 in a state that I could test colors (and flags) in the same way?

gotzl, 2024-04-23:
> LFS kinda works on VirtualBox but I get no LEDs at all, though it may be because of the game itself.

Afaiu, games use the `Fanatec SDK` in windows to drive the extended features of the wheel. I don't think that this is integrated into LFM ...

> I'm somewhat confused about `wheel_id` ...

I found this `wheel_id` by looking into what field changes when putting on another wheel.  I found `0x04` for my `CSL_ELITE_STEERING_WHEEL_WRC_ID`. So it seems that the way I try to get the `wheel_id` is not correct :(

> I managed to write to the sysfs file (0 or 1) for RPM LEDs and they do turn on or off

Ok, so then you should get some blinking by using `hid-fanatecff-tool`, I guess ...

> Is https://github.com/gotzl/hid-fanatecff/pull/60 in a state that I could test colors (and flags) in the same way?

I think in principle yes. But the sysfs files it creates depend on the `wheel_id`, and the functions for the Formula wheel are selected when this is `0x0a` ... So in the current state you won't get the correct sysfs files. I guess we've to figure the `wheel_id` out, first. (You could modify the IDs in `hid-ftec.h` to fit what you're getting, though ...)
Also note that it is under development. So I don't guarantee that it doesn't crash your PC ;)

closed 2025-02-01 by gotzl