gotzl/hid-fanatecff #40: CSL Steering Wheel P1 V2 support

ZakMcKrack3n, 2023-04-28:
Thanks to you excellent driver I got the fanatec bundle with the CSL Steering Wheel P1 V2.

I dont know how similar it is to the WRC wheel, so will report my findings anyway:
- LED is initialized as 'green' when turning the wheel base on
- driver reports all LEDs as 0
- setting any LED to 1 then back to 0 turns it off completely
- RPM 1-3 all turn on the GREEN LED
- RPM 4-6 all turn on the RED LED
- RPM 7-9 all turn on the BLUE LED
- any combination is possible , so GREEN / YELLOW / RED / PURPLE / BLUE /  CYAN and WHITE can be set mixing the appropriate colored LEDs
- 3x7 segment display: sending numbers work as expected and -1 turns it of, initialized off.

How do I go about to find out the wheel ID ? Is this already accessible in the `wheel` node of the device ?
-> this reports it as 0008 for me
```
#define CSL_STEERING_WHEEL_P1_V2 0x0008
#define CSL_ELITE_STEERING_WHEEL_WRC_ID 0x0112
```

gotzl, 2023-05-10:
Thanks for your feedback!
> Is this already accessible in the wheel node of the device ?
Yes, that's it. I'll add it to the header.

So, is there s.t. missing or s.t. that could be improved? I would think about splitting `wheel` part from the `wheel base` part, so that each wheel gets its own driver. For this wheel for example, the sysfs could the offer simpler interface to the LED.

ZakMcKrack3n, 2023-05-11:
No , everything seems to work , I am not sure about the LED init (GREEN on as default) , for now I simply immediately run a script to turn it off again , not sure , maybe a side effect of some initialization code for this wheel.

As for now I did not check how it works with the python LED server , I only tested with a custom python script and AMS2.

gotzl, 2023-05-11:
>  I am not sure about the LED init
This is on purpose, I find it practical to see whether the driver properly loaded.
> I did not check how it works with the python LED server ... LED progression is RED -> YELLOW -> WHITE as all the bits are turned on with increasing rpm.
Jea, this thing also needs some attention.. But no time so far.. On my wheel, I have the LEDs in a row, so each RPM sysfs entry represents one LED on the wheel (and wheel-base). So, low RPM would mean RPM1-3 are on, high RPM would mean RPM1-9 is on. I guess, you could modify the led-server to do s.t. different when it detects your wheel ...

ZakMcKrack3n, 2023-05-12:
> 
> This is on purpose, I find it practical to see whether the driver properly loaded.
> 
Ah ok , so I think I can find that and disable it in my branch for now , I am not sure if its visible if all LEDS are turned on then immediately off again ? (to save the same purpose)
> 
> Jea, this thing also needs some attention.. But no time so far.. (and I play mostly VR lately..) On my wheel, I have the LEDs in a row, so each RPM sysfs entry represents one LED on the wheel (and wheel-base). So, low RPM would mean RPM1 is on, high RPM would mean RPM1-9 is on. Seems to fit your findings?. I guess, you could modify the led-server to do s.t. different when it detects your wheel ...

Yes, in the custom script I already mixed colors to have something completely different, so from the driver side they are all working and addressable.

So if implementing "quirks" based on this and probably the WRC wheel is a different "feature" , I will close this one for now, at least the ID is known so it can be handled.

closed 2023-05-12 by ZakMcKrack3n

gotzl, 2023-10-15:
Hey, thanks for the PR with the changes to the display! I'll check that out later.
Just wanted to let you know that there is another PR on the LED server side that will add AMS2 support: https://github.com/gotzl/hid-fanatecff-tools/pull/2 Maybe this of interest to you..