gotzl/hid-fanatecff #43: No FFB Gran Turismo DD

voxors, 2023-10-14:
Hi,
I am trying to use this driver for FFB, but it doesn't seem to work on my side.
The driver seems to load but I got no FFB 
```
[ 1097.389284] usb 1-2: new full-speed USB device number 12 using xhci_hcd
[ 1097.708840] usb 1-2: New USB device found, idVendor=0eb7, idProduct=0020, bcdDevice= 6.25
[ 1097.708851] usb 1-2: New USB device strings: Mfr=1, Product=3, SerialNumber=0
[ 1097.708854] usb 1-2: Product: FANATEC Wheel
[ 1097.708858] usb 1-2: Manufacturer: Fanatec
[ 1097.732736] hid_map_usage: 34 callbacks suppressed
[ 1097.732740] Fanatec FANATEC Wheel: Invalid code 768 type 1
[ 1097.732746] Fanatec FANATEC Wheel: Invalid code 769 type 1
[ 1097.732748] Fanatec FANATEC Wheel: Invalid code 770 type 1
[ 1097.732751] Fanatec FANATEC Wheel: Invalid code 771 type 1
[ 1097.732753] Fanatec FANATEC Wheel: Invalid code 772 type 1
[ 1097.732755] Fanatec FANATEC Wheel: Invalid code 773 type 1
[ 1097.732758] Fanatec FANATEC Wheel: Invalid code 774 type 1
[ 1097.732760] Fanatec FANATEC Wheel: Invalid code 775 type 1
[ 1097.732762] Fanatec FANATEC Wheel: Invalid code 776 type 1
[ 1097.732765] Fanatec FANATEC Wheel: Invalid code 777 type 1
[ 1097.732863] input: Fanatec FANATEC Wheel as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/usb1/1-2/1-2:1.0/0003:0EB7:0020.000E/input/input29
[ 1097.733118] ftec_csl_elite 0003:0EB7:0020.000E: input,hiddev98,hidraw5: USB HID v1.11 Joystick [Fanatec FANATEC Wheel] on usb-0000:02:00.0-2/input0
[ 1097.733383] ftec_csl_elite 0003:0EB7:0020.000E: Hires timer: period = 2 ms
```
fftest report no ffb effect implemented too.

gotzl, 2023-10-14:
Hi there, could you post the output of `evtest`? The Output should include s.t. like this
```
  Event type 21 (EV_FF)
    Event code 80 (FF_RUMBLE)
    Event code 81 (FF_PERIODIC)
    Event code 82 (FF_CONSTANT)
    Event code 83 (FF_SPRING)
    Event code 85 (FF_DAMPER)
    Event code 88 (FF_SQUARE)
    Event code 89 (FF_TRIANGLE)
    Event code 90 (FF_SINE)
    Event code 91 (FF_SAW_UP)
    Event code 92 (FF_SAW_DOWN)
```

voxors, 2023-10-14:
There is the result of evtest.
```
No device specified, trying to scan all of /dev/input/event*
Not running as root, no devices may be available.
Available devices:
/dev/input/event6:      Logitech G513 RGB MECHANICAL GAMING KEYBOARD
/dev/input/event7:      Logitech G513 RGB MECHANICAL GAMING KEYBOARD Keyboard
/dev/input/event8:      Fanatec FANATEC Wheel
Select the device event number [0-8]: 8
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0xeb7 product 0x20 version 0x111
Input device name: "Fanatec FANATEC Wheel"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 288 (BTN_TRIGGER)
    Event code 289 (BTN_THUMB)
    Event code 290 (BTN_THUMB2)
    Event code 291 (BTN_TOP)
    Event code 292 (BTN_TOP2)
    Event code 293 (BTN_PINKIE)
    Event code 294 (BTN_BASE)
    Event code 295 (BTN_BASE2)
    Event code 296 (BTN_BASE3)
    Event code 297 (BTN_BASE4)
    Event code 298 (BTN_BASE5)
    Event code 299 (BTN_BASE6)
    Event code 300 (?)
    Event code 301 (?)
    Event code 302 (?)
    Event code 303 (BTN_DEAD)
    Event code 704 (BTN_TRIGGER_HAPPY1)
    Event code 705 (BTN_TRIGGER_HAPPY2)
    Event code 706 (BTN_TRIGGER_HAPPY3)
    Event code 707 (BTN_TRIGGER_HAPPY4)
    Event code 708 (BTN_TRIGGER_HAPPY5)
    Event code 709 (BTN_TRIGGER_HAPPY6)
    Event code 710 (BTN_TRIGGER_HAPPY7)
    Event code 711 (BTN_TRIGGER_HAPPY8)
    Event code 712 (BTN_TRIGGER_HAPPY9)
    Event code 713 (BTN_TRIGGER_HAPPY10)
    Event code 714 (BTN_TRIGGER_HAPPY11)
    Event code 715 (BTN_TRIGGER_HAPPY12)
    Event code 716 (BTN_TRIGGER_HAPPY13)
    Event code 717 (BTN_TRIGGER_HAPPY14)
    Event code 718 (BTN_TRIGGER_HAPPY15)
    Event code 719 (BTN_TRIGGER_HAPPY16)
    Event code 720 (BTN_TRIGGER_HAPPY17)
    Event code 721 (BTN_TRIGGER_HAPPY18)
    Event code 722 (BTN_TRIGGER_HAPPY19)
    Event code 723 (BTN_TRIGGER_HAPPY20)
    Event code 724 (BTN_TRIGGER_HAPPY21)
    Event code 725 (BTN_TRIGGER_HAPPY22)
    Event code 726 (BTN_TRIGGER_HAPPY23)
    Event code 727 (BTN_TRIGGER_HAPPY24)
    Event code 728 (BTN_TRIGGER_HAPPY25)
    Event code 729 (BTN_TRIGGER_HAPPY26)
    Event code 730 (BTN_TRIGGER_HAPPY27)
    Event code 731 (BTN_TRIGGER_HAPPY28)
    Event code 732 (BTN_TRIGGER_HAPPY29)
    Event code 733 (BTN_TRIGGER_HAPPY30)
    Event code 734 (BTN_TRIGGER_HAPPY31)
    Event code 735 (BTN_TRIGGER_HAPPY32)
    Event code 736 (BTN_TRIGGER_HAPPY33)
    Event code 737 (BTN_TRIGGER_HAPPY34)
    Event code 738 (BTN_TRIGGER_HAPPY35)
    Event code 739 (BTN_TRIGGER_HAPPY36)
    Event code 740 (BTN_TRIGGER_HAPPY37)
    Event code 741 (BTN_TRIGGER_HAPPY38)
    Event code 742 (BTN_TRIGGER_HAPPY39)
    Event code 743 (BTN_TRIGGER_HAPPY40)
    Event code 744 (?)
    Event code 745 (?)
    Event code 746 (?)
    Event code 747 (?)
    Event code 748 (?)
    Event code 749 (?)
    Event code 750 (?)
    Event code 751 (?)
    Event code 752 (?)
    Event code 753 (?)
    Event code 754 (?)
    Event code 755 (?)
    Event code 756 (?)
    Event code 757 (?)
    Event code 758 (?)
    Event code 759 (?)
    Event code 760 (?)
    Event code 761 (?)
    Event code 762 (?)
    Event code 763 (?)
    Event code 764 (?)
    Event code 765 (?)
    Event code 766 (?)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      0
      Min        0
      Max    65535
    Event code 1 (ABS_Y)
      Value      0
      Min        0
      Max    65535
    Event code 2 (ABS_Z)
      Value      0
      Min        0
      Max    65535
    Event code 3 (ABS_RX)
      Value      0
      Min        0
      Max    65535
    Event code 4 (ABS_RY)
      Value      0
      Min        0
      Max    65535
    Event code 5 (ABS_RZ)
      Value      0
      Min        0
      Max    65535
    Event code 6 (ABS_THROTTLE)
      Value      0
      Min        0
      Max      255
    Event code 7 (ABS_RUDDER)
      Value      0
      Min     -128
      Max      127
    Event code 16 (ABS_HAT0X)
      Value      0
      Min       -1
      Max        1
    Event code 17 (ABS_HAT0Y)
      Value      0
      Min       -1
      Max        1
    Event code 40 (ABS_MISC)
      Value      0
      Min        0
      Max      255
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
  Event type 21 (EV_FF)
    Event code 80 (FF_RUMBLE)
    Event code 81 (FF_PERIODIC)
    Event code 82 (FF_CONSTANT)
    Event code 83 (FF_SPRING)
    Event code 85 (FF_DAMPER)
    Event code 88 (FF_SQUARE)
    Event code 89 (FF_TRIANGLE)
    Event code 90 (FF_SINE)
    Event code 91 (FF_SAW_UP)
    Event code 92 (FF_SAW_DOWN)
Properties:
Testing ... (interrupt to exit)

```

voxors, 2023-10-14:
Sorry for the inconvenience. I get some ffb with fftest 
```
Force feedback test program.
HOLD FIRMLY YOUR WHEEL OR JOYSTICK TO PREVENT DAMAGES

Device /dev/input/event8 opened
Features:
  * Absolute axes: X, Y, Z, RX, RY, RZ, Throttle, Rudder, Hat 0 X, Hat 0 Y, Misc ,
    [FF 00 03 00 00 01 00 00 ]
  * Relative axes: 
    [00 00 ]
  * Force feedback effects types: Constant, Periodic, Spring, Damper, Rumble, 
    Force feedback periodic effects: Square, Triangle, Sine, Saw up, Saw down, 
    [00 00 00 00 00 00 00 00 00 00 2F 1F 00 00 00 00 ]
  * Number of simultaneous effects: 16

Uploading effect #0 (Periodic sinusoidal) ... OK (id 0)
Uploading effect #1 (Constant) ... OK (id 1)
Uploading effect #2 (Spring) ... OK (id 2)
Uploading effect #3 (Damper) ... OK (id 3)
Uploading effect #4 (Strong rumble, with heavy motor) ... OK (id 4)
Uploading effect #5 (Weak rumble, with light motor) ... OK (id 5)
```

Now I just need to understand why overrsteer don't seem to work.

closed 2023-10-14 by voxors

gotzl, 2023-10-14:
Alright, no worries! :)
Note that some buttons on your wheel are not supported, like the 4 round knob thingies.
One thing oversteer needs is to have the correct rights for the sysfs files, see the 'Permissions' section in the readme. Also, oversteer dev doesn't have fanatec wheels, so it is questionable what works and what not...