gotzl/hid-fanatecff #46: Wheel not recognized in F1 23

gotzl, 2023-11-08:
The game starts with 'default' Steam Input settings, but it does not recognize the wheel. 
When setting `Disable Steam Input`, the game doesn't start at all.

gotzl, 2023-11-08:
I've got the game to recognize the wheel by setting `Disable Steam Input`, then starting the game with Proton 6.3-8 (the game doesn't fully start, but some frames seem to be enough). Then switching back to current Proton version.

gotzl, 2023-11-08:
@ciscoyo I'll create a separate ticket when I had time to confirm the raceroom issue. Could you check if the Proton 6.3-8 trick works for you as well?

ZakMcKrack3n, 2023-11-17:
"Good news everyone"
As its currently the free weekend I also downloaded the game , got it to run using the latest custom from glorious eggroll:

https://github.com/GloriousEggroll/proton-ge-custom/releases/download/GE-Proton8-23/GE-Proton8-23.tar.gz

Force feedback also seems to work, wheel base was in native mode and displayed in the game as "CSL DD Pro".

The archive has to be extracted and put into `/home/yourUserName/.local/share/Steam/compatibilitytools.d/` and then can be selected in steam game settings proton version dropdown (if steam was running , needs restart to recognize the custom proton version).

So , no apparent driver issue using this version here.

ZakMcKrack3n, 2023-11-18:
@ciscoyo Proton Experimental for me still crashed for me yesterday, maybe I have to try again with a clean prefix.

For me pedals are also both marked like this, I think its just internal reuse of game pad symbology and reminiscent of former codemasters stuff, it looks weird , but if its works , just ignore it.

dinuxlt, 2023-11-18:
F1 23 this weekend is free, so decided to download and test.
My default proton is: 7.0-6
It started with no issues.

CLS Elite + McLaren V2 wheel + V3 pedals worked out of the box. FFB, buttons, wheel, pedals.. all good and as expected (not sure about FFB because I don't have windows to compare).

ZakMcKrack3n, 2023-11-18:
> u just installed ge proton and everything works out of the box? what distro r u using?

Yes it did and of course I use arch btw 😅.
Wanting to try again I set up image sharpening by VKBASALT (totally unrelated) , now I get a denuvo anti tamper message and it links to a page I got to wait 24h.
So retesting anything is no longer possible for me at the moment.

ZakMcKrack3n, 2023-11-19:
@ciscoyo It seems switching proton versions triggered denuvo lockout , every new proton version is like an activation for them , found a bunch of threads for other games where this also happened, not great feature when the game crashes and we want to get it to run in the first place....
So VKBASALT is probably not to blame , I was merely following your tip and tried experimental , this was one switch to much.

magguns, 2024-01-08:
I just want to add, this whole switching Proton versions from experimental -> 6.3-8 -> experimental was absolutely necessary for me on all FF games I've tried on a fresh Ubuntu 23.10 install with the hid-fanatecff driver in order to make them run. They would either crash on startup (AMS2, EA WRC) or when starting a race/stage (DR2.0) 100% of the time until the until the 6.3-8 trick was performed. When switching to Proton 6.3-8, starting the game, and switching to experimental again they all worked. I've tested this multiple times and am able to reproduce it, both with the snap-distributed Steam package and the native package.

magguns, 2024-01-08:
Perhaps the Proton 6.3-8 trick should be mentioned in the README for new users coming here?

gotzl, 2024-01-18:
I've created an entry in the README. Closing this as solved as the workaround exists, I'll try to figure out what's going on with this, maybe in a separate ticket.

closed 2024-01-18 by gotzl

melroy89, 2024-01-24:
I needed to downgrade back to `GE-Proton8-25` to get the wheel detected in Assetto Corsa. It's a big issue I think.