gotzl/hid-fanatecff #53: Raceroom Racing Experience - no ffb

SynthetikzZ, 2023-12-19:
is this something thats fixeable by you? or can i contribute in some way to make it happen ?

game detects wheel and pedals, i can drive, just with no ffb at all.

game runs only with proton GE 6.25 for me.

gotzl, 2023-12-19:
Hmm.. for me it is working. But I had to go through proton6/7 to get the wheel recognized and FFB. [Here is the comment](https://github.com/gotzl/hid-fanatecff/issues/42#issuecomment-1817918717) where I described what I did. Maybe this helps for you as well?

That said, after the recent update of the game, I think the FFB feels different and a bit weak. So I'll have to look into that...

SynthetikzZ, 2023-12-19:
i will try it later when at home.. does switching proton versions do anything to the prefix like installing files or something?

SynthetikzZ, 2023-12-20:
so your right , ffb works. very lightly but i could up it trough the ingame multiplier. but weirdly raceroom decides the lock range of each car and i cant adjust it from ingame, so i have to append the wheelrange trough sysfs.. doing echo 450 >> range gives me this:
`synthetikz@synthetikz-pc /s/m/h/d/h/0003:0EB7:0020.0001> echo 450 >> range
warning: An error occurred while redirecting file 'range'
open: Keine Berechtigung

`
it says no permissions (im in the games group)

doing 

`echo 450 | sudo tee -a range
`
works, but nothing changes, also the soft lock of the wheel does not change.

so rightnow i have to do 540° rotation to on side to do 90° ingame and i cant fix that.

SynthetikzZ, 2023-12-20:
i tried it out for couple of minutes and all these "trueforce" effects work great, but on ffb multiplier 3x there is strong force but it is more like a constant spring, giving no information about tire slip at all. all important ffb effects kinda missing. i dont know how it should feel on windows tho. very arkward.

gotzl, 2023-12-20:
> so rightnow i have to do 540° rotation to on side to do 90° ingame and i cant fix that.

jea, the range sysfs file doesn't work for DD wheel.. Not sure why no one complaint so far, or I overlooked it ^^ But I'm working on that in another branch. In the meantime, just use the tuning menu on the wheel directly and set the 'SEN' entry. I've put 400deg, also make sure to align this with the visuals in-game (video -> advanced -> wheel-rotation).

>  all these "trueforce" effects work great
What do you mean with that? Note that 'trueforce' is a technical term for proprietary logitech stuff...

What I feel makes a large difference is setting `FOR` and `DPR` to 100. With these, it feels kind of alright to me.. At least I feel the over/understeer, (weak) curb effects etc ...

SynthetikzZ, 2023-12-20:
That's why I put trueforce into the quotation marks. 

Raceroom has some effects( they are turned off by default, you can turn them on in the settings) for motor vibrations, road effects and gear shift jolts. Basically the same what Logitech is doing with trueforce (also just software). Does feel really nice.

SynthetikzZ, 2023-12-20:
I mean weird as like rf2 I have my wheel setup to 1080, and rf2 does the wheel lock thing based on this, I don't have to change settings per car no matter what wheel lock it has. In raceroom I have to switch wheel configs if changing car.  Raceroom should have a global wheel range I can set to the range my wheel is setup to, and then adjusting wheel lock per car on a software basis.

SynthetikzZ, 2023-12-20:
You don't get it. In rf2 I set my overall wheel range to 1080 and never ever have to adjust range to the car. It always uses the correct wheel range for the specific car I drive, without the need to adjust in the wheelmenu /driver. In rre I have to match the range in the wheel menu to every car I drive in the game which is an extra step that's not really needed. In rf2 my wheel is set to 1080 but it still uses like example 450 degrees of rotation for the specific car I drive.

SynthetikzZ, 2023-12-21:
No wheel is set to 1080 fixed, and in rf2 its set to 1080 globally. Rf2 does the rest in matching wheel range on a per car basis. Nomatter what you drive you dont have to touch your wheel config at all. 

In rre rightnow I have to adjust the sen setting for every car I hop into the wheel movement in the game is identical to my wheel. Setting sen to automatic changes nothing for me. I also have to set the sen setting to the half of what the car in rre has.

Like car has 440 degrees rotation I have set sen to 220 so it matches up. All that fiddling I don't have to do on other sims.

gotzl, 2023-12-21:
Just a quick comment: writing to the sysyfs file or using range slider in oversteer is (or will be in the future) the same as adjusting `SEN` on the wheel.

gotzl, 2023-12-21:
Let's take rF2 as an example. In this game, you've to calibrate the wheel first. So the game asks you to turn the wheel to either the full left/right limit, and then to 90 degrees. Now the game knows the translation from raw input (in terms of the raw values it got from the wheel) to 'degrees'. With this, it can create a Soft-Lock effect at a given rotation that depends on the car you're sitting in. And it will adjust the Steering-Ratio according to the car you're sitting in (sitting in F1 car, 90degree gives more turning than sitting in a truck).
This calibration surly depends on the `SEN` setting. If you change the `SEN` setting, say from 360 to 720, then you'd have to turn double the rotation as before.

> but what is, when u have different values/ratio in these?

You can't. sysfs/oversteer/SEN all refer to the same value. Or let's say, changing one updates the others as well.

liszca, 2024-01-02:
@SynthetikzZ 
Does RRRE work for you? Is it only the thing with the steering angle that needs to be adjusted for every car?

SynthetikzZ, 2024-01-02:
Yeah only have to adjust sensitivity. Switching proton versions from 6 over 7 to latest Experimental did the trick for me. But I'm not too much into this game I enjoy rf2 way more :D. 

Can you give me some examples of games using friction and inertia effects(happy if they are somewhat older games that I can get cheap, happy if it's not ea haha) ? Have everything setup to make you the USB dumps.

SynthetikzZ, 2024-01-18:
From my side there is nothing. 

Regarding the protontrick. Maybe you are able to compare a fresh proton6 prefix with a fresh protonexperimental prefix with the diff function to list all differences.

closed 2024-01-27 by gotzl