gotzl/hid-fanatecff #57: possible code source for friction / inertia effects ?

SynthetikzZ, 2023-12-27:
so i kept thinking how you could implement it correctly and then i stumbled upon openffbboard. it has its own opensource software so i started searching the code and found a hint 
```

#define FFB_EFFECT_CONSTANT     0x01
#define FFB_EFFECT_RAMP         0x02
#define FFB_EFFECT_SQUARE       0x03
#define FFB_EFFECT_SINE         0x04
#define FFB_EFFECT_TRIANGLE     0x05
#define FFB_EFFECT_SAWTOOTHUP   0x06
#define FFB_EFFECT_SAWTOOTHDOWN 0x07
#define FFB_EFFECT_SPRING       0x08
#define FFB_EFFECT_DAMPER       0x09
#define FFB_EFFECT_INERTIA      0x0A
#define FFB_EFFECT_FRICTION     0x0B
#define FFB_EFFECT_CUSTOM       0x0C
 
```

here is the link to the file i found that  https://ultrawipf.github.io/OpenFFBoard/doxygen/ffb__defs_8h_source.html

my programming skills are non existent so i dont know what im searching exactly for but maybe thats a good source for you ?

https://github.com/Ultrawipf/OpenFFBoard

gotzl, 2023-12-27:
Interesting peace of code, but it doesn't help much, unfortunately. While the idea is similar, the protocol how to communicate with the device (eg the command codes you've posted) are vendor specific.

There are only 2 ways to find out the Fanatecs protocol:
a) they provide a specification for it
b) by reverse engineering (using USB-trafffic dumps of known input).

Since a) is not likely to happen, we're stuck with b).. So I think what we need is a game using `FF_FRICTION` and another one using `FF_INERTIA` and a windows user that is willing to create some USB-traffic dumps ....
For one of these, `EA Sports WRC` could be used for example.

SynthetikzZ, 2023-12-28:
In the next couple of days I will setup a windows machine ( have some parts laying around for a second PC) and install windows, i will dump all the USB traffic for you

gotzl, 2023-12-29:
> In the next couple of days I will setup a windows machine ( have some parts laying around for a second PC) and install windows, i will dump all the USB traffic for you
> 
> There has to be some sort of documentation for you with what I did with the wheel for any given time right?

This would be great! Not much docu is needed for this. Just playing the game for a few seconds should be enough. Then one can filter the data for unknown command codes.

SynthetikzZ, 2024-01-01:
Is EA sports WRC the only game you know with the effects? Have every

closed 2024-01-01 by SynthetikzZ

gotzl, 2024-01-02:
@SynthetikzZ Up to now, besides WRC I only found Dirt Rally 2 that requires `FF_FRICTION`. Do you have that?
I did not find a game utilizing `FF_INERATIA` yet ...

SynthetikzZ, 2024-01-02:
Yes I have it in my steam library. I will do a dump as soon as I am at home again( im at work right now).

Do you need other dumps out of the fanatec driver itself? (Wheel centering/calibration or whatever ?)

I can provide the dumps as pcap or as JSON what do you prefer?

gotzl, 2024-01-02:
Great! Other than the game, what I would be interested in is the 'initialization' sequence. So, a capture of the traffic when powering on the device. (Make sure that fanatec tray app is not running, and also no other fanatec app is open at that time). And then a dump of the traffic when opening the fanatec controller settings app.

> I can provide the dumps as pcap or as JSON what do you prefer?

I'd prefer pcap. Thank you!

SynthetikzZ, 2024-01-02:
CSL DD wheelbase with the CSL P1 v2 wheelrim. 

I'll do the dumps later :)

SynthetikzZ, 2024-01-03:
https://github.com/SynthetikzZ/fanatec_usb_dump

you can find the files there

if you need something else just ask. and thank you for your work

gotzl, 2024-01-03:
Thanks! But the DirtRally2 file lacks content... The other files look fine, though. Could you redo DR2?

SynthetikzZ, 2024-01-03:
tought it so, dr2 had very light ffb for me (barely noticible but it was there). seems to be a problem of the game as its seems to be a problem a lot of people have :D

have to resolve it first then or have to find another game :D

gotzl, 2024-01-03:
Hmm... Maybe try editing `DiRT Rally 2.0/input/devices/device_defines.xml` file? Or try it in compatibility mode.

SynthetikzZ, 2024-01-03:
i just did some other things and had good ffb strength going on from the tarmac etc. i just uploaded a new dump could you check ?

SynthetikzZ, 2024-01-03:
added a dump in csw mode.. definitly csw gave me the best ffb in the game now

gotzl, 2024-01-03:
Thanks! Had a brief look into the data, but sadly couldn't identify where the `FF_FRICTION` is set ... Have to think about that...

SynthetikzZ, 2024-01-03:
did you tried the csw dump instead? ffb felt alot better there.

gotzl, 2024-01-04:
> did you tried the csw dump instead? ffb felt alot better there.

The csw dump looks similar. Strange that the FFB felt different. Did you use the same setting in tuning-menu and in-game?
Another question, when did you start the capture? Could you start the capture before starting the game?

> im not sure how this all work, but i think u have to trigger these eventcodes, no?

Yes, you have to 'trigger' them. But what I saw in Dirt Rally 2 (proton) was, that it sends `FF_FRICTION` already at a standstill at the beginning of the stage...

SynthetikzZ, 2024-01-04:
Yes same settings, csw mode feels thousand times nicer compared to PC mode, also display and LEDs working in csw mode, not in PC mode. 

I started the dump while already in-game, later today I'll do a complete dump from wheelbase initialization to in-game driving.

dinuxlt, 2024-01-04:
(maybe this issue should be `open`? it's `closed` now)

Hi @gotzl I have suggestion here.

I have windows application, the title says `iRacing Force Feedback Test 1.72` `WheelCheck.exe` downloaded from this link https://www.racedepartment.com/downloads/lut-generator-for-ac.9740/

I have windows only in VirtualBox. Connected my CSL Elite to linux box and forwarded my wheel to windows VM. Installed all drivers and this application sees my wheel. I can enable effect, for example `Simple Spring` and I get this on wheel.

There is Damping Force list: Damper, Inertia, Friction

[image]

Maybe this is possibility to use this application, change effects and monitor whats happens on USB port ?

**WARNING !!!** in some off these effects, wheel goes crazy

[image]

gotzl, 2024-01-04:
@dinuxlt Hey there! Yes, I've this tool as well and using it in qemu is exactly what I did to find out the other effects. Strange thing here is, that choosing Dampler/Inertia/Friction all result in the same op-code... (and I also believe that the wheel didn't feel differently)

gotzl, 2024-01-04:
> Yes same settings, csw mode feels thousand times nicer compared to PC mode, also display and LEDs working in csw mode, not in PC mode.
> 
> I started the dump while already in-game, later today I'll do a complete dump from wheelbase initialization to in-game driving.

Just fired up DR2 (proton) again. What I see is that the game already sends `FF_FRICTION` just after the stage got loaded. So even before choosing 'Start' and sitting in the car. So I would also be interested in a capture of only the loading screen + some seconds in the menu before the stage.

gotzl, 2024-01-04:
> > @dinuxlt Hey there! Yes, I've this tool as well and using it in qemu is exactly what I did to find out the other effects. Strange thing here is, that choosing Damper/Inertia/Friction all result in the same op-code... (and I also believe that the wheel didn't feel differently...)
> 
> yea, im testing right now.. so for me `ndp` / damper and `nfr` / friction feels completely the same now.. it wasnt before the update though. before update, the behavior was better, or lets say it **WAS** how it should feel! but not anymore. `nin` / inertia is fine.

Which update do you mean?

To my understanding, NDP/NFR/NIN are sort of base effects that are applied by the wheel directly, independent of the information coming from games. That's why they are called 'Natural ...'. You can change these values w/o a game running and you'll feel the difference. So the driver shouldn't have an influence on how these settings feel.

melroy89, 2024-01-16:
> @SynthetikzZ Up to now, besides WRC I only found Dirt Rally 2 that requires `FF_FRICTION`. Do you have that? I did not find a game utilizing `FF_INERATIA` yet ...

I'm also willing to dump USB packages if you wish. I also have a Windows machine running to the right of me, that have the official Fanatac software running.

I believe that Natural Friction is used in Asseta Corsa as well. But I could be wrong. I do notice quite some oscillation in Asseta Corsa with this Linux driver, I hope that natural friction (NFR) can help to prevent oscillation. Or like you said maybe it's just there in the firmware (or the Fanatec Windows software) that deals with this, instead of the game itself.

SynthetikzZ, 2024-01-16:
Pretty sure that the natural ff effects are all independent from the game you are playing.

Sorry @ gotzl for not updating the dumps. Something happened in Reallife, I wasn't at home since then . Tomorrow I can do the dumps you asked for.

melroy89, 2024-01-16:
Should we just create a new issue for implementing friction / inertia? And do not close that ticket, until some version of friction/invertia is implemented?

gotzl, 2024-01-19:
@melroy89 thanks for offering to test with windows as well

I've recently found [ForceTest](https://github.com/denisn73/JoyFFB/blob/master/Software/ForceTest.exe), but it doesn't work in qemu for whatever reason. It does work with wine, though. So, @melroy89 @SynthetikzZ, what I would be interested in is 
* usb-dump when enabling the friction effect
* qualitative comparison of how the wheel feels in windows vs wine/linux when setting friction/damping

SynthetikzZ, 2024-01-19:
`Enumerated Joystick : FANATEC Wheel
Enumerated Joystick : FANATEC Wheel
Supported Effects : 
   - Constant
   - Ramp Force
   - Square Wave
   - Sine Wave
   - Triangle Wave
   - Sawtooth Up Wave
   - Sawtooth Down Wave
   - Spring
   - Damper
   - Inertia
   - Friction
   - CustomForce

Creating effects for FANATEC Wheel
Error creating Spring Effect. Your joystick may not support this effect.  : 
E_INVALIDARG : 
An invalid parameter was passed to the returning function
Error creating Friction Effect. Your joystick may not support this effect.  : 
E_INVALIDARG : 
An invalid parameter was passed to the returning function
Error creating Damper Effect. Your joystick may not support this effect.  : 
E_INVALIDARG : 
An invalid parameter was passed to the returning function
Error creating Constant Force Effect. Your joystick may not support this effect.  : 
E_INVALIDARG : 
An invalid parameter was passed to the returning function
Error creating Periodic Effect. Your joystick may not support this effect.  : 
E_INVALIDARG : 
An invalid parameter was passed to the returning function
Error creating Pavement Groove Effect. Your joystick may not support this effect.  : 
E_INVALIDARG : 
An invalid parameter was passed to the returning function`

not working for me.. throws me this in its diagnosticswindow

SynthetikzZ, 2024-01-19:
https://github.com/SynthetikzZ/fanatec_usb_dump/blob/main/dirtrally_bootup_to_racestart_menu.pcapng

added the dump for the usbtraffic before the racemenu.. defenitly some commnunication happening just before it loads up.

gotzl, 2024-01-20:
> ```
> Enumerated Joystick : FANATEC Wheel
> Enumerated Joystick : FANATEC Wheel
> Supported Effects :
>    - Constant
>    - Ramp Force
>    - Square Wave
>    - Sine Wave
>    - Triangle Wave
>    - Sawtooth Up [...]

Oh, looks similar to what I got in qemu... Then it seems this prog just doesn't like windows....

SynthetikzZ, 2024-01-20:
It was win 11 tho maybe I can try it in compability mode or maybe it works for win 7. I will have a look into it as soon as I have spare time for it.

melroy89, 2024-01-20:
To be honest this forcetest. exe is also 6 years old. Moreover, the same log output (logfile.txt) was submitted to the repo: https://github.com/denisn73/JoyFFB/blob/cb8fd82d052a8348af35eea3b370def19f93e85e/logfile.txt#L31

melroy89, 2024-01-20:
Ps. I also tried this [ForceTest.exe](https://github.com/denisn73/JoyFFB/blob/master/Software/ForceTest.exe) same results as you're seeing. I tried compatibility mode Windows 7 and even Windows XP. No luck. I also tried running as admin. Also no luck.
I bet the binary is compiled with a too old .NET maybe?

Anyhow, I have everything up and running under Windows (I have the drivers installed already, I have AC running, I have recently now installed Device Monitoring Studio Ultimate as well):

[image]
[image]

gotzl, 2024-01-30:
> What is next?

In terms of the `FF_FRICTION/FF_INERTIA`, I actually don't know. The most recent dumps don't contain more information, unfortunately. And I would say that with the current code, DR2 should produce similar USB traffic/cmds in Linux vs Windows... But does it feel similar?

Another topic would be to create USB traffic dumps for FW upgrades for various parts, especially the base. Not sure if FW update will ever be supported in Linux, but having the dumps surly doesn't hurt...

melroy89, 2024-01-30:
> In terms of the `FF_FRICTION/FF_INERTIA`, I actually don't know. The most recent dumps don't contain more information, unfortunately. And I would say that with the current code, DR2 should produce similar USB traffic/cmds in Linux vs Windows... But does it feel similar?

Not sure about DR2, but under AC (assetto corsa) I do feel a big difference in friction / inertia under Windows vs Linux. It doesn't oscillate as much under Windows (under Linux my steer under AC, if I let it go, goes from left to right, to left to right very intense and fast). Also I noticed the steering wheel is going back to enter again when you drive in a straight line and leave your hands off the wheel, just like a real car.

In terms in USB dumps, I think it's basically the same kind of USB packages. Maybe that is why you do not really see a difference?

Getting USB traffic dumps during a firmware upgrade would be nice indeed.

gotzl, 2024-01-30:
Hmm... I don't think I have this oscillation in AC. I'll check later.

> In terms of USB dumps, I think it's basically the same kind of USB packages. Maybe that is why you do not really see a difference?

What do you mean by that? I wanted to say, that the commands that are send to the wheel look similar in Linux and Windows. So in terms of the fanatec-usb-protocol, I don't see a difference...

melroy89, 2024-01-30:
> What do you mean by that?

Maybe you expected different kind of USB packages for friction and/or inertia...

gotzl, 2024-01-30:
> > What do you mean by that?
> 
> Maybe you expected different kind of USB packages for friction and/or inertia...

Yes, I expect different packages for damper/friction/inertia, but what the windows dumps show so far is that they are not different. That's what's puzzling...

melroy89, 2024-01-30:
I think it's just the same kind of packages, but let's say the racing game force feedback says rotate the wheel 20 degrees, with inertia and friction algorithms within the driver it will only rotate 10 degrees for example and reduce the torque or visa versa.

Just thinking out loud here... Not sure if this is how the driver is working under Windows.

gotzl, 2024-01-30:
Ah, yes, I was thinking in this direction too ... Maybe have to give it a shot ...

gotzl, 2024-02-23:
Does s.o. own F1 2X and is willing to capture windows usb-traffic for it? There are still some doubts about the effects these games use ...