gotzl/hid-fanatecff #59: Try Earlier Commit

liszca, 2024-01-07:
How can I try an earlier commit?

I want to try again commit: 9525d77e6f7b5e533e291d41e6986582dcf2b4bf

As I have the feeling something is missing in the forcefeedback, but not sure what it is.

gotzl, 2024-01-08:
Just do `git checkout 9525d77` and `make` it. Then you can remove the currently loaded version (if it's loaded) with `sudo rmmod hid_fanatec` and use the version you've just compiled with `sudo insmod hid_fanatec.ko`.

> As I have the feeling something is missing in the forcefeedback, but not sure what it is.

Which game?

liszca, 2024-01-12:
> Which game?

Its RRRE.
 In the above commit my wheel doesn't center. In RRRE it gives me more feeling what the cars rear is doing. I prefer that behavior.

In ACC I have to use Scaling Peak, but its fine, I feel what the car is doing. But the bugs when doing adjustments to the wheel came back.

having the FFB of this commit with the better wheel button-handling of the current version.

How about fixing my problem by Using Oversteers "Center wheel" Toggle under Tools? Or is it not as simple as that?

ZakMcKrack3n, 2024-01-13:
> Its RRRE. In the above commit my wheel doesn't center. In RRRE it gives me more feeling what the cars rear is doing. I prefer that behavior.

@liszca I also have auto center now, if i toggle the wheel base mode to compatibility and back (CSL DD here), its gone.

@gotzl  not sure what is different, guess something in the initialization ?

liszca, 2024-01-13:
> 
> wait what? can u be more precise on what exactly do u feel better from the rear? 
with commit 9525d77 I can feel something is going on at my cars rear. The effect is very slight, so that with the Auto Centering of the other commit I can only feel something when its already to late!

liszca, 2024-01-14:
> what u (should) feel from the rear, is the oversteer wheel/tire axi physics result only. the wheels/tire/axi want to go the way with the least resistant, always. that means, the more oversteer happen, the forces will increase, when u hold the steering wheel tight/same angle.

Yes, thats what happens. I prefer to feel the slight forces to so I don't Overdrive the car. And thats what I have with that older commit. 

And the "Auto Centering" eats that feedback effect so I don't recognize it anymore.

liszca, 2024-01-14:
@gotzl 
Can you check how to fix it? I want that back into the newest Version, would call it a regression fix.

liszca, 2024-01-16:
> @gotzl Can you check how to fix it? I want that back into the newest Version, would call it a regression fix.

@gotzl back on the current version it looks all good to me compared to 7925d77

closed 2024-01-16 by liszca