gotzl/hid-fanatecff #61: Initial make errors out...

smbika007, 2024-01-09:
Hi, I am running Ubuntu 22.04.2 in a VM on Windows 10. I was attempting to make a quick dry run on installing the drivers before I do it for real on a workstation running the same with a FANATEC CSL Elite Wheel Base PS4. I cd-ed to the folder with the Makefile and code and issued the make command which resulted in:6.2.0-39-generic

-e 
:: Compiling Fanatec kernel module
========================================
make -C /lib/modules/`uname -r`/build M=$PWD
make[1]: Entering directory '/usr/src/linux-headers-6.2.0-39-generic'
make[1]: *** No rule to make target 'Drivers/hid-fanatecff'.  Stop.
make[1]: Leaving directory '/usr/src/linux-headers-6.2.0-39-generic'
make: *** [Makefile:8: default] Error 2

I am running KVERSION 6.2.0-39-generic.

Please advise?

Thanks!
Sean

SynthetikzZ, 2024-01-09:
To get your wheel work in a VM you have to setup iommu hardware passthrough (hardware has to support it) and in windows the fanatec driver has to be uninstalled. Also you have to passthrough your GPU to get any game inside the VM run at a acceptable performance, if you only have 1 GPU that is not possible from a windows host. 

You better off trying it dualbooting from Linux

smbika007, 2024-01-09:
> To get your wheel work in a VM you have to setup iommu hardware passthrough (hardware has to support it) and in windows the fanatec driver has to be uninstalled. Also you have to passthrough your GPU to get any game inside the VM run at a acceptable performance, if you only have 1 GPU that is not possible from a windows host.
> 
> You better off trying it dualbooting from Linux
> 
> I use the CSL DD and Linux support thanks to this driver is not 100% complete but it's complete enough for me not needing windows at all anymore.

Thanks for the quick response! I actually don't need it to un on my VM. I am just going through the motions to prep for when  I do it for real on the above mentioned dual boot system at work. I am hoping this issue doesn't occur on the dual boot. It's hard to say if the issue is my configuration or just the fact that I am running on a VM. I looked at the other issues posted and found one quite similar (**https://github.com/gotzl/hid-fanatecff/issues/35**). The answer said to make sure headers/sources were pointed to in the /lib/modules/<kernel>/build folder. That folder has an "include" folder which is my only indication that headers may be pointed to there. Sources did not seem to be visible. I may have to do what they suggested which is to run the following command (probably, they said): **sudo apt install linux-headers-$(uname -r)**.

If that rings a bell, let me know if you can, but again, thanks for the quick response - much appreciated!

Sean

smbika007, 2024-01-09:
So, I ran that command, sudo apt install linux-headers-$(uname -r) and the result was that it was aleady up to date and the result from make command was still that same :-(

gotzl, 2024-01-09:
Hmm.. that's strange.. Can you show the output of `ls /usr/src/linux-headers-6.2.0-39-generic`?

smbika007, 2024-01-09:
> Hmm.. that's strange.. Can you show the output of `ls /usr/src/linux-headers-6.2.0-39-generic`?

Sure thing:

sean@sean-WINTERMUTE:/mnt$ ls /usr/src/linux-headers-6.2.0-39-generic
arch   certs   Documentation  fs       init      ipc     Kconfig  lib       mm              net   samples  security  tools   usr
block  crypto  drivers        include  io_uring  Kbuild  kernel   Makefile  Module.symvers  rust  scripts  sound     ubuntu  virt

smbika007, 2024-01-09:
maybe a better view:

gotzl, 2024-01-09:
Hmm.. this looks good. Now please the output of `make -n` (inside the Drivers/hid-fanatecff).

smbika007, 2024-01-09:
**sean@sean-WINTERMUTE:~/FANATEC Drivers/hid-fanatecff$** make -n
echo -e "\n::\033[32m Compiling Fanatec kernel module\033[0m"
echo "========================================"
make -C /lib/modules/`uname -r`/build M=$PWD
make[1]: Entering directory '/usr/src/linux-headers-6.2.0-39-generic'
make[1]: *** No rule to make target 'Drivers/hid-fanatecff'.  Stop.
make[1]: Leaving directory '/usr/src/linux-headers-6.2.0-39-generic'
make: *** [Makefile:8: default] Error 2

gotzl, 2024-01-09:
Hmm ;) Was hopping for a little more output ^^ I've the same Ubuntu in a VM and for me it all works ^^ So, bit puzzling, so guess we're having to do some back-and-forth here ... 

What kind-of looks strange to me is that it doesn't report the full path here `No rule to make target 'Drivers/hid-fanatecff'`. Can you output `$PWD` (again, inside the Drivers/hid-fanatecff).

smbika007, 2024-01-09:
you may have to help me there - I  issue the command "@ echo $PWD" and I get this:

sean@sean-WINTERMUTE:~/FANATEC Drivers/hid-fanatecff$ make -n
echo -e "\n::\033[32m Compiling Fanatec kernel module\033[0m"
echo "========================================"
echo 'WD'
make -C /lib/modules/`uname -r`/build M=$PWD
make[1]: Entering directory '/usr/src/linux-headers-6.2.0-39-generic'
make[1]: *** No rule to make target 'Drivers/hid-fanatecff'.  Stop.
make[1]: Leaving directory '/usr/src/linux-headers-6.2.0-39-generic'
make: *** [Makefile:9: default] Error 2

gotzl, 2024-01-09:
Ahhh, I see now ;) There is a space in the path ;) Sooo... there are quotations missing in the Makefile! Can you do `git pull origin next` to get the latest version of the code and try again?

smbika007, 2024-01-09:
> Ahhh, I see now ;) There is a space in the path ;) Sooo... there are quotations missing in the Makefile! Can you do `git pull origin next` to get the latest version of the code and try again?

Will do - give me a sec...

smbika007, 2024-01-09:
ok - did that (after stashing the mods to Makefile) and this is the current Makefile (default section):

KVERSION ?= `uname -r`
KERNEL_SRC ?= /lib/modules/${KVERSION}/build
MODULEDIR ?= /lib/modules/${KVERSION}/kernel/drivers/hid

default:
	@echo -e "\n::\033[32m Compiling Fanatec kernel module\033[0m"
	@echo "========================================"
	$(MAKE) -C $(KERNEL_SRC) M="$$PWD"

and the result of the make -n

echo -e "\n::\033[32m Compiling Fanatec kernel module\033[0m"
echo "========================================"
make -C /lib/modules/`uname -r`/build M="$PWD"
make[1]: Entering directory '/usr/src/linux-headers-6.2.0-39-generic'
Makefile:149: *** building multiple external modules is not supported.  Stop.
make[1]: Leaving directory '/usr/src/linux-headers-6.2.0-39-generic'
make: *** [Makefile:8: default] Error 2

smbika007, 2024-01-09:
do i need to upgrade make?

gotzl, 2024-01-09:
Ah damn ... Should've checked first. You did everything right!
Looks I need to do some research how to support spaces in the path. Can you for now just rename `FANATEC Drivers/` to `FANATEC_Drivers/`? This should workaround the issue ...

smbika007, 2024-01-09:
Ah, i see now - no prob!

yeah - THAT was dramatically better!

Thanks - sometimes it's just staring you in the face...much appreciated!
Sean

gotzl, 2024-01-09:
Yes, sure ;) I'll have to check how other external-module maintainers handle this case ^^ Until then, I'd like to keep this open...

smbika007, 2024-01-09:
Sounds good!

closed 2026-10-24 by gotzl