gotzl/hid-fanatecff #62: get raw data and ForceFeedBack Control on Linux

NeighborDva, 2024-01-15:
[image]

Hello.

For the purpose of developing remote control for autonomous driving
As I'm looking for force feedback control (handle control) and raw data of the steering wheel as shown in the image above, I'm asking you because the racing wheel most similar to the actual car seems to be Fanatec's dd series.

Do you have any advice on controlling the handle with force feedback and how to get raw data for the handle and pedal?

https://github.com/kuriatsu/ros-g29-force-feedback
Referring to the link above, it seems to be based on linux/input.h. 
Can Fanatec's product also control raw data acquisition and handle with force feedback based on that library?

gotzl, 2024-01-18:
> Can Fanatec's product also control raw data acquisition and handle with force feedback based on that library?

I had a quick look into the code. I would say that the code is not too g29 specific. The parameters will be surly different with other wheel, but the general logic should work for other wheels as well. As you said, it uses `linux/input.h`, which is common API for wheels, pedals, buttons ...

gotzl, 2024-07-07:
Closing due to no activity. Please reopen if you've further questions.

closed 2024-07-07 by gotzl