gotzl/hid-fanatecff #64: New release?

melroy89, 2024-01-16:
I wondering, we are on this `next` branch for about 3 years? What about creating a new release :)? 

The new release doesn't need to be 'perfect', it will never be "perfect" but always an ongoing journey. Software development always is like that.

Anyhow, I hope you will just merge the `next` branch into the master branch. Switch the default branch, and create a tagged/release version in GitHub. This helps you maybe mentally but also it shows the commitment and creating a new release shows activity of the project for the outside world. 

And continue creating feature branches or if you wish a `develop` branch. But I think just a master/main branch should be sufficient.

RayBa82, 2024-01-21:
I would love this! I found out, that archlinux distributes an [AUR](https://aur.archlinux.org/packages/hid-fanatecff-dkms) of this project, but it is really outdated :-D

And it would be really cool to have this integrated into linux gaming distros like [ChimeraOS](https://github.com/ChimeraOS/chimeraos) or [bazzite](https://github.com/ublue-os/bazzite).

And the driver itself seems to be very stable, as far as I can see, you are just constantly adding new features and nothing breaks. So what should happen? :-D 

This is open source and we are very thankful for your commitment.

melroy89, 2024-02-20:
Just do it ;) New release, add a version number, this will become the tag: https://github.com/gotzl/hid-fanatecff/releases/new

gotzl, 2024-02-21:
Yes, I'll do that next week. I was hoping to find the issue with the proton-prefixes first ....

melroy89, 2024-02-22:
Yea that make sense.

MpJupp, 2024-02-25:
Sounds good. Does this mean the AUR will be updated as well?

melroy89, 2024-02-25:
> Sounds good. Does this mean the AUR will be updated as well?

Somebody called mcoffin is maintainer of this package.

gotzl, 2024-03-05:
Didn't find the issue with the proton-prefixes yet, but still went for the release. I hope mcoffin will react and update his package. I also hope that people using other distributions contribute by creating/maintaining packages. I don't expect the driver to end-up in linux-kernel for some while.

closed 2024-03-05 by gotzl

melroy89, 2024-03-05:
Congrats with your first tagged release! No worries it's only a start, more releases will eventually come out. But I think it's a good step. Tagging my first release always feel good, I hope for you as well.