gotzl/hid-fanatecff #66: `AE SPORTS WRC` crashes when fanatec base connected

dinuxlt, 2024-01-22:
Game: https://store.steampowered.com/app/1849250/EA_SPORTS_WRC/
ProtonDB: https://www.protondb.com/app/1849250

This https://www.protondb.com/app/1849250#cl6HSkl-qE comment says that proton GE Proton 8-20 and GE Proton 8-21 works.
What I found personally, that GE 8-20, 8-21, 8-25 does not work for me.

If fanatec base (CSL Elite) connected and game started, then game will crash at the end of first video when AE logo displayed.
If I run game with NO base connected, then game will start and will be playable with keyboard.
If I connect base in game menu, or in game - game will crash. 

From other comments (from protondb), looks like Logitech wheels does not have this issue.

While testing I deleted wine (proton) prefix few times and now I get `You can no longer authenticate on new devices` It means, with new prefix game things that he is installed on new device. And now I reached 5 new devices per day limit.

gotzl, 2024-01-23:
Hi,

I can reproduce this behavior, it is also similar in Dirt Rally 2. Current workaround is to first start the game with proton6 (and a fresh prefix), then switch to the proton version you actually want.

I've not yet found the cause of this ...

dinuxlt, 2024-01-23:
Thanks it works now.

What I did:
* Deleted game prefix `1849250`
* Selected Proton 6.3-8
* Connected the wheel
* Started the game, game crashed
* Selected Proton 8.0-5
* Started the game. And game just works with no issues.

closed 2024-01-23 by dinuxlt