gotzl/hid-fanatecff #67: Issue with pfx'es created by proton7/8

gotzl, 2024-01-27:
A number of games crash when proton prefix was created by proton7/8 and a wheel is connected.
The following games I've found to suffer from this problem
```
AMS2 - crash on startup
DR2 - crash on loading stage
EA WRC - crash on startup
F1 22/23 - crash on startup
```
As mentioned in the README, the workaround is to start the game once with proton6 to create the pfx, then switch to proton7/8.

I'm currently investigating the issue, but no conclusion yet. I'll update the ticket with my findings...

Note: Some games seem to work fine for me with prefix created by proton7/8. So far, these are `rF2, R3E, PC2`.

liszca, 2024-04-01:
> AMS2 - crash on startup

Could it be they have the same Issue I have?
The game does not start when the wheel is powered on. How can I check for those prefixes? Having the problem after installing Linux Mint 21.3

Helge102, 2024-04-01:
Am Samstag, 27. Januar 2024, 15:42:30 CEST schrieb gotzl:
> A number of games crash when proton prefix was created by proton7/8 and a
> wheel is connected. The following games I've found to suffer from this
> problem
> ```
> AMS2 - crash on startup

I just installed ASM2 with GE-Proton-9.2 a few days ago.
This has nothing to do with Proton 7/8, but for me it works perfectly with the 
CSL Elite.

-- 
Gruß
Helge

liszca, 2024-04-01:
> I just installed ASM2 with GE-Proton-9.2 a few days ago. This has nothing to do with Proton 7/8, but for me it works perfectly with the CSL Elite.

How can I figure out which proton versions are installed? On Steam some Proton Versions don't show me their numbers

Helge102, 2024-04-01:
Am Montag, 1. April 2024, 17:49:38 CEST schrieb liszca:

> How can I figure out which proton versions are installed? On Steam some
> Proton Versions don't show me their numbers

I use ProtonUp-Qt to manage the versions.
I have no idea why you don't see some version numbers in Steam.

https://davidotek.github.io/protonup-qt/

melroy89, 2024-05-23:
I needed to downgrade back to GE-Proton8-25 to get the wheel detected in Assetto Corsa (AC). Instead of using the latest proton 8.

Any idea what is the root cause?

gotzl, 2024-05-28:
For me, AC with proton 8 works fine (I use ContentManager, but shouldn't make a difference). But I also did not do any further testing with AC (like starting with a clean proton pfx) because it is one of my main games and I don't want to break it ;)

> Any idea what is the root cause?

First, the issue is really not caused by hid-fanatec. The crash also happens without having hid-fanatec loaded, it is enough to have the wheel connected and powered-on.
It seems that the issue is with HID enumeration and how games access HID enumerated devices. However, I'm not yet able to pinpoint where it goes wrong as debugging crashes leaves me in game-code where I cannot see what's going on.
I'm currently collecting ammunition for creating a ticket in the wine/proton bug-tracker, maybe s.o. else has a good idea. (There was also a kind of similar issue with Unity games https://github.com/ValveSoftware/Proton/issues/7284)

gotzl, 2024-06-21:
I think I've finally found the root cause of these games crashing. It seems to be an issue of the `FanatacSDK` (or how it is integrated into these games). The `FanatecSDK` code-path is only entered when a Fanatec VID/PID combination is found and inside this code-path, it expects to find certain HID devices with very specific naming.
However, this specific naming of HID devices ([multi TLC](https://learn.microsoft.com/en-us/windows-hardware/drivers/hid/hidclass-hardware-ids-for-top-level-collections)) is not yet implemented in wine, and thus the code runs into an exception...

So, I think I'm going to look at implementing `multi TLC` in wine. This would also potentially allow the `FanatecSDK` to do it's thing (like driving LEDs/displays/etc).

melroy89, 2024-06-21:
This is awesome news! And glad you finally found the root cause of all this.

nrbk999, 2024-09-15:
I had this problem just yesterday.  Tried with proton version no option/experimental/9
Raceroom 64bits/Vulan/VR crashed
Raceroom 32bits didn't

After much hours wasted  launching "wine control" in the game prefix seems to have fixed the error
Manjaro+KDE

melroy89, 2024-11-23:
Is this still an issue or is there any update?

gotzl, 2025-01-03:
This is still an issue. Plan is to stabilize #86 and get proton/wine patched to be able to make use of this in the coming weeks.

closed 2025-02-01 by gotzl

gotzl, 2025-02-03:
Update: I've created the PR to get the required changes into wine-proton https://github.com/ValveSoftware/wine/pull/269