gotzl/hid-fanatecff #69: ClubSport Steering Wheel F1® Esports V2

salsadinner, 2024-01-31:
Hi!
This is my first racing wheel setup, so please excuse my ignorance if any.

I am trying to setup the [ClubSport Steering Wheel F1® Esports V2](https://fanatec.com/eu-en/steering-wheels/clubsport-steering-wheel-f1-esports-v2) with CSL DD1 on Arch linux. Everything is detected, but I see that the steering does not work as expected. I need two complete rotations in either direction during calibration to hit -100 or +100.

How do I go about fixing this ?

gotzl, 2024-02-01:
Which game is this? 
Generally you can adjust the range of the wheel with the SEN setting in the tuning-menu directly on the wheel. This is probably set to 'AUTO'... Try if it works better for you to set it eg to 108 (with would be 1080 degrees).

salsadinner, 2024-02-01:
Thank you so much, that fixed it!

closed 2024-02-01 by salsadinner