gotzl/hid-fanatecff #70: Conflict between Playstation 5 controller & Fanatec CSL DD

liszca, 2024-02-11:
On my system the conflict shows up with following symptoms, all ACC related didn't check other games:

Normaly the wheel is detected as "Fanatec Wheel".

Since I have used the Playstation5 controller in other games my Fanatec CSL DD wheel only shows up under the Playstation 5 controllers name "Wireless Controller". 

What I tried:
Disable Bluetooth - no change
remove device from Manager - no change
remove device from Manager with additional reboot - no change

But the thing is I don't want to have to disconnect the Playstation 5 controller.

Am I alone with this problem?

gotzl, 2024-02-11:
You are not alone, I have similar problems and I think there are also comments like this on acc proton GitHub issue tracker. I think it's s.t. in wine/proton and how Acc enumerates devices.

In my case, I have pedals and the wheel-base connected via USB and I have to connect them to the PC always in the same order, else acc confuses them.

Not sure what the best solution is in your case, probably make sure to always have only the wheel-base connected?

I think I'll make an entry to the readme about this issue, but I do think there is nothing I can do on the driver side to fix it...

liszca, 2024-02-13:
> In my case, I have pedals and the wheel-base connected via USB and I have to connect them to the PC always in the same slot/order, else acc confuses them.

I think that was the same under Windows, you had to use the same USB slot otherwise config is different.