gotzl/hid-fanatecff #71: Fanatec Handbrake Inverted

Oortael60, 2024-02-16:
Hi @gotzl ,
first off all thank you for the driver. Using it with my CSL DD is a charm.

I've recently purchased a Fanatec ClubSport Handbrake v1.5 for WRC. The game recognises the handbrake, but has inverted scales. When the lever is in neutral position the brake is at 100%.

Is there a way to tell the driver via configuration to reverse the direction?

Best regards
Stefan

gotzl, 2024-02-18:
Hi,

I have the brake as well. For me it works fine, but I've connected it directly to the wheel. I assume you use the USB adapter?

Oortael60, 2024-02-25:
Yes, directly connecting worked after setting the device to CSW compatibility (yellow) and back to standard (red).

But, I needed to remap the button for the handbrake.

closed 2024-02-25 by Oortael60