gotzl/hid-fanatecff #72: No FFB on manjaro

MpJupp, 2024-02-24:
Hi, 

first of all, thanks for all the efforts. Your driver might be the last missing piece for me to fully switch to linux. 

I am currently running manjaro and I want to use the csl dd. I installed the driver from the AUR and it seemed everything worked well, but ingame I am not getting any FFB. I can steer and the wheelbase is found, but no ffb (running in pc mode). 

Is there anything I can provide to track the issue? 

Thanks in advance.

gotzl, 2024-02-24:
Hi, can you try the proton6 trick? See the readme for more info..

MpJupp, 2024-02-24:
I tried, but still no FFB. I deleted the AUR packages because I read somewhere that it is outdated. 
Then I wanted to do a manual install as written in the readme, but the make fails:

:: Compiling Fanatec kernel module
========================================
make -C /lib/modules/`uname -r`/build M=$PWD
make[1]: Verzeichnis „/home/user/Downloads/hid-fanatecff-next“ wird betreten
make[1]: *** /lib/modules/6.6.16-2-MANJARO/build: Datei oder Verzeichnis nicht gefunden.  Schluss.
make[1]: Verzeichnis „/home/user/Downloads/hid-fanatecff-next“ wird verlassen
make: *** [Makefile:8: default] Fehler 2

MpJupp, 2024-02-24:
Issue is solved. 

I was missing the linux-headers on the relatively fresh install of manjaro. It seems that the AUR package is outdated. I used the zip-file and make. Now it works like a charm. Thanks for your support.

closed 2024-02-24 by MpJupp

gotzl, 2024-02-25:
Ah, glad you managed! The aur PKG also misses the dep to Linux-Headsets... @mcoffin could you update the aur?