gotzl/hid-fanatecff #74: WRC 8 no force feedback, wheel detection issues

Alan-James, 2024-03-20:
While Dirt Rally 2.0 works without issue, I'm having trouble with WRC 8 FIA World Rally Championship.

In PC mode my Fanatec CSL DD is not detected as a wheel but shows up as a controller. 
In compatibility mode it does show up as a Clubsport Wheel Base V2.5  but there's no force feedback (and I can't change any FFB settings)

I've tried different versions of proton from 6 up to 9 without any improvement.

Today the game is on sale (90% off), so I'm hoping someone else can try to reproduce this issue.

gotzl, 2024-03-30:
Missed the sale... Will look into it when it's on sale again.

gotzl, 2026-11-03:
I did some digging and found the issue. In essence, the game uses 'offsets' in object enumeration which are not translated correctly by wine. Fanatec wheels are special here, because their object enumeration doesn't start with 'X' at offset 0, but with 'POV'. This stems from the HID descriptor order, so other wheels probably have 'X' as first entry in the descriptor and don't hit this problem.

I've created a wine MR to fix the offset translation (https://gitlab.winehq.org/wine/wine/-/merge_requests/12499) as well as a corresponding PR for proton-ge (https://github.com/GloriousEggroll/proton-ge-custom/pull/928).