gotzl/hid-fanatecff #77: Left stick not detected on Formula wheel and Button Module Endurance

Cykyrios, 2024-04-27:
Some wheels, such as all variations of the Formula wheel, the Button Module Endurance, and I assume the same is true for the Clubsport Steering Wheel RS and BMW GT2, feature both a funky switch and an analog stick. The analog stick is not detected at all, except for its button trigger when pressing it.
I have already checked the stick is working properly on the Fanatec app on Windows, so definitely not a hardware issue.

gotzl, 2024-04-28:
Hi,
I can reproduce this with the Clubsport F1. It seems that the min/max values for these axis is not correct. A workaround would be to issue
`evdev-joystick --evdev /dev/input/by-id/usb-Fanatec_*-event-joystick --axis 3 --min -128 --max 128 --axis 4 --min -128 --max 128`

If this works for you as well I will think about a more permanent solution.

Cykyrios, 2024-04-28:
Yes, with this LFS now sees one of the 2 axes, but evtest does show values going from -128 to 128 on both RX and RY, so I'll have to assume there is an issue with LFS not being able to detect both (or wine), too bad it only detects RY when I would want to use RX...
The remaining issue is probably not related to this, so I think we can consider `evdev-joystick` fixes it. It seems SDL has been an issue on wine for some games, I'll see if I can find something there.

gotzl, 2024-04-28:
You can check what wine sees with `wine control joy.cpl`

Cykyrios, 2024-04-28:
Okay, I might have missed something, I checked wine control earlier and only had RY, but now I do have both RX and RY.

Cykyrios, 2024-05-01:
Just a quick note as I tried again this morning: if I try to set both axes at the same time, only one works, I have to set them separately for both axes to be detected by wine.

commit 918ff0a3 (2024-07-07): set analog stick ranges via udev rule (fixes #77)
closed 2024-07-07 by gotzl