gotzl/hid-fanatecff #78: BeamNG CSL DD problem

Riptils, 2024-05-02:
Hello!
I recently bought the CSL DD bundle pack on the fanatec website.
I am a linux user (archlinux to be precise) and I therefore used this driver to get my steering wheel working.
I also bought a Windows PC to update the firmware on the CSL DD, the steering wheel and the quick release. 

I play mainly (if not exclusively) BeamNG and I've noticed problems with slicing, movement not "real" force feedback poorly proporionated (crashes far too aggressive and roads too soft and lack of force feedback on the relief of the field ...).

I've made a video to show my main bug, which is excessive vibration of the steering wheel to make it go straight again. Here it is: https://drive.google.com/file/d/1Ofe8Y-C739JrGBHOTXhE8En3HKHtFS3X/view?usp=sharing

I saw that there was already an issue on BeamNG and that apparently bugs were known. If possible, I would like more detailed information on the problem and how it works.

Thank you

OverStyleFR, 2024-05-04:
Salut (translate to french to understanding).

J'ai eu un problème similaire a toi, cette à dire que mon Fanatec DD1 était reconnu mais je n'avais pas de force feedback ni de tremblement.

J'ai tout d'abord installer les drivers du repository actuelle (https://github.com/gotzl/hid-fanatecff).
Ensuite j'ai tout simplement redémarrer mon jeu, si ca marche pas désactive le fait d'utiliser Proton sur steam.

OverStyleFR, 2024-05-04:
@Riptils

Riptils, 2024-05-04:
Salut(translate to french to understanding).

Je parle français moi aussi.
Merci pour ta réponse. À vrai dire j'avais également testé avec Wine sous BeamNG(avec les dépendances nécessaires).

Pour info de mon côté ce que j'ai fait est : 
1. Mettre à jour les firmwares avec un pc Windows et calibrer le centre du volant avec fanalab.
2. Installer le paquet hid-fanatecff-dkms sur l'AUR de ArchLinux.
3. Redémarrer, brancher le volant, aller dans ses paramètres et définire le setup 1 définir le bon angle de braquage.
4. Lancer le jeu, mettre des paramettres, définir sur **Full PUIS sur FAST** (Full pour qu'il le détecte puis après avoir fait un tour le mode Fast fonctionne)
5. Faire la vidéo pour montrer les tremblements étranges.

A noter que j'ai testé le driver sur Assetto Corsa 1.16.4 et sur le jeu LiveForSpeed (dernière version :0.7E ). Le retour de force est bluffant et super réaliste sur ces jeux. Mais il ne sagit pas de BeamNG, le jeu auquel je désire jouer par dessus tous les autres.

De l'aide de la part de la communauté et du développer serait grandement appréciée ! 

Merci

Riptils, 2024-05-04:
@OverStyleFR

OverStyleFR, 2024-05-05:
> Salut(translate to french to understanding).
>
> Je parle français moi aussi. Merci pour ta réponse. À vrai dire j'avais également testé avec Wine sous BeamNG(avec les dépendances nécessaires).
>
> Pour [...]

Etrange...

Ce que tu as déjà fait ressemble a ce que j'ai fait. Malheuresement je pense pas que je serais le plus apte a t'aider :(

gotzl, 2024-05-12:
Hi, I just tried to replicate your issue, but I don't see this oscillation.
As I'm not too familiar with this game, can you tell me how I can load-up the exact situation you're showing in the video?

Riptils, 2024-05-13:
Hello, 
I've already replied to this in a previous message. Could you ask me for clarification on the part that you don't find precise instead?
I can add as information that I am on Archlinux, that I used the last versions of kernel (linux-mainline) with last drivers vulkan. The game runs without any additional parameters. If you want more information on what I use as an arch installation, here's the installation script I use: https://gitlab.com/greyxor/obscais and my dotfiles: https://gitlab.com/Riptils/dotfiles.
Finally, if you want to see my settings applied in the game, I show them in the video (for the FFB). I've assigned acceleration and braking to buttons (because I don't have any pedals at the moment). All assists are deactivated and the game is set as "Realistic".

I'd be happy to answer your questions if they were a bit more specific. Thanks for your help!

gotzl, 2024-05-13:
Well, all I wanted to know was which map/car combination you're using in the video. So I just used a 'Time Trail' on 'Industrial Site' with a 'Ibishu 200BX Type-LS (M)' which seems to come close to what you have in the video. I've applied all your settings but still, I cannot reproduce this issue. However, I'm using the ClubSport DD, so maybe this issue only shows with the CSL DD...

Some questions:
* do you get the issue in both the windows (proton) and the linux-native versions of the game?
* if you have windows, you don't get issue in windows-native?
* is the issue car dependent?

Riptils, 2024-05-13:
Okay, then: 
I use the ETK Driver Experience Centre card on the "West".
It was an Ibishu 200BX Type LS (A).

I launched the game again and did the text and miracle! the bug disappeared, but I find the force feedback weird, you can feel that it's not real and there's a sort of texture in the FFB of the steering wheel. The force feedback doesn't look very well simulated.

The effect seems to be corrected by switching to the yellow mode on the CSL DD.

I can't launch the binary version of linux.
I don't have a Windows machine, I had one just to update the firmware but it's not mine and clearly not powerful enough for beamng.
The problem wasn't specific to the car.

gotzl, 2024-05-13:
Thanks for your clarifications.

> I launched the game again and did the text and miracle! the bug disappeared, but I find the force feedback weird, you can feel that it's not real and there's a sort of texture in the FFB of the steering wheel. The force feedback doesn't look very well simulated.

This is with the exact same tuning-menu (and in-game FFB) settings as before ?

> The effect seems to be corrected by switching to the yellow mode on the CSL DD.

So you would say, in 'yellow mode' everything feels alright?

Riptils, 2024-05-13:
Yes, I did this test with the same settings. I had just set the force feedback to full at the start but I changed it back to fast after seeing that it didn't change anything apart from the need to go through the ffb full for the ffb fast to work properly.

I can't say that the ffb works perfectly because I still have a few small details in the driving that aren't very real, like too much oscillation of the steering wheel in a straight line, or details that are too grainy on tarmac and not enough on bumps/kerbs. I get excessive shaking when I lower the steering wheel smoothness, which gives me more detail on the road but less understanding of what's really going on under the wheel. On the other hand, increasing the smoothness gives a more real texture but makes the steering wheel feel more toy-like and gimmicky.

In a way, the yellow mode seems to be better supported than the red mode, but neither offers a real feel. As I said, I had tested this steering wheel on assetto corsa and live for speed and it was nothing like it, and despite the games physics being much less advanced I found that the ffb felt more real.

For information, I used to have a g27 for beamng, assetto and liveforspeed and they all worked very well with very well simulated force feedback. I just don't have the same feeling as before with beamng.
Thanks for all your help!

gotzl, 2024-07-13:
> I still have a few small details in the driving that aren't very real

Well, this can have multiple sources
* issue of the game
* issue of wine
* issue of the driver

I don't have any indication that there is s.t. fishy in wine or the driver here. BeamNG seems to use only `FF_CONSTANT`, you can't do much wrong here. So this might be an issue of the game itself and one would have to compare to how it feels in windows (with the exact same settings) to point it again in the direction of wine/the driver.

> yellow mode seems to be better supported than the red mode

This I don't understand. They should feel exactly the same, supposed you're using the same tuning-menu settings.

Note that the g27 is very weak compared to the CSL DD. So I would expect that it feels quite different and I would also not take how it felt with g27 as a reference .... The reference feeling from my perspective is what the device is doing ind Windows (with the same settings).

gotzl, 2025-12-07:
Just tested with `v0.37.6` of the game and `v0.2.2` of the driver. The native and proton-10.0-3 version of the game feel quite similar to me. I was able to use the CSL DD and Fanatec CSL Elite pedals w/o issue.
Please reopen if your experience differs.

closed 2025-12-07 by gotzl