gotzl/hid-fanatecff #80: Support for Fanatec ClubSport USB adapter

gotzl, 2024-05-12:
Fanatec offers a [USB-adapter](https://fanatec.com/eu-de/accessories/pedal-accessories/clubsport-usb-adapter) that allows connecting the CSL Handbrake/Shifter/Pedals to the PC directly (therefore they can be used w/o a Fanatec wheel).

The USB adapter has different modes to support the different devices. In windows, a firmware update tool allows switching between the different modes. (It seems to put the device into a bootloader-mode, then flashes the appropriate firmware).

Should be doable, but not sure how many people would actually use this ....

Insprill, 2026-02-05:
Support for this would be great to have :)

I have my CSL Elite Pedals hooked up to it, and they're currently only recognized in games running under Proton when it's explicitly set to use HIDRAW (`PROTON_ENABLE_HIDRAW=0x0eb7/0x1a95`). In native games (BeamNG) or Proton games (Dirt Rally 2) without that env var, they aren't recognized at all. I tried applying the same `FTEC_PEDALS` quirk used for other pedals, but either I didn't do it right, or it didn't work.

hosj12, 2026-04-07:
I have been attempting to get this exact use case to work, I am using a Logitech g29 wheel and a fanatec csl petal set. I haven't been able to get my petals to work with anything, but i can see that the device works. If you have a fix for this, any help would be appreciated!