gotzl/hid-fanatecff #81: Doesn't Work For OpenSuse?

RagingTout, 2024-05-18:
I am running Regata OS, based off OpenSuse. The make and install commands don't work. This is what I get when running the 'make' command from within the hid-fanatecff-0.1 folder:

make -C /lib/modules/`uname -r`/build M=$PWD
make[1]: Entering directory '/lib/modules/6.8.8-lp155.9-default/build'
make[1]: *** No targets specified and no makefile found.  Stop.
make[1]: Leaving directory '/lib/modules/6.8.8-lp155.9-default/build'
make: *** [Makefile:8: default] Error 2

I'm a noob at this stuff but I've been able to install this on Debian platforms so I am confused. Am I missing a dependency?

gotzl, 2024-05-18:
Looks to me you're missing what's called `kernel-headers` in most distros. Can you find a package like that and try again?

RagingTout, 2024-05-18:
It seems getting Linux-kernel-headers installed in RegataOS is beyond my scope, I am going back to Debian. Thanks anyways.

gotzl, 2024-07-13:
In case anyone tries again, `sudo zypper install kernel-default-devel` might do it?

closed 2024-07-13 by gotzl