gotzl/hid-fanatecff #87: Incorrect wheel_id of CSL Steering Wheel P1 V2

HardTy, 2024-07-14:
I'm not sure if this is the right place here.
I tried to get [gotzl/hid-fanatecff-tools](https://github.com/gotzl/hid-fanatecff-tools) running but got the following trace:

```text
Traceback (most recent call last):
  File "/home/rene/projects/public/hid-fanatecff-tools/tools/./fanatec_led_server.py", line 274, in <module>
    raise e
  File "/home/rene/projects/public/hid-fanatecff-tools/tools/./fanatec_led_server.py", line 252, in <module>
    threads.append(typ(ev, args.dbus, args.device, args.display))
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/rene/projects/public/hid-fanatecff-tools/tools/f1_23.py", line 15, in __init__
    fanatec_led_server.Client.__init__(self, ev, dbus, device, display)
  File "/home/rene/projects/public/hid-fanatecff-tools/tools/fanatec_led_server.py", line 65, in __init__
    self.wheel = self.get_wheel_code()
                 ^^^^^^^^^^^^^^^^^^^^^
  File "/home/rene/projects/public/hid-fanatecff-tools/tools/fanatec_led_server.py", line 74, in get_wheel_code
    return wheels_dict[wheel_id]
           ~~~~~~~~~~~^^^^^^^^^^
KeyError: '00'
```

I did some testing and found out that I get a wrong/no ID when I print the `wheel_id` in sysfs.

```shell
cat /sys/module/hid_fanatec/drivers/hid\:fanatec/0003\:0EB7\:0020.000D/wheel_id 
0x00
```

I fixed the problem by changing https://github.com/gotzl/hid-fanatecff-tools/blob/master/dbus/fanatec_input.py#L24, but that doesn't seem to be the right way.  :)

I am using the latest git tag [gotzl/hid-fanatecff:0.1.1](https://github.com/gotzl/hid-fanatecff/tree/0.1.1)

gotzl, 2024-07-21:
Yes... Unfortunately looks like reading out wheel-id needs more work .. I wonder why it's only `0x00` for you. Does it keep that value even after you've turned the wheel?

HardTy, 2024-07-24:
Yes, it keeps the same value.

mferraci, 2024-09-10:
I can confirm wheel_id only displays `0x00` whatever wheel is mounted on my Clubsport v2.5 base

SynthetikzZ, 2024-09-10:
I have the same issue, only displays 0x00.. if I cycle trough csw mode back to PC mode, it shows 0x04 for sometimes a couple of seconds to some minutes, then it goes back to 0x00. 

Fixed the fantec led server temporarily for me by hard coding the wheel code into it, otherwise it throws a key value error "00"

gotzl, 2024-09-17:
> How can I try the hidraw branch ? Any tutorial on patching proton so it can talk to the wheel ?

I just came back from holidays and some personal stuff, but I hope I can write some instructions within the next 2 weeks.

> Ps: also the wheel goes wild pressing all buttons mapped to the wheel when I change something in the tuning menu, resulting in me teleporting to pits and honking until I press the honk again.

Please create a new issue, this sounds really strange. Is this with the latest firmware files?

SynthetikzZ, 2024-09-17:
Was this like always.. only happens if I go to tuning menu and can happen once or twice afterwards.