gotzl/hid-fanatecff #89: How to disable rumble?

junelva, 2024-07-22:
I'm not familiar with interacting with sysfs. I tried `echo 0 >/sys/module/hid_generic/drivers/hid:hid-generic/0003:0EB7:183B.0005/rumble` but it said permission denied, sudo or not. Unfortunately, despite dmesg showing the pedals by name, without your driver they do not work for me in certain sims. If possible, I'd like to ask if you could teach me how to disable rumble when using your driver.

```
[    6.838289] input: FANATEC ClubSportPedal as /devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9:1.0/0003:0EB7:183B.0005/input/input21
[    6.838584] hid-generic 0003:0EB7:183B.0005: input,hidraw3: USB HID v1.11 Joystick [FANATEC ClubSportPedal] on usb-0000:00:14.0-9/input0
```

Thank you, hope it's no trouble to ask.

gotzl, 2024-07-22:
Hi,
doing this with sudo involves a little trick, s.t. like `echo 0 | sudo tee /sys/module/hid_generic/drivers/hid:hid-generic/0003:0EB7:183B.0005/rumble`
But I wonder why you require sudo in the first place. Could you show the output of `ls -l /sys/module/hid_generic/drivers/hid:hid-generic/0003:0EB7:183B.0005/rumble`? This should show that the file is writable by group `games`. If it's not showing the games group, then there is an issue in the udev rule...
So the best way to talk to these sysfs files would be to add your user to the `games` group (note that adding user to a group only takes effect in new sessions, check with `groups` what groups you're currently in).

I also wonder why you specifically want to disable rumble. Is the rumble active by default for you?

junelva, 2024-07-22:
Thank you. I noticed that the AUR package is out of date, and that was why things weren't right. So I built the latest version here. This resulted in correct devices and while I was able to echo 0 to the rumble file, there was no change in DR2. However, I haven't had time to try after a system reboot. Arch is finicky like this. But it's not a big deal, since DR1 doesn't have rumble for these pedals. The reason I wish to disable rumble is because I live in an apartment building and do not wish to vibrate the floor or wall.

gotzl, 2024-07-22:
Ah, I see. The `rumble` sysfs file is only there to control rumble via the linux driver. However, in DR2, the rumble is not using the linux driver, but speaks directly to the pedals via the FanatecSDK (afaiu).
If the game has no option to disable the rumble, then one possibility would be to remove the FatantecSDKs ability to talk to the device with `protontricks -c "wine reg add 'HKLM\System\CurrentControlSet\Services\winebus' /v DisableHidraw /t REG_DWORD /d 0x1" 690790`. I'm not sure what the side effects of this are, though.

junelva, 2024-07-24:
Very well, and it's no issue with these pedals thankfully; after a little observation, I saw that the vibration motors were wired with connectors accessible without even removing the control board. With a pair of pliers, I solved my issue. Thank you for the driver- works great. :)

closed 2024-07-24 by junelva