gotzl/hid-fanatecff #91: Support for CSL Elite Pedals V2

pmarkee, 2024-07-24:
Hi, first of all thank you for the existence of this project!

My question is basically the title, I take it the CSL Elite V2 pedals are not yet supported? It is seen by Oversteer, however it does not detect analog input, the pedals sit at 0% and jump to 50% when almost fully pressed in.

gotzl, 2024-07-24:
Hi,
in principle, you don't need an extra driver for the pedals. Their functionality is covered by the generic HID driver in the kernel. 
What could explain your issue would be that the ranges and/or deadzones are miss-configured. Could you check what `evtest` is seeing? And the output of `evdev-joystick -s /dev/input/by-id/usb-Fanatec_*-event-joystick`?
Does `evdev-joystick --evdev /dev/input/by-id/usb-Fanatec_*-event-joystick --deadzone 0 --fuzz 0 --flatness 0` maybe help?

pmarkee, 2024-07-24:
It seems to be possibly an issue with Oversteer, `evtest` shows analog inputs as expected and it also shows up correctly in-game. I plugged it into the wheelbase, but it also works via USB. ProductID is `0eb7:6206`. Thanks for the help!

One more thing if you don't mind, is it possible to set the force of the brake pedal, like in the official Fanatec software under Windows?

gotzl, 2024-07-24:
Ok, good to hear that it works! In case you connect it via USB, it might work similarly to the V1. For a quick test you could change https://github.com/gotzl/hid-fanatecff/blob/dcb93dc4af8ed9837b5fc5345c5eb26df02c3f0c/hid-ftec.h#L17 to your PID, rebuild and reload the module. Then, you should get a sysfs file called `load` which allows you to set the force...

If you connect it to the wheel, then currently, the only way to adjust the force is via the tuning-menu directly on the wheel.

SynthetikzZ, 2024-07-24:
> It seems to be possibly an issue with Oversteer, `evtest` shows analog inputs as expected and it also shows up correctly in-game. I plugged it into the wheelbase, but it also works via USB. [...]

Yes, you can change settings directly in the menu on your wheelrim. 

I don't know if gotzl fixed permission issues writing directly to sysfs files tho.

pmarkee, 2024-07-24:
Btw @gotzl do you accept donations? You really deserve a nice cold beer for all this work!

gotzl, 2024-07-24:
> Rebuilt and installed the driver, the value in the load file is 0 right now

Forgot to say that reading back the value doesn't work currently ;)

> But also the pedals are not showing up under evtest anymore 

That's unexpected ^^ Hmm.. If you have the time, could you retest? Maybe observe what `journalctl -xfk` says when you plug-in the pedals...

> Btw @gotzl do you accept donations? You really deserve a nice cold beer for all this work! And all other contributors too of course.

Thanks, I'll take the beer if you get to Germany :) Not sure about donations, I'll think about it ...

SynthetikzZ, 2024-07-24:
Wenn du in Köln wohnst kannste von mir direkt ne Kiste haben hahahaha

pmarkee, 2024-07-24:
> That's unexpected ^^ Hmm.. If you have the time, could you retest? Maybe observe what `journalctl -xfk` says when you plug-in the pedals...

```
júl 24 19:44:55 coolhostname kernel: usb 3-2.2: new full-speed USB device number 6 using xhci_hcd
júl 24 19:44:55 coolhostname kernel: usb 3-2.2: New USB device found, idVendor=0eb7, idProduct=6206, bcdDevice= 0.01
júl 24 19:44:55 coolhostname kernel: usb 3-2.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
júl 24 19:44:55 coolhostname kernel: usb 3-2.2: Product: FANATEC CSL Elite Pedals LC V2
júl 24 19:44:55 coolhostname kernel: usb 3-2.2: Manufacturer: FANATEC
júl 24 19:44:55 coolhostname kernel: input: FANATEC FANATEC CSL Elite Pedals LC V2 as /devices/pci0000:00/0000:00:08.1/0000:13:00.3/usb3/3-2/3-2.2/3-2.2:1.0/0003:0EB7:6206.0009/input/input24
júl 24 19:44:55 coolhostname kernel: fanatec 0003:0EB7:6206.0009: input,hidraw7: USB HID v1.11 Joystick [FANATEC FANATEC CSL Elite Pedals LC V2] on usb-0000:13:00.3-2.2/input0
```

Seems no `/dev/input/eventX` file for it. Also, I guess this is expected but with the unmodified 0.1.1 version the sysfs for it is under `/sys/module/hid_generic/drivers/hid:hid-generic/0003:0EB7:6206.XXXX`, with the patched one it's `/sys/module/hid_fanatec/drivers/hid:fanatec/0003:0EB7:6206.XXXX`.

gotzl, 2024-07-25:
> Seems no /dev/input/eventX file for it.

It is 'implicitly' created by the input subsys. You should see it at `/dev/input/by-id/usb-FANATEC_*`.
The change of the sysfs directory is expected, it shows that your patch really did work and the hid-fanatec driver is loaded for the device.

> Edit: not sure if it matters but it's also the same if the wheelbase is also plugged in
If you connect the pedals via USB, they are completely independent from the wheel-base ...

I just found I don't get input from the pedals anymore after `rmmod hid-fanatec` followed by `modprobe hid-fanatec`, unless I replug the USB. Maybe this was the case for you as well?

pmarkee, 2024-07-31:
Sorry for the late reply, I was away from home for a few days.

I've tried again and it's still not showing up in `evtest` by default, but via `sudo evtest /dev/input/by-id/usb-FANATEC_*` it works. Setting brake force through the `load` file works as well.

> I just found I don't get input from the pedals anymore after rmmod hid-fanatec followed by modprobe hid-fanatec, unless I replug the USB. Maybe this was the case for you as well?

I've managed to get this exact behavior. Worth noting that with the patched version, when I re-plug it it doesn't respect the previously set brake force (not referring to readback, but the actual input I see in `evtest`) so you have to set it again through the `load` file.

Anyway thanks again for your help!

commit 1d600ebf (2025-06-08): add support for csl add CSL Elite Pedals LC V2 loadcell (closes #91)
closed 2025-06-08 by gotzl