gotzl/hid-fanatecff #92: Installation issue on Arch / FFB not working

T0jan, 2024-07-27:
Hi, 

Thx for your great work projects like this motivated me to finally switch to linux also for gaming. However I am struggling a bit to get the Force Feedback to work. I build the files from source and installed them afterwards and everything seems to have worked, there were at least no error messages. The wheel gets detected and steering and all buttons work, but FFB isn't. The fftest output reported that none of the FFB functions are implemented:

```
fftest /dev/input/by-id/usb-Fanatec_FANATEC_Wheel-event-joystick 
Force feedback test program.
HOLD FIRMLY YOUR WHEEL OR JOYSTICK TO PREVENT DAMAGES

Device /dev/input/by-id/usb-Fanatec_FANATEC_Wheel-event-joystick opened
Features:
  * Absolute axes: X, Y, Z, RX, RY, RZ, Throttle, Rudder, Hat 0 X, Hat 0 Y, 
    [FF 00 03 00 00 00 00 00 ]
  * Relative axes: 
    [00 00 ]
  * Force feedback effects types: 
    Force feedback periodic effects: 
    [00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ]
  * Number of simultaneous effects: 0

Uploading effect #0 (Periodic sinusoidal) ... Error:: Function not implemented
Uploading effect #1 (Constant) ... Error: Function not implemented
Uploading effect #2 (Spring) ... Error: Function not implemented
Uploading effect #3 (Damper) ... Error: Function not implemented
Uploading effect #4 (Strong rumble, with heavy motor) ... Error: Function not implemented
Uploading effect #5 (Weak rumble, with light motor) ... Error: Function not implemented
Enter effect number, -1 to exit
```

The FFB test in oversteer also returns a similar error:

```
Exception in thread Thread-3 (test1):
Traceback (most recent call last):
  File "/usr/lib/python3.12/threading.py", line 1073, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.12/threading.py", line 1010, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/lib/python3.12/site-packages/oversteer/test.py", line 129, in test1
    left_effect = self.create_left_effect(0)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/oversteer/test.py", line 67, in create_left_effect
    left_effect.id = self.input_device.upload_effect(left_effect)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/evdev/device.py", line 340, in upload_effect
    ff_id = _input.upload_effect(self.fd, data)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: [Errno 38] Function not implemented
```

I also noticed, that there was no folder `hid_fanatec` created in `/sys/module/`, where the FFB settings would be expected to be stored.

Any idea what could have gone wrong here?

gotzl, 2024-07-28:
Hi,
after installing the driver, best is to unplug/replug the USB cable (or power cycle the wheelbase). If the hid_fanatec folder still doesn't appear then the driver is not properly installed for whatever reason... Which distro is this?

T0jan, 2024-07-28:
hmm I am pretty sure i tested that yesterday and even restarted the whole pc multiple times while trying to get it working but now it suddenly works. Sorry for the confusion and once again thanks for your great work.

closed 2024-07-28 by T0jan