gotzl/hid-fanatecff #93: Install nearly broke my system(?)

ProbablyJonathan, 2024-08-19:
I don't know if I did something wrong here, since i am rather new to Linux, but here's what happened:
I installed everything as per instructions and it appeared to be working. I am using a CSL DD, and it was recognized, but the FFB wasn't working yet. Also the wheel kept self centering, so i thought i must've done something wrong and I tried start over and install it again, except this time, when I ran "sudo make install" it just returned: "sudo: /usr/bin/sudo must be owned by uid 0 and have the setuid bit set".
Turns out that a bunch of my system's directories got assigned to the "games" group (e.g. bin, etc, lib, ...) and possibly changed other permissions (couldn't verify this), which made it borderline unusable. Luckily I was able to use Timeshift to restore an older system state.
Again, maybe I did something wrong here, but i'm 99% sure this install was the cause, so I thought I would let you guys know regardless.

gotzl, 2024-08-20:
Hi,
this sounds bad! I don't quite understand yet why this happened, but I've found a potential candidate that could cause this kind of behavior and submitted a fix for it just now. Thanks alot for reporting!
I'm glad that you were able to rescue your system, and I'm very sorry for the troubles.. Could you give it another shot and check your file perms?
Please open another ticket if you still have problems getting FFB working.
Thanks again!

mfatiga, 2024-10-04:
Is this fixed now? I do not want to risk it if this is still a problem.

ProbablyJonathan, 2024-10-04:
Hey, because im currently dual booting Windows I kinda forgot about this, sorry!
I will test it tomorrow :)

ProbablyJonathan, 2024-10-09:
This issue seems to be fixed, just installed everything without problems - FFB seems to be working as well!
Thanks a lot!

mfatiga, 2024-10-10:
Yep, seems to be working. I've tried to install it also and had no issues.

SynthetikzZ, 2024-10-11:
> Hey, because im currently dual booting Windows I kinda forgot about this, sorry! I will test it tomorrow :)

Dual booting with Windows is also known for destroying Linux installations. Seems unrelated to this topic, just fyi