gotzl/hid-fanatecff #98: Fanatec DD CSL - Wheel axis resolution low compared to windows

nrbk999, 2024-09-15:
Installed everything yesterday.
Manjaro+KDE, so I've used AUR packages.
Also installed Oversteer.

Wheel axis rotation resolution issues:
-is dependent on Wheel Sensitivity setting, the lower the setting the better the resolution.
With high sensitivty settings like 1080 or Auto I could clearly see it in game (raceroom) or oversteer or raw output from /dev/js1.

The minimum wheel turn to get a change in game axis or raw output was small but clearly noticeable.
In game was enough to feel strange.

gotzl, 2024-09-17:
Hi,
i don't follow. Could you try to explain again what you mean, preferably with examples for example for the raw output and what you would expect instead?
Thanks

nrbk999, 2024-09-19:
Issue is gone after Manjaro reinstall, I did not install Oversteer.
Tested with:
-Steam+Gamescope+ACC
-jstest /dev/input/whatever
Behavior is ok, high resolution.

Manjaro KDE and drivers from AUR hid-fanatecff-dkms 0.1.1-1.

closed 2025-02-01 by gotzl