gotzl/hid-fanatecff #99: Automobilista 2 crash at startup

sigonze, 2024-09-17:
Hi,
I notice that AMS2 is tagged as supported but I have a crash at the startup of the game (just after the first logo). 
After trying multiple combinations, I discovered that when my fanatec wheel is OFF (CSL DD), I have no issue at the moment I power it ON, the game just crash. 
I tried multiple proton (8, 9, experimental, GE) nothing change. 
I also did a fresh install. 
So if you have any clue I am very intrested. 
Thannks.

gotzl, 2024-09-17:
Hi,
this is a known issue, see #67. The workaround is to start the game once with proton6, then switch to a more recent proton version.

closed 2024-09-17 by gotzl

sigonze, 2024-09-17:
Sorry did not see the other issue.
I just confirm it works. Thank you for your help. :+1:

gotzl, 2024-09-17:
Glad it works now, no need to appologize!