gotzl/hid-fanatecff #102: FR: Add dkms

tim-gromeyer, 2024-10-24:
Hello,
can you please add support for dkms? So I don't need to manually build and install this module after every kernel update?

gotzl, 2024-11-17:
I personally don't use dkms. There is however the arch-pkg, which uses dkms afaik, so you could set it up based on that, and I'm happy to include it via a PR.

Helge102, 2025-03-01:
Here is my solution for DKMS with Debian. I think it will work with other distros too.
Copy the clone of the repo to /usr/src/hid-fanatec-0.1.2
Create 2 scripts in this folder and make them executable.

**post_install.sh**

_#!/bin/bash
cp -v /usr/src/hid-fanatec-0.1.2/fanatec.rules /etc/udev/rules.d/99-fanatec.rules
depmod_

**post-remove.sh**

_#!/bin/bash
rm -fv /etc/udev/rules.d/99-fanatec.rules
depmod_

You also need a dkms.conf in this folder.

_PACKAGE_NAME="hid-fanatec"
PACKAGE_VERSION="0.1.2"
MAKE[0]="make -C ${kernel_source_dir} M=${dkms_tree}/${PACKAGE_NAME}/${PACKAGE_VERSION}/build"
CLEAN="make -C ${kernel_source_dir} M=${dkms_tree}/${PACKAGE_NAME}/${PACKAGE_VERSION}/build clean"
BUILT_MODULE_NAME[0]="hid-fanatec"
DEST_MODULE_LOCATION[0]="/kernel/drivers/hid/"
AUTO INSTALL="yes"_

_POST_INSTALL="/usr/src/hid-fanatec-0.1.2/post_install.sh"
POST_REMOVE="/usr/src/hid-fanatec-0.1.2/post_remove.sh"_

Finally, the module must be added, built and installed.

_sudo dkms add -m hid-fanatec -v 0.1.2
sudo dkms build -m hid-fanatec -v 0.1.2
sudo dkms install -m hid-fanatec -v 0.1.2_

PR #129 (2025-09-19): Add DKMS
Hello,

I've re-used the DKMS scripts from *xone* to make them work here.

```sh
# dkms status
hid-fanatec/0.2.0-6-gd3b10ca, 6.14.4-mainline, x86_64: installed
```

Closes #102
closed 2025-09-19 by gotzl