gotzl/hid-fanatecff #103: Issue with deadzone and calibration.

webermaximilian92, 2024-10-29:
I installed the drivers and the wheelbase is shown and every button works. (Funky switch is based on axis somehow but nvm)

Now the first 90° left and right rotation is dead. (Value 0)

When I use the calibration it works very awkwardly, only half of the axis is working now the other half is flickering.

gotzl, 2024-11-17:
Hi,

Which wheelbase is it? what do you mean with `When I use the calibration`? Which calibration?

Can you use `evtest` to check if the deadzone is visible there as well? Can you post the output of `evdev-joystick -s /dev/input/by-id/usb-Fanatec_FANATEC_Wheel-event-joystick` (path may vary depending on your wheelbase)

webermaximilian92, 2024-12-03:
Hi,
oh sorry,

I missed so much information for this issue and also propably didn't installed the drivers correctly, but let's find out. 
Thank you!

- I do have a CSLDD
- I was talking about the linux (popos) intern device calibration (jstest-gtk)
- For you request the following was returned:

in between left and right there is no dead zone!

gotzl, 2024-12-08:
Could you post the output of `evdev-joystick -s /dev/input/by-id/usb-Fanatec_FANATEC_Wheel-event-joystick`?
Also is there a reason why you're using the wheel in compatibility mode?

webermaximilian92, 2024-12-08:
I think I have been wrong the whole time.
My AMS2 crashed as soon as my CSLDD was connected to my linux and I tried to check the inputs with the tool I mentioned but it simply was broken. 
I rechecked the input with your terminal commands and with rf2 and everything is fine now. 
Compatibility Mode was a mess up from me as well but it wasn't the main reason.

So thanks a lot. Cheers

closed 2024-12-08 by webermaximilian92

gotzl, 2024-12-08:
Alright, thanks for clarifying.
For the AMS2, see #67 and the workaround mentioned there.