gotzl/hid-fanatecff #104: typo in Udev rule

Helge102, 2024-11-01:
in line 3 of 'fanatec.rules' it says DRIVERS, but it must be DRIVER

gotzl, 2024-12-08:
Why do you think this is a typo? For me the rule works just fine
```
$ udevadm test /dev/input/by-id/usb-Fanatec_FANATEC_Wheel-event-joystick
....
  RUN{program} : /bin/sh -c '/usr/bin/evdev-joystick --evdev /dev/input/event28 --deadzone 0 --fuzz 0 --flatness 0; /usr/bin/evdev-joystick --evdev /dev/input/event28 --axis 3 --min -128 --max 128; /usr/bin/evdev-joystick --evdev /dev/input/event28 --axis 4 --min -128 --max 128'
...
```

Also, look at the output of `udevadm info -a /dev/input/by-id/usb-Fanatec_FANATEC_Wheel-event-joystick`

Helge102, 2024-12-09:
Honestly? ChatGPT told me.
Something seems to have gone wrong in my /etc/udev/rules.d folder.
I have now cleaned it up and no longer have any problems with the udev rule.

closed 2024-12-09 by Helge102

gotzl, 2024-12-09:
Ah, I see. Take it as a lesson, ChatGPTs lies are quite convincing ..
Glad your issue is resolved, though.