gotzl/hid-fanatecff #108: Rennsport Crashes when trying to launch with Fanatec CSL connected

azlekayn, 2024-11-11:
Rennsport Playtest on Steam crashes when ever I try to open the game. 

Only crashes when I have my CSL wheel and pedals connected, without them connected game boots up with no issues.

Operating System: Fedora Linux 41
KDE Plasma Version: 6.2.3
KDE Frameworks Version: 6.8.0
Qt Version: 6.8.0
Kernel Version: 6.11.6-300.fc41.x86_64 (64-bit)
Graphics Platform: Wayland
Processors: 16 × AMD Ryzen 7 7800X3D 8-Core Processor
Memory: 30.9 GiB of RAM
Graphics Processor: AMD Radeon RX 7800 XT

Tried last 2 versions of the Driver from this git hub page, same problem.

Tried different versions of proton - 

GE-9.13
Proton Experimental
Proton 9.0-3
Proton 8.0-7

removed prefixes and Shader cache

Nothing in Linux terminal to indicate what is crashing, just asks if I want to send a report.

azlekayn, 2024-11-11:
Installed the version from Heroic games and got the same problem, crashes in the exact same place with my wheel connected.

dinuxlt, 2024-11-11:
Maybe related with this one ? https://github.com/gotzl/hid-fanatecff/issues/67
> the workaround is to start the game once with proton6 to create the pfx, then switch to proton7/8

azlekayn, 2024-11-11:
> Maybe related with this one ? #67
> 
> > the workaround is to start the game once with proton6 to create the pfx, then switch to proton7/8

Nuking my prefix, installing with proton 6, then plugging in my wheel and booting with Proton Experimental fixed it!

N i c e!

Been looking forward to trying rennsport!

gotzl, 2024-11-17:
Closing as duplicate of #67

closed 2024-11-17 by gotzl